KLayout/klayout #5: __LP64__ not defined

devengr, 2017-02-28:
Conditional code has been added to db/dbHash.h with a comment "Specialization missing for size_t on WIN32" --on linux it looks like the directive __LP64__ should be set so this code is skipped, but that is not happening so compiling is currently broken.  If this small section is removed, compiling works correctly.

klayoutmatthias, 2017-02-28:
I can't confirm that. I'm using Linux myself. If I apply the change you propose, it gets broken in my case since there is a conflicting template specialization in the STL already.

What is the distribution you are using? Is it 32bit maybe? What is the compiler? Could you please give the error message?

Matthias

closed 2017-02-28 by klayoutmatthias

devengr, 2017-03-06:
Confirmed that problem is now fixed.
(Note--problem was occurring on 32 bit Ubuntu 14.04.5 using gcc 4.8.4)

commit 194a4cc4 (2017-02-28): Fixed #5