KLayout/klayout #8: Labels size should change when zoom in

osascruz, 2017-03-29:
Hi,  I just tried klayout , it is great job !
Just one quick feedback, if I zoom in the layout, the text of labels stay very small and do not scale.
I checked the property, in my case size is 0.02 . I would assume that zooming to grid size, the text should be  big.   
Thank you !

win8.1 -64bit   15.6 display 4K

klayoutmatthias, 2017-03-29:
That's actually intended, because this way you will see the labels always. If you're interested in finding a few small labels 

But there is a way to configure the behaviour you expect: see [http://www.klayout.de/doc/manual/create_text.html](http://www.klayout.de/doc/manual/create_text.html)

klayoutmatthias, 2017-03-30:
There is one more setting on the setup page: Display/General: Default Font Size. You can set it to "Large" for your display. 

Please carry this discussion to the forum: [https://www.klayout.de/forum/](https://www.klayout.de/forum/). This is the right place to discuss such things.

closed 2020-04-18 by klayoutmatthias