KLayout/klayout #12: Build fails due to float = qreal != double

tobp, 2017-04-24:
Building klayout 0.24.10 on openSUSE_13.2/armv7l fails with:

```javascript
src/gsiDeclQColor.cc:636:51: error: no matching function for call to 'QColor::getRgbF(double*&, double*&, double*&, double*&)'
((QColor *)cls)->getRgbF (arg1, arg2, arg3, arg4);
src/gsiDeclQColor.cc:636:51: note: candidate is: [...]
/usr/include/QtGui/qcolor.h:114:10: note: void QColor::getRgbF(qreal*, qreal*, qreal*, qreal*) const
      void getRgbF(qreal *r, qreal *g, qreal *b, qreal *a = 0) const;
```

The caller is (`src/gsiDeclQColor.cc:636`):

```javascript
static void _call_f_getRgbF_c4704 (const qt_gsi::GenericMethod *decl, void *cls, gsi::SerialArgs &args, gsi::SerialArgs &ret)
{
  double *arg1 = args.read<double * > ();
  double *arg2 = args.read<double * > ();
  double *arg3 = args.read<double * > ();
  double *arg4 = args ? args.read<double * > () : (double *)(0);
  ((QColor *)cls)->getRgbF (arg1, arg2, arg3, arg4);
```

klayoutmatthias, 2017-04-25:
Thanks for mentioning this ... getRgbF cannot be bound to the script system anyway because of the pointers. I'll drop that one.

Are there any other fails?

tobp, 2017-04-26:
`_call_f_getHsvF_c4704`  which calls `((QColor *)cls)->getHsvF (arg1, arg2, arg3, arg4);` has the same problem.

Untested, but I think the same problem as with `getRgbF` and `getHsvF` occurs with `getCmykF` and `getHslF` - all take a pointer to `qreal`.

tobp, 2017-04-27:
Yet another case:
```javascript
static void _call_f_stops_c0 (const qt_gsi::GenericMethod *decl, void *cls, gsi::SerialArgs &args, gsi::SerialArgs &ret)
{
  ret.write<qt_gsi::Converter<QVector<QPair<double, QColor> > >::target_type > ((qt_gsi::Converter<QVector<QPair<double, QColor> > >::target_type)qt_gsi::CppToQtAdaptor<QVector<QPair<double, QColor> > >(((QGradient *)cls)->stops ()));
}
```

Again, the problem is double <-> float:
```
no known conversion for argument 1 from 'QGradientStops {aka QVector<QPair<float, QColor> >}' to 'qt_gsi::CppToQtAdaptor<QVector<QPair<double, QColor> > >::source_type& {aka QVector<QPair<double, QColor> >&}'
```

tobp, 2017-04-28:
And another case:

```
src/gsiDeclQGraphicsAnchorLayout.cc: In member function 'void QGraphicsAnchorLayout_Adaptor::cbs_getContentsMargins_c4704_0(double*, double*, double*, double*) const':

src/gsiDeclQGraphicsAnchorLayout.cc:428:71: error: no matching function for call to 'QGraphicsAnchorLayout_Adaptor::getContentsMargins(double*&, double*&, double*&, double*&) const'
      QGraphicsAnchorLayout::getContentsMargins(left, top, right, bottom);

src/gsiDeclQGraphicsAnchorLayout.cc:428:71: note: candidate is:
/usr/include/QtGui/qgraphicslayout.h:66:10: note: virtual void QGraphicsLayout::getContentsMargins(qreal*, qreal*, qreal*, qreal*) const

note:   no known conversion for argument 1 from 'double*' to 'qreal* {aka float*}'

src/gsiDeclQGraphicsAnchorLayout.cc: In member function 'virtual void QGraphicsAnchorLayout_Adaptor::getContentsMargins(double*, double*, double*, double*) const':

src/gsiDeclQGraphicsAnchorLayout.cc:436:73: error: no matching function for call to 'QGraphicsAnchorLayout_Adaptor::getContentsMargins(double*&, double*&, double*&, double*&) const'
QGraphicsAnchorLayout::getContentsMargins(left, top, right, bottom);

note: candidate is:
/usr/include/QtGui/qgraphicslayout.h:66:10: note: virtual void QGraphicsLayout::getContentsMargins(qreal*, qreal*, qreal*, qreal*) const
```

tobp, 2017-05-02:
Yet another one:

```
src/gsiDeclQGraphicsAnchorLayout.cc:446:75:
error: no matching function for call to 'QGraphicsAnchorLayout_Adaptor::getContentsMargins(double*&, double*&, double*&, double*&) const'
          QGraphicsAnchorLayout::getContentsMargins(left, top, right, bottom);
                                                                            ^
note: candidate is:
/usr/include/QtGui/qgraphicslayout.h:66:10:
virtual void QGraphicsLayout::getContentsMargins(qreal*, qreal*, qreal*, qreal*) const
```

tobp, 2017-05-02:
Yet another one in the same file:

```
src/gsiDeclQGraphicsAnchorLayout.cc:446:75:
error: no matching function for call to 'QGraphicsAnchorLayout_Adaptor::getContentsMargins(double*&, double*&, double*&, double*&) const'
          QGraphicsAnchorLayout::getContentsMargins(left, top, right, bottom);
                                                                            ^
note: candidate is:
/usr/include/QtGui/qgraphicslayout.h:66:10:
virtual void QGraphicsLayout::getContentsMargins(qreal*, qreal*, qreal*, qreal*) const
```