KLayout/klayout #16: build fails on f53a3be (master as of today)

svenn71, 2017-08-22:
running a git pull
./build -qt5
yields the following error message.

I can build git revision 639eef0 without problems.

Am I missing some recently required development packages? I run on debian sid x86_64

```
g++ -m64 -Wl,-O1 -Wl,-rpath,/home/svennb/work/prg/git/klayout/bin-release -Wl,-rpath-link,/usr/lib/x86_64-linux-gnu -o ../../../../build-release/strm2cif main.o   -L/usr/X11R6/lib64 -L/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../.. -lklayout_bd -lQt5Designer -lQt5PrintSupport -lQt5MultimediaWidgets -lQt5Svg -lQt5Widgets -lQt5Multimedia -lQt5Gui -lQt5XmlPatterns -lQt5Network -lQt5Xml -lQt5Sql -lQt5Core -lGL -lpthread
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `db::compare_layouts(db::Layout const&, unsigned int, db::Layout const&, unsigned int, unsigned int, int, unsigned long, bool)'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `vtable for tl::ArgBase'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::CommandLineOptions::m_license[abi:cxx11]'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::CommandLineOptions::CommandLineOptions()'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::CommandLineOptions::~CommandLineOptions()'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `typeinfo for tl::ArgBase'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::CommandLineOptions::m_version[abi:cxx11]'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::CommandLineOptions::operator<<(tl::ArgBase const&)'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::ArgBase::~ArgBase()'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::CommandLineOptions::parse(int, char**)'
/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif/../../../libklayout_bd.so: undefined reference to `tl::ArgBase::ArgBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
Makefile:211: recipe for target '../../../../build-release/strm2cif' failed
gmake[3]: *** [../../../../build-release/strm2cif] Error 1
gmake[3]: Leaving directory '/home/svennb/work/prg/git/klayout/build-release/buddies/src/strm2cif'
Makefile:76: recipe for target 'sub-strm2cif-make_first' failed
gmake[2]: *** [sub-strm2cif-make_first] Error 2
gmake[2]: Leaving directory '/home/svennb/work/prg/git/klayout/build-release/buddies/src'
Makefile:43: recipe for target 'sub-src-make_first' failed
gmake[1]: *** [sub-src-make_first] Error 2
gmake[1]: Leaving directory '/home/svennb/work/prg/git/klayout/build-release/buddies'
Makefile:455: recipe for target 'sub-buddies-make_first' failed
gmake: *** [sub-buddies-make_first] Error 2
```

klayoutmatthias, 2017-08-27:
This issue was fixed with a0d864f3b6d903270d4dbb721ee654c95cf13b1e. 

I'm trying to ensure the build is succeeding, but I'm running incremental builds because my resources are limited. This case wasn't caught by the incremental build because the buddy exe's did not get linked again. 

Sorry for the inconvenience. Please try again with the lastest build. 

Matthias

closed 2017-08-27 by klayoutmatthias

svenn71, 2017-08-28:
Just to confirm that klayout now builds and runs.