KLayout/klayout #20: OSX version?

lukasc-ubc, 2017-11-19:
Thank you Matthias for the new version.  Very much looking forward to seeing if we can bring https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK into a KLayout package. That would certainly make it easier for the users to install and update their installations.

I'm wondering if you know if anyone is working on compiling an OSX version of 0.25?

Thank you
Lukas

klayoutmatthias, 2017-11-22:
Is anyone able to help here?

Matthias

Kazzz-S, 2017-11-27:
I started working on this!

Kazzz

lukasc-ubc, 2017-12-10:
Hi Kazzz,
Any further progress? I am thinking of hiring a junior programmer to help.

Kazzz-S, 2017-12-10:
Hi Lukasc,

I was busy these days and just came back to my work a couple of hour ago.
Since Matthias recommends to work with Qt5 first, I'm doing so.
One problem I found so far is that some Qt5 constants are not defined in OSX version, which causes compile errors.

To be honest, I'll need more time to go. So, any help is highly appreciated.

Kazzz

lukasc-ubc, 2017-12-14:
While building the new version, it might be worth re-considering which Python version is used. If the Windows version of KLayout uses Python 3, it would be nice to have consistency thus have the OSX version also use Python3. Too bad OSX ships with Python2, which would then require users to install Python3.  Unless it can be packaged within the build like on Windows?

klayoutmatthias, 2017-12-15:
Hi Lukas,

My experience is that using some "standard" installation will make it  
easier to obtain 3rd party packages. So if Python 2.7 is the standard  
on MacOS, then NumPy for example should be easy to obtain and install.  
And if MacOS behaves similar to Linux, these new packages become  
available to KLayout immediately, because they are in the default  
path. The availability of 3rd party packages and their simple  
installation is a considerable advantage.

I not sure how this works on MacOS. Maybe it's possible to achieve the  
same with Python 3 too. But maybe you can ask kazzz about this. BTW:  
on Linux, Python 3 is also not available always.

My personal opinion is that Python 3 is nice, but for simple  
applications, Python 2 and 3 are not that different. If wide  
deployment is required, confining the script to the common Python  
subset may be an option. In the worst case, scripts can ask for the  
python version they run on and use different code for these cases.

Best regards,

Matthias

He's making progress,

jasonfakidis, 2017-12-21:
Hi All, 

My name is Jason. I was introduced to this project by lukasc-ubc. I wanted to offer to help out with the OSX build for 0.25. 

I am generally free during daytime working hours (9-5) and am located in UTC-08:00 Pacific. 

You can contact me directly at: jfakidis@gmail.com

Regards,
JMF

lukasc-ubc, 2017-12-22:
Hi Matthias,

Related topic, iOS for iPads, etc...

There is not GDS viewer in the Apple Store, and I’ve had people ask me and they've looked around and not found anything… 

in 2012:

> That's a nice suggestion :-) But as far as I know, iOS does not support Qt. Hence, to my knowledge, there is no feasible way to port a Qt application to an iPad …

and in 2013:

> Regarding IOS: Qt5 differs substantially from Qt4 (for example *all* include paths have changed) and first experiences strongly suggest to wait for Qt5 to mature before shifting to that platform. Until then, KLayout won't build on Qt5, so IOS support is out of scope right now :-(

iOS supports Qt5.  So the question is whether there is a plan for KLayout to switch to Qt5 (now that it has been several years of Qt5, so presumably must be mature)...

https://wiki.qt.io/Transition_from_Qt_4.x_to_Qt5

> The transition from Qt 4.x to Qt 5 is not expected to be significant

jasonfakidis, 2017-12-27:
A summary of issues surrounding porting KLayout to mobile devices.

Q: Can we port KLayout to IOS/Android
A: Qt able to do that, but it is commercial (charge for ability to distribute an IOS/Android.)
-> Qt started as an environment for development
-> expensive to buy Qt license

 If we ported the application we would need to have UX/UI adaptations for iOS/ Android

-> As far as IOS/Android development goes, the framework is open source. However, deployment demands a license. We are not sure if you can deploy IOS except through the apple store (what about online?)

-----------KLayout Mac Issues---------
-> application port to mobile would not commute...
-> Another problem is that mobile applications much leaner than desktop applications. Could cause potential functionality issues.

** Work on this side project **
-> Investment has been skipped b/c of cost factor
-> Not able to accurately assess effort needed

klayoutmatthias, 2017-12-29:
Hi Jason,

kazzz has sent some information about the MacPort configuration. Will  
this be helpful to continue? I still am curious about what is  
preventing the application to start up.

Best regards,

Matthias

lukasc-ubc, 2018-01-10:
Hi Matthias and Kazzz,

It sounds like you are making great progress on compiling things for OSX.  Would it be beneficial to have me test the binaries you are producing?

thank you
Lukas

klayoutmatthias, 2018-01-11:
Hi Lukas,

I'm mainly focussing on debugging - my binary is running, but it's a  
plain binary and does not come as a package or as an installer. I have  
not installed a full suite as I am just running a weak VirtualBox  
instance, so right now I'm not able to produce more than that.

Best regards,

Matthias

Kazzz-S, 2018-01-11:
Hi Matthias and Lukas,

I'm preparing a new script to generate a DMG installer that works in combination with 
"build4mac.py" tool. 
I'm planning to add it after release of version 0.25.1.

Some works still remain, however, I've confirmed that it can be used as previous ones
(up to version 0.24.10).
Please give me some more time.

Warm regards,
Kazzz

Kazzz-S, 2018-01-16:
Hi Matthias and Lucas,

DMG installers are ready for trial.
Please visit,
https://www.dropbox.com/sh/174y11laig1y919/AADf_VGEatsA9_D6kJf5gYAva?dl=0　

There are two compressed DMG files.
1. klayout-0.25.1-MacOSX-Yosemite-1-Qt593mp.dmg
2. klayout-0.25.1-macOS-HighSierra-1-Qt593mp.dmg

No. 1 is for Yosemite, El Capitan, and Sierra.
No. 2 is for High Sierra.

Warm regards,
Kazzz

klayoutmatthias, 2018-01-16:
Hi kazzz,

thanks a lot! I tried High Sierra on my VirtualBox instance.

The installation was a new experience for me, but I finally noticed I  
had to drag the icons :-)

The application was copied successfully. Initially I wasn't able to  
run the application because the default security settings did not  
allow running applications from a unidentified source. I could solve  
this by allowing this application to open in the security settings. I  
had a similar issue on Windows where signed binaries are required now.  
While this issue is solved by a public/private key pair on Linux, I  
had to purchase a code signing certificate to be able to provide a  
signed Windows binary. I hope that's not the way MacOS goes too ...

But even after solving that issue, the application did not start  
(libruby.2.0.0.dylib missing). I only have a screenshot which I have  
attached. Hopefully, this is of help.

Thanks and best regards,

Matthias

lukasc-ubc, 2018-01-16:
Hi Kazzz, Matthias,

Amazing! Thank you for getting it working. I have been able to get my SiEPIC packages installed and tested. Some feedback below.

Initial dialog box cropped fonts.
￼

Tools | Manage Package: SiEPIC_EBeam_PDK installation causes KLayout to crash, even before it starts loading the files.
￼

Tools | Manage Package: SiEPIC_Tools installation ok.

2nd try: 
Tools | Manage Package: SiEPIC_EBeam_PDK installation ok.

In SiEPIC, the menus are attached using the built-in macro binding. The keys don’t work initially, but when you click on the menu (SiEPIC), then escape, then they work.  e.g, Shift+W. But it is temporary and the keys stop working after a while and you need to click the menu again.  Interestingly another way to fix it is Command-Tab to another program, then bag again; seems to reset the key bindings.
To reproduce:
- Install both EBeam and Tools. 
- new layout
- select layer Si
- create a path with a few vertices
- menu SiEPIC | Waveguides | Path to Waveguide (or W)
- then undo the waveguide, using Shift-W (Waveguide to Path)
Do it back and forth a few times, W, shift-W.  Sometimes it works, and other times you have to do the trick above..
￼
I tried to create a simple case that shows this problem by creating a macro in the local environment, but couldn’t reproduce it.  Perhaps this only happens for a package.

Somehow I got a greyed out set of menus in SiEPIC.  Never seen this before.
￼
Switching away from KLayout and back again seems to have fixed the grey menus.

[image]

[image]

[image]

[image]

Kazzz-S, 2018-01-17:
Hi Matthias,

Thank you for testing. It's weird: VirtualBox issue???  :-(

1. The Ruby bundled with **High Sierra** is version 2.3.x as below. Please confirm this in your environment.
[image]
2. **High Sierra Version KLayout** is built and deployed to link the Ruby library with its absolute path.
[image]
3. On successful start up, "About" dialog shows this.
[image]

----
On the other hand, 
1. **Yosemite Version KLayout** is build and deployed to link the Ruby version 2.0.x.
[image]
2. The Ruby bundled with **Sierra** is version 2.0.x as below.
[image]
3. So, on successful start up, "About" dialog shows this.
[image]
----

In summary...
- **Yosemite**, **El Capitan**, and **Sierra** are bundled with **Ruby 2.0.x**. So I made one package for these three. 
- However, **High Sierra** is bundled with **Ruby 2.3.x**. So I needed to make anther package specially for **High Sierra**.
----
If you have installed **Yosemite Version KLayout** to **High Sierra** by chance, it attempts to link Ruby version 2.0.x but there is Ruby 2.3.x only. 
Such discrepancy (no binary compatibility) should lead to failure.

As you have experienced, the installation procedure is just copying files by drag-and-drop, where platform compatibility is not checked.
I've seen several commercial tools that check platform compatibility during installation.
There must be a way to do such a sanity check. But it's beyond my knowledge so far.

Warm regards,
Kazzz

klayoutmatthias, 2018-01-17:
Hi kazz,

thanks very much ... you're perfectly right - that was my mistake.  
Everything is fine now. I posted my complete reply in the Google group.

Thanks for your efforts and best regards,

Matthias

lukasc-ubc, 2018-01-17:
Error message dialog windows are wider than my screen. Example:

[image]

lukasc-ubc, 2018-01-18:
1) Greyed out menus - I am seeing this all the time, and constantly need to switch away from KLayout and back.

2) minor issue: When running with multiple user accounts, switch back and forth with KLayout open, the program ended up in B&W mode.  Restarting KLayout solved it.

[image]

Kazzz-S, 2018-01-18:
Merged Kazzz-S#23 and rebuilt 1d30cc88.
The DMG installers under https://www.dropbox.com/sh/174y11laig1y919/AADf_VGEatsA9_D6kJf5gYAva?dl=0
have been replaced.

Kazzz

klayoutmatthias, 2018-01-18:
> <img width="661" alt="screen shot 2018-01-16 at 2 30 36 pm"
> <img width="1309" alt="screen shot 2018-01-16 at 2 31 17 pm"
> <img width="588" alt="screen shot 2018-01-16 at 2 42 26 pm"
> <img width="617" alt="screen shot 2018-01-16 at 3 01 00 pm"

Hi Lukas,

I noticed the gray menu issue, but so far only on the "recent file"  
menu. But having this on the bound menu items is annoying. I'll take a  
look into that.

I don't get a crash when I try to install siepic_ebeam_pdk from  
Salt.Mine (at least not in version 0.3.4). When I let it run the  
macros after the installation I get some errors saying "Stream has  
unknown format". I assume the macros are trying to load a layout file  
and for some reason that fails. But the package is installed.

Best regards,

Matthias

lukasc-ubc, 2018-01-18:
The "Stream has unknown format” is something I was going to report, but it is a warning so I’m not too worried about it.  It happens on both Windows and OSX.  And it is indeed occurring when loading GDS files, which were created by KLayout. So I have no idea why this is happening.  It occurs when creating a library, in the macro: SiEPIC-IME Library.lym

>    self.layout().read(os.path.join(os.path.dirname(os.path.realpath(__file__)), "SiEPIC-EBeam.gds"))

lukasc-ubc, 2018-01-21:
Here are screenshots from 0.24.10 and 0.25.1, illustrating the problem.  OSX High Sierra.  Screenshot taken using Hotkey Shift+Command+4.

[image]
[image]

klayoutmatthias, 2018-01-21:
> <img width="909" alt="screen shot 2018-01-20 at 10 18 55 pm"
> <img width="908" alt="screen shot 2018-01-20 at 10 12 29 pm"

Hi Lukas,

thanks for the images. I can't look into this issue currently, but  
I'll come back to you.

Best regards.

Matthias

lukasc-ubc, 2018-01-27:
Matthias, Kazzz,

Just wanted to say thank you for building the OSX version.  I had a workshop this past week where I taught silicon photonics design using KLayout and SiEPIC-Tools. We had about 10 people using MacOS and the above 0.25.1 version.  Things generally worked well!  At least I don't have any new KLayout bugs to report (we sure found many in SiEPIC-Tools particularly since we were nightly introducing new features...)

thanks again,
Lukas

klayoutmatthias, 2018-01-29:
Hi Lukas, hi kazzz,

I have fixed a few potential issues on the master branch. But a major  
issue remains: the disabled menus appear to be a Qt bug  
(https://bugreports.qt.io/browse/QTBUG-63866). I tried to find a  
workaround, but so far without success. I'm not familiar enough with  
the Cocoa API to understand the details of the Qt implementation.

I'll try a few things, but if Qt4 provides a better solution, my  
suggestion was to switch to Qt4.

Unfortunately I have little bandwidth to spend on this right now.  
Hopefully I'll be back in a few days.

Best regards,

Matthias

lukasc-ubc, 2018-01-30:
Hi Matthias,
That’s a pretty big bug, which I’m surprised they haven’t fixed considering that Qt has been around for such a long time, and that it is a company…
At least we have a work-around (Command-Tab, twice).  I think it only shows up as a problem when a dialog or GUI appears.

Kazzz-S, 2018-01-30:
Hi Matthias,

I've attempted to build the latest snapshot with Qt4. But got some compile errors.

Kazzz
[qt4.build.macos-Yosemite-release.zip](https://github.com/klayoutmatthias/klayout/files/1677003/qt4.build.macos-Yosemite-release.zip)

OfficialVinsmokeSanji, 2018-01-31:
Hi Kazzz 
thank you very much for compiling the dmg file and uploading. 
i am using Sierra and this version works with Qt593 and gives much better graphics than the 0.24 provided on the klayout website with Qt487

thomaslima, 2018-02-06:
Hi Kazzz, Matthias,

I think most Mac users use `brew` instead of `macports` for package management. I added a few options for ruby2.5, python3.6 and qt5.10 from brew and am testing some compilation. If it works, I will post here. I can probably submit a merge request too.

Best,
Thomas

Kazzz-S, 2018-02-06:
Hi Thomas,

Good to know that "brew" can be used, too.

> Update: Compilation went well, but qt5.10 causes bad layout rendering.

I think this is the same issue as Kazzz-S#17.
Please try 5.9.3 instead.

Regards,
Kazzz

thomaslima, 2018-02-06:
It works well with 5.9.4 that I downloaded. I used python 3.6.4 and ruby 2.5 from Brew. Let me upload the updated code for build4mac to my fork.

klayoutmatthias, 2018-02-06:
Hi Thomas,

do you see any issues with the Menus on this combination? For example,  
disabled menu items in the "File/Open Recent" submenu?

Thanks and kind regards,

Matthias

thomaslima, 2018-02-07:
hi Matthias,

Nope, not at all. I have not noticed any GUI issues yet. The only issue I have is that I cannot distribute this compiled version to anyone. They need to have the exact same python, ruby and qt5 directories otherwise the program crashes. I wonder if there is a way to package them together with the app...

PS: Now that you mentioned it actually happened once. the file/open recent files were greyed out. But it seems rare and random. Problem disappeared when I tried again and could not reproduce.

lukasc-ubc, 2018-02-25:
Hi @klayoutmatthias, @Kazzz-S,

Regarding the Qt issue of the menus being disabled on MacOS, here is a discussion where someone was able to solve it. I don't know if that will apply to your case.

https://forum.qt.io/topic/79992/qt-5-8-main-menu-items-disabled-in-osx/15

thomaslima, 2018-03-07:
Hello everyone,

Sorry for the delay. I have submitted a pull request after having done some more work on this. I finally was able to produce a release .dmg with qt5.9.4 that worked on at least one computer other than mine. Please read PR #86 for more details. I have also attached a .dmg for download.

This is for those of you with python3 installed via homebrew (or otherwise shipped with a Framework located in `/usr/local/opt/python/Frameworks/Python.framework`). What is curious is that I opened the klayout binary with the terminal in a computer without this installation of python, and it still worked (unexpectedly to me). 

[image]

Another feature of this build is that you can open klayout from terminal with a symbolic link. I use `ln -s /Applications/klayout.app/Contents/MacOS/klayout /usr/local/bin/klayout`.

thomaslima, 2018-09-22:
I was just navigating through the open issues I've participated in... I think this one can be closed, given that we have good binaries for klayout 0.25.x for mac. 

I'd like to point out that my tip from above is not a good one because it changes the @executable_path to `/usr/local/bin` instead of `/Applications/klayout.app/Contents/MacOS/`, this will cause python to malfunction inside. Instead, create a bash script in /usr/local/bin/klayout like such:
``` bash
#!/usr/bin/env bash
/Applications/klayout.app/Contents/MacOS/klayout "$@"
```
Now you can launch klayout from command line from anywhere.