KLayout/klayout #21: Autorun(-early) doesn't seem to run when lym files are inside a package

davidnhutch, 2017-11-21:
Something is broken about the macro autorun feature, under certain circumstances, in 0.25

Assumed problem:

It seems  the autorun-on-startup feature probably only looks in KLayout/macros folder right now. I think it should also look in KLayout/salt/*/macros and KLayout/salt/*/*/macros. (The latter case is applicable when you have a package inside a group.) This enables dependencies between packages without needing to "load 'filename.rb'" to manually run one code from the other code.

Steps to reproduce the problem:

# Create file1.lym and file2.lym under macros folder (the main macros folder, not inside a package)

file1.lym:
    # Choose properties > Run early on startup
    module MyModule
      class MyClass
        def MyMethod
          p 'hello world'
        end
      end
    end

file2.lym:
    include MyModule
    MyClass.new.MyMethod

# Make file1.lym be auto-run early.
# Close then open KL.
# Run file2.lym via the Macro Editor. It works! (Nothing is wrong.)
# Now, create a package named "mypackage" using Tools > Manage Packages. Use the Ruby template.
# Move file1.lym and file2.lym to salt/mypackage/macros using the OS filesystem. (Or create them anew there and delete the old ones.)
# Close then open KL.
# Try to run file2.lym via the Macro Editor. It doesn't work. (Apparently file1.lym has not run, so it doesn't recognize MyModule.) 
# Now, if you manually run file1.lym in the Macro Editor, then run file2.lym, it works. This indicates that there is nothing broken about the code. It is just that file1.lym is not autorunning on startup.

klayoutmatthias, 2017-11-22:
I see. You're are probably referring to our discussion in the forum :-) 

My suggestion would be to use plain ruby libraries in the "ruby" subfolder. They can be included through the standard "require". 

But I'll take a look at the issue. Thanks for reporting the problem.

klayoutmatthias, 2017-12-02:
I have tried to provide a fix on the master branch. The fix goes a bit deeper than I was expecting. Maybe you have a chance to test it and give feedback.

Thanks, 

Matthias

commit 1bb662b9 (2017-12-02): Fixed #21 (Autorun(-early) doesn't seem to run when lym files are inside a package)
closed 2017-12-02 by klayoutmatthias