KLayout/klayout #22: DRC not acting on texts

davidnhutch, 2017-11-21:
The following DRC command acts on shapes like polygons, but not on texts: input(1).output(2)

Is this intentional? I assumed everything on a given layer will be read in on "input" command, including texts.

klayoutmatthias, 2017-11-22:
Hi David,

The DRC engine is a geometry engine. It's working objects are edges and polygons. Texts are not handled - even if you just input/output a layer, texts won't be copied. So that is a known limitation.

The only exception is "texts" which can select texts by their string and turn them into dot-like polygons. It can be used on original layers only and will use the layer's meta information to take the texts from. So it's not a true text processing function. A description can be found [here](http://www.klayout.de/doc-qt4/about/drc_ref_layer.html#h2-1893).

davidnhutch, 2017-11-22:
I see the reason. Just checking it wasn't a bug.

No big deal, I can override DRCEngine methods to do whatever I want I suppose. Thanks!

closed 2017-11-22 by davidnhutch