KLayout/klayout #24: Text insert dialog possible bug in 0.25 but not in 0.24

davidnhutch, 2017-11-21:
There seems to be an issue with the text insert dialog. Steps to reproduce the problem:

1) KLayout 0.25 on Win10 64 bit. New layout, new cell named TOP, new layer 1/0.
2) Click "Text" tool, dialog pops up. Change string to something like "test". 
3) Click OK then place the text. It doesn't work! (it places the old string not the new one). If you click Apply then place the text it DOES work though.
4) Note that clicking OK then placing the new text DOES work on 0.24.10 but not on 0.25.0.

klayoutmatthias, 2017-11-22:
Thanks for reporting this. I think the same is true for the path properties and instance dialog. Closing these dialogs with "Ok" does not imply "Apply".

davidnhutch, 2017-11-22:
Ok

Clicking OK did imply "Apply+Close" in version 0.24.x (and possibly earlier versions), according to a brief test I did. I suppose that's a typical action implied by "OK" in OK/Apply/Close dialogs too, so it worked well.

I guess I'm wondering if there was a reason for the change?

klayoutmatthias, 2017-12-01:
Actually too many lines got deleting during one of one code restructuring activities - it was easy to recover. Thanks for reporting this issue.

commit 54a37357 (2017-12-01): Fixed #24 (Text insert dialog possible bug in 0.25 but not in 0.24)
closed 2017-12-01 by klayoutmatthias