KLayout/klayout #25: Linux compiliation fail with tlHttpStream  (sendCustomRequest)

Moles1966, 2017-11-22:
Hello Matthias,
great new features we want to try with 0.25 ... but I run  into a compiling issue with the linux build, we're using RedHat  6.6.

If I edit the Makefile in build-release/tl/tl and delete the http file from compling it won't stop.
./build.sh : 

.....
g++ -c -m64 -pipe -fvisibility=hidden -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -m64 -mtune=generic -D_REENTRANT -Wall -W -pedantic -Woverloaded-virtual -Wsign-promo -Wsynth -Wno-deprecated -Wno-long-long -Wno-strict-aliasing -Wno-deprecated-declarations -fPIC -DHAVE_QTBINDINGS -DHAVE_PYTHON -DHAVE_RUBY -DHAVE_RUBY_VERSION_CODE=10901 -DKLAYOUT_VERSION=0.25 -DKLAYOUT_VERSION_REV= -DKLAYOUT_VERSION_DATE=2017-11-22 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=25 -DKLAYOUT_TINY_VERSION=0 -DMAKE_TL_LIBRARY -DQT_NO_DEBUG -DQT_SCRIPT_LIB -DQT_SQL_LIB -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I/usr/lib64/qt4/mkspecs/linux-g++-64 -I../../../src/tl/tl -I/usr/include/QtDesigner -I/usr/include/QtCore -I/usr/include/QtNetwork -I/usr/include/QtGui -I/usr/include/QtXml -I/usr/include/QtSql -I/usr/include/QtScript -I/usr/include -I/home/scratch/andreas/kl/build-release/tl/tl/. -I/home/scratch/andreas/kl/build-release/tl/tl/. -I../../../src/tl/tl -I. -o tlHttpStream.o ../../../src/tl/tl/tlHttpStream.cc

../../../src/tl/tl/tlHttpStream.cc: In member function 'void tl::InputHttpStream::issue_request(const QUrl&)':
../../../src/tl/tl/tlHttpStream.cc:189: error: 'class QNetworkAccessManager' has no member named 'sendCustomRequest'

../../../src/tl/tl/tlHttpStream.cc:195: error: 'class QNetworkAccessManager' has no member named 'sendCustomRequest'

gmake[2]: *** [tlHttpStream.o] Error 1

Best Regards,
Andy

klayoutmatthias, 2017-11-22:
Hi Andy,

The build requirements are documented here: [http://www.klayout.de/build.html](http://www.klayout.de/build.html). Qt 4.7 is required at least. The Qt version available by default in RH6/CentOS6 is too old. 

Unfortunately there is nothing that substitutes "sendCustomRequest" in Qt <4.7, so the only option is to disable the package manager.

Matthias

Moles1966, 2017-11-22:
Thanks for feedback Matthias, 
I need to crosscheck but I used 4.8 something. 
Especially Salt is a very nice and useful add-on to distribute packages to different departments.

BR,
Andreas

closed 2017-11-22 by Moles1966