KLayout/klayout #35: 32bit build on debian stretch fails due to size mismatch

timkb4cq, 2017-12-12:
```
In file included from ../../../src/db/db/dbOASISWriter.h:31:0,
                 from ../../../src/db/db/dbOASIS.cc:26:
../../../src/db/db/dbHash.h: In instantiation of ‘std::size_t __gnu_cxx::hfunc(const T&) [with T = long long int; std::size_t = unsigned int]’:
../../../src/db/db/dbHash.h:124:84:   required from here
../../../src/db/db/dbHash.h:107:15: error: no match for call to ‘(__gnu_cxx::hash<long long int>) (const long long int&)’
     return hf (t);
            ~~~^~~
Makefile:3716: recipe for target 'dbOASIS.o' failed
```

klayoutmatthias, 2017-12-12:
I'm no longer trying to support 32bit on Linux. That would double my continuous integration efforts.

Please try to replace line 89 of src/db/db/dbHash.h with this:

```
#elif defined(_WIN32) || (defined(__GNUC__) && defined(__i386__))
```

and let me know whether this works.

danfe, 2021-07-12:
Yes, this fixes the build on 32-bit architectures, thanks! (powerpc suffers the same as i386 here), so either both `__i386__` and `__powerpc__` should be listed there, or some more generic check, e.g. `#elif sizeof(void *) == 4` used instead.