KLayout/klayout #37: OK button in dialogs doesn't "Apply" parameters

lukasc-ubc, 2017-12-16:
In v0.24 and earlier, one would either click on Apply or OK, and the changes in dialogs would be accepted, e.g., in Path, Instance, etc.  

Now, in v0.25, if I click on Path, then change it to 0.5, then OK, the path that I draw comes out at 0.1, instead of 0.5.  In order to get 0.5, I have to click Apply, then OK.  So OK isn't performing the "Apply" operation first...

klayoutmatthias, 2017-12-16:
Hi Lukas,

this looks like a duplicate of #24 which is already fixed.

Regards,

Matthias

mustafacc, 2017-12-20:
Hi Matthias, I'm having the same issue with the 0.25 version downloaded from the website (tried on both Windows and Ubuntun).  Is the issue fixed on the released version?

mustafacc, 2017-12-20:
Just confirmed that the issue is only with the website downloaded version. I built the master branch version on my Ubuntu machine and the issue is fixed.

klayoutmatthias, 2017-12-21:
Hi Mustafa,

Thanks for this confirmation. The master branch is were development happens and this big fix will be released in 0.25.1.

Kind regards,

Matthias

closed 2018-02-24 by klayoutmatthias