KLayout/klayout #47: klayout-0.25 build error in Gentoo (-expert)

akhuettel, 2017-12-27:
I'm getting the following build error for 0.25... still investigating what's happening, but any advice is appreciated. Best, Andreas

```
x86_64-pc-linux-gnu-g++ -c -march=native -O2 -pipe -g -fvisibility=hidden -Wall -W -pedantic -Woverloaded-virtual -Wsign-promo -Wsynth -Wn
o-deprecated -Wno-long-long -Wno-strict-aliasing -Wno-deprecated-declarations -D_REENTRANT -fPIC -DHAVE_QTBINDINGS -DHAVE_PYTHON -DHAVE_RU
BY -DHAVE_RUBY_VERSION_CODE=20200 -DKLAYOUT_VERSION=0.25 -DKLAYOUT_VERSION_REV= -DKLAYOUT_VERSION_DATE=2017-12-27 -DKLAYOUT_MAJOR_VERSION=
0 -DKLAYOUT_MINOR_VERSION=25 -DKLAYOUT_TINY_VERSION=0 -DMAKE_GSIQT_LIBRARY -DQT_NO_DEBUG -DQT_DESIGNER_LIB -DQT_UIPLUGIN_LIB -DQT_PRINTSUP
PORT_LIB -DQT_MULTIMEDIAWIDGETS_LIB -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_XMLPATTERNS_LIB -DQT_NETWORK_LIB -
DQT_XML_LIB -DQT_SQL_LIB -DQT_CORE_LIB -I../src/gsiqt -I/chroot/largetmp/portage/sci-electronics/klayout-0.25/work/ruby22/gsiqt -I../src/t
l/tl -I../src/gsi/gsi -I../src/db/db -I../src/gsiqt -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtDesigner -isystem /usr/include/q
t5/QtUiPlugin -isystem /usr/include/qt5/QtPrintSupport -isystem /usr/include/qt5/QtMultimediaWidgets -isystem /usr/include/qt5/QtSvg -isys
tem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtMultimedia -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtXmlPatte
rns -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtXml -isystem /usr/include/qt5/QtSql -isystem /usr/include/qt5/QtCore -
I/chroot/largetmp/portage/sci-electronics/klayout-0.25/work/ruby22/gsiqt -isystem /usr/include/libdrm -I/usr/lib64/qt5/mkspecs/linux-g++ -
o gsiDeclQIconEngine.o ../src/gsiqt5/gsiDeclQIconEngine.cc
In file included from ../src/gsi/gsi/gsiDecl.h:45:0,
                 from ../src/gsiqt/gsiQt.h:27,
                 from ../src/gsiqt5/gsiDeclQIconEngine.cc:36:
../src/gsi/gsi/gsiClass.h: In instantiation of ‘void gsi::_assign(tl::true_tag, void*, const void*) [with X = QIconEngine_Adaptor]’:
../src/gsi/gsi/gsiClass.h:824:16:   required from ‘void gsi::Class<T, A>::assign(void*, const void*) const [with X = QIconEngine_Adaptor; 
Adapted = gsi::NoAdaptorTag]’
../src/gsiqt5/gsiDeclQIconEngine.cc:910:1:   required from here
../src/gsi/gsi/gsiClass.h:465:14: error: use of deleted function ‘QIconEngine_Adaptor& QIconEngine_Adaptor::operator=(const QIconEngine_Adaptor&)’
   *(X *)dest = *(const X *)src;
   ~~~~~~~~~~~^~~~~~~~~~~~~~~~~
../src/gsiqt5/gsiDeclQIconEngine.cc:335:7: note: ‘QIconEngine_Adaptor& QIconEngine_Adaptor::operator=(const QIconEngine_Adaptor&)’ is implicitly deleted because the default definition would be ill-formed:
 class QIconEngine_Adaptor : public QIconEngine, public qt_gsi::QtObjectBase
       ^~~~~~~~~~~~~~~~~~~
../src/gsiqt5/gsiDeclQIconEngine.cc:335:7: error: ‘QIconEngine& QIconEngine::operator=(const QIconEngine&)’ is private within this context
In file included from /usr/include/qt5/QtGui/QIconEngine:1:0,
                 from ../src/gsiqt5/gsiDeclQIconEngine.cc:30:
/usr/include/qt5/QtGui/qiconengine.h:96:18: note: declared private here
     QIconEngine &operator=(const QIconEngine &other) Q_DECL_EQ_DELETE;
                  ^~~~~~~~
../src/gsiqt5/gsiDeclQIconEngine.cc:335:7: error: use of deleted function ‘QIconEngine& QIconEngine::operator=(const QIconEngine&)’
 class QIconEngine_Adaptor : public QIconEngine, public qt_gsi::QtObjectBase
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/qt5/QtGui/QIconEngine:1:0,
                 from ../src/gsiqt5/gsiDeclQIconEngine.cc:30:
/usr/include/qt5/QtGui/qiconengine.h:96:18: note: declared here
     QIconEngine &operator=(const QIconEngine &other) Q_DECL_EQ_DELETE;
                  ^~~~~~~~
make[1]: *** [Makefile:13731: gsiDeclQIconEngine.o] Error 1
```

akhuettel, 2017-12-27:
I suspect this broke when switching from Qt 5.7 to Qt 5.9 here. (Can't really test that though since a Qt downgrade on my machine would be a major pain.)

I also suspect the following commit in qtgui is the cause:
https://github.com/qt/qtbase/commit/d8b678661b962c45e6aeb3e61856b823bcfee272

klayoutmatthias, 2017-12-27:
Hi Andreas,

There was a ticket for this issue already (#34). On the master, this issue is fixed. If you want to apply this patch, please use https://github.com/klayoutmatthias/klayout/commit/11cb5eb99836e043d71b556af010b29624c3f569

I want to release the master as 0.25.1 as soon as the MacOS build is functional.

BR,

Matthias

akhuettel, 2017-12-27:
Thanks a lot Matthias; sorry I should've looked more thoroughly. I added 11cb5eb to klayout-0.25; now I ended up with a different error (see below). 

I'll try to build git master overnight and see what happens.

```
x86_64-pc-linux-gnu-g++ -c -march=native -O2 -pipe -g -fvisibility=hidden -Wall -W -pedantic -Woverloaded-virtual -Wsign-promo -Wsynth -Wno-deprecated -Wno-long-long -Wno-strict-aliasing -Wno-deprecated-declarations -D_REENTRANT -fPIC -DHAVE_QTBINDINGS -DHAVE_PYTHON -DHAVE_RUBY -DHAVE_RUBY_VERSION_CODE=20200 -DKLAYOUT_VERSION=0.25 -DKLAYOUT_VERSION_REV= -DKLAYOUT_VERSION_DATE=2017-12-27 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=25 -DKLAYOUT_TINY_VERSION=0 -DMAKE_GSIQT_LIBRARY -DQT_NO_DEBUG -DQT_DESIGNER_LIB -DQT_UIPLUGIN_LIB -DQT_PRINTSUPPORT_LIB -DQT_MULTIMEDIAWIDGETS_LIB -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_XMLPATTERNS_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_SQL_LIB -DQT_CORE_LIB -I../src/gsiqt -I/chroot/largetmp/portage/sci-electronics/klayout-0.25/work/ruby22/gsiqt -I../src/tl/tl -I../src/gsi/gsi -I../src/db/db -I../src/gsiqt -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtDesigner -isystem /usr/include/qt5/QtUiPlugin -isystem /usr/include/qt5/QtPrintSupport -isystem /usr/include/qt5/QtMultimediaWidgets -isystem /usr/include/qt5/QtSvg -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtMultimedia -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtXmlPatterns -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtXml -isystem /usr/include/qt5/QtSql -isystem /usr/include/qt5/QtCore -I/chroot/largetmp/portage/sci-electronics/klayout-0.25/work/ruby22/gsiqt -isystem /usr/include/libdrm -I/usr/lib64/qt5/mkspecs/linux-g++ -o gsiDeclQt_4.o ../src/gsiqt5/gsiDeclQt_4.cc
../src/gsiqt5/gsiDeclQt_4.cc:157:44: error: ‘WindowOkButtonHint’ is not a member of ‘Qt’
     gsi::enum_const ("WindowOkButtonHint", Qt::WindowOkButtonHint, "@brief Enum constant Qt::WindowOkButtonHint") +
                                            ^~
../src/gsiqt5/gsiDeclQt_4.cc:158:48: error: ‘WindowCancelButtonHint’ is not a member of ‘Qt’
     gsi::enum_const ("WindowCancelButtonHint", Qt::WindowCancelButtonHint, "@brief Enum constant Qt::WindowCancelButtonHint"),
                                                ^~
make[1]: *** [Makefile:31003: gsiDeclQt_4.o] Error 1
```

akhuettel, 2017-12-28:
I guess this still needs 28ff493de71cbafa38e214b75dd6874b0226813f on top of 0.25 ... testing ...

akhuettel, 2017-12-28:
With this patch (adapted from 28ff493) 
https://github.com/gentoo/gentoo/blob/master/sci-electronics/klayout/files/klayout-0.25-qt59.patch
on top of the previous one, klayout-0.25 builds fine here. :)

klayoutmatthias, 2017-12-29:
Hi Andreas, 

very good - I actually already wanted to suggest this.

Frankly, 0.25 will undergo some changes because of MacOS builds and some bug fixes. I'd recommend to wait for 0.25.1 which will contain all those patches.

Thanks,

Matthias

closed 2018-02-24 by klayoutmatthias