KLayout/klayout #51: macos-build branch: segmentation faults when closing modal dialogs

klayoutmatthias, 2018-01-02:
Upon closing modal dialogs (like the assistant), segmentation faults are observed.

The stack trace points to a place inside the event handler, specifically the QTimerEvent handling. Apparently, some things are messed up in the menu area.

Experiments show that the menu synchronisation is causing this issue. In this step, the main menu is cleared and rebuilt from scratch. Apparently this is not allows on MacOS. Probably because of the merging of the application menu with the system menu.

Further experiments show that the issue can be avoided if remove/add sequences are avoided on the menus. Such sequences happen inside the synchronization algorithm to establish a certain order of menu items. 

Just adding or just removing items seems not to cause any problems. 

So a quick solution is to disable reordering for menus. New items will always appear at the menu end initially. This happens when a new macro or package is added. Upon restart of the application, the menu is built from scratch and the proper order is established.

klayoutmatthias, 2018-01-02:
@Kazzz-S  

I am trying to fix this issue with higher prio because it's my main blocker. Did you observe this problem?

Kazzz-S, 2018-01-02:
No, so far. 
I can open and close the "assistant" several times without segmentation fault.

klayoutmatthias, 2018-01-04:
Hmmm ... Maybe that's a particular issue with Qt5.9 and/or High Sierra. Nevertheless it should be fixed. Thanks, 
Matthias

commit ae6485a0 (2018-01-02): Fixed #51: segmentation fault on return to main window and other opportunities
commit 6f2c0542 (2018-01-06): Merge pull request #18 from klayoutmatthias/macos-build

Fixed #51: segmentation fault on return to main window
closed 2018-01-04 by klayoutmatthias