KLayout/klayout #52: [BUG] MacOS+Qt5: QApplication: invalid style override passed, ignoring it.

Kazzz-S, 2018-01-02:
Style options passed to QApplication are ignored.
This worked up to version 0.24.10.

```
MacBookPro(1)$ ./klayout -v
KLayout 0.25

MacBookPro(2)$ ./klayout -e -style=motif
Warning: QApplication: invalid style override passed, ignoring it.
```
The "-e" option is accepted and KLayout starts in the editor mode.

klayoutmatthias, 2018-01-04:
Hi kazzz,

apparently "motif" is no longer supported and has been removed in Qt5. 

"windows" is available. Also "fusion" which is my favorite:

```
klayout -e style=fusion
```

The MacOS native styling looks ugly I admit. A lot of polishing will be required to blend KLayout into MacOS nicely.

Matthias

Kazzz-S, 2018-01-05:
Hi Matthias,

Thank you for clarification.
I started using Qt5 very recently with KLayout 0.25 and did not notice such a big change.
I'm going to change my scripts accordingly.

Kazzz

closed 2018-01-05 by Kazzz-S