KLayout/klayout #55: [BUG] MacOS+Qt5: Blurred image in the main canvas

Kazzz-S, 2018-01-02:
About Mac:
 1. OS: Yosemite (10.10.5)
 2. Model: Macbook Pro (13-inch, Mid 2012)
 3. Memory: 16GB
 4. Graphics: Intel HD Graphics 4000 1024MB

**KLayout 0.24.10:**
[image]

**KLayout 0.25:**  Partially transparent to the backgroud
[image]

Kazzz-S, 2018-01-03:
I've added three more different combinations as above.
It looks this is a problem in combination with Qt 5.10.0.

Qt 5.9.3 is fine.
I'm going to use Qt 5.9.3 only.

closed 2018-01-03 by Kazzz-S

Kazzz-S, 2018-01-05:
@klayoutmatthias 

Luckily, this is not a blocker.

Kazzz

thomaslima, 2018-03-16:
Dear @Kazzz-S . 

I believe this was fixed in #89 . I have no issues with qt5.10 from brew. Try building from [this commit](https://github.com/lightwave-lab/klayout/commit/fa96f6d35232e81134de467bc04b96931ab644f6#diff-9c5dcb7b173366d2e4a62fa1f450a339).

Kazzz-S, 2018-03-16:
Dear @thomaslima,

Thank you for your effort.
I'll try ;-)

Kazzz

Kazzz-S, 2018-03-18:
Hi @thomaslima,

I've merged dfd5bb1381a747f582c3dc6ab89df40e0a7aa4a4 and tried to build it on Yosemite.

```
$ brew install python3 qt  ===> Success (Qt is 5.10.1 now)

$ ./build4mac.py -p B36 -q Qt5Brew  ==> Success

$ ./build4mac.py -p B36 -q Qt5Brew -Y ==> Success
```
However ...
```
$ pwd
$HOME/GitWork/klayout-0.25/qt5.pkg.macos-Yosemite-release/klayout.app/Contents/MacOS

$ ./klayout -e
```
**ERROR: Got empty string where a real number was expected**
The same message appears in a GUI dialog on start-up and the program terminates
when [OK] button is pressed.
The default build (-p sys -r sys -q qt5macports) shows the same error.

Have you experienenced the same error?
I'm planning to repeat this on "High Sierra", too.

Regards,
Kazzz

thomaslima, 2018-03-18:
Yes I think I got that error and posted in #85. But it was fixed by Matthias. Try starting layout with a fresh .klayout

Kazzz-S, 2018-03-19:
Hi @thomaslima,

Thank you for feedback. I've overlooked the post #85. 
With a fresh ./klayout/, I could start the tool normally.

I've also confirmed that the original issue captured in this ticket was fixed as below.
KLayout 0.25.2 is working fine with Qt 5.10.x.

Qt\OS     | Yosemite |High Sierra|
-----------|-----------|--------------
5.10.0 MacPort|  OK  | OK  
5.10.1 Home Brew|  OK  |  OK  

Thank you once again for your effort.
I'm going to modify the relevant files.

Regards,
Kazzz