KLayout/klayout #57: Salt.mine GitHub feature gone

lukasc-ubc, 2018-01-04:
Hi Matthias,

Attempting to add a Package:

https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK.git/branches/v0.3.0/klayout_dot_config/grain.xml

(gives me an error: 410. Feature Gone. Sorry, but this feature has been retired. Check out the GitHub Blog for more information.)

or 
https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK.git/trunk/klayout_dot_config/grain.xml

gives me "Not Found"

I note that our Dec 13th submission URL doesn't work anymore either:
https://github.com/lukasc-ubc/SiEPIC-Tools.git/branches/development/klayout_dot_config/grain.xml

Could you please update Salt.Mine "Click here to view the package definition file" error message with updated instructions?

Thank you
Lukas

klayoutmatthias, 2018-01-04:
Hi Lukas,

the message is looking worse than it actually is. Actually you cannot access the grain.xml URL with the browser, because this is the SVN port that talks WebDAV. It's formatted as a browser URL because that's the default rendering, but in fact that URL is not meant for the browser. 

For some reason, the response is sometimes 410 and in other cases it's 404. However, the URL actually works (almost - see below).

Here is the background: KLayout uses the WebDAV protocol to access information from servers. WebDAV is spoken by SVN and the SVN port of GitHub, that's why I have chosen it as the default method.

I'm puzzled that "Click here to view the package definition isn't working". I'll need to check that. Plus some feature in KLayout was helpful to install a package from an arbitrary source for testing in the real environment- I'll consider that as a feature. 

But here are two ways to test whether your package can be installed:

### SVN method 

Try to check out this URL with Subversion: `svn co <package-url>`. If that works, it's likely that KLayout will be able to install that package too.

### curl quick test

Use curl to emulate the WebDAV access:

```
curl -H "User-Agent: SVN" -H "Depth: 1" -d '<?xml version="1.0" encoding="utf-8"?><propfind xmlns="DAV:"><prop><resourcetype xmlns="DAV:"/></prop></propfind>' -X PROPFIND https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK.git/branches/v0.2.0/klayout_dot_config
```

If this works, and gives you some like this:

```
<?xml version="1.0" encoding="utf-8"?>
<D:multistatus xmlns:D="DAV:" xmlns:ns0="DAV:">
    
      <D:response xmlns:lp1="DAV:">
        <D:href>/lukasc-ubc/SiEPIC_EBeam_PDK.git/branches/v0.2.0/klayout_dot_config/</D:href>
        
          <D:propstat>
            <D:prop>
              
                <lp1:resourcetype><D:collection/></lp1:resourcetype>
              
            </D:prop>
            <D:status>HTTP/1.1 200 OK</D:status>
          </D:propstat>
        
        
      </D:response>
...
</D:multistatus>
```

then KLayout is very likely to download the package.

### Emulator Salt.Mine

1.) Create a file with this content and save it somewhere (i.e. ~/index.xml):

```
<salt-mine>
  <salt-grain>
    <name>siepic_ebeam_pdk</name>
    <version>0.3.0</version>
    <url>https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK.git/branches/v0.3.0/klayout_dot_config</url>
  </salt-grain>
</salt-mine>
```

This is the minimum package index file for a single package. "name" must be name of your package (needs to match the one from grain.xml) and version the version of the package.

2.) Run KLayout with a sandbox and using this index file instead of the Salt.Mine server (for details see [http://www.klayout.de/doc-qt4/about/packages.html](http://www.klayout.de/doc-qt4/about/packages.html):

```
export KLAYOUT_HOME`~/.klayout-test     # use ~/.klayout-test as a sandbox (use any path you like - the path doesn't even need to exist)
export KLAYOUT_SALT_MINE=~/index.xml    # or wherever you put this file 
klayout
```

KLayout will then start with a pristine environment and in the package manager you'll only this particular package. You can now test-install it.

However, in the case of the 0.3.0 version, you'll note a few error messages:

```
ERROR: Error downloading file from 'https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK.git/!svn/bc/880/branches/v0.3.0/klayout_dot_config/tech/EBeam/pymacros/SiEPIC-EBeam-competition/ebeam_competition2018T1_TM_%3Cyour_GitHub_username%3E.gds':
Error 404: Not Found, fetching https://github.com/lukasc-ubc/SiEPIC_EBeam_PDK.git/!svn/bc/880/branches/v0.3.0/klayout_dot_config/tech/EBeam/pymacros/SiEPIC-EBeam-competition/ebeam_competition2018T1_TM_%3Cyour_GitHub_username%3E.gds
```

The package is still installed, but those files will be missing. Apparently the angular brackets are escaped somewhere they shouldn't. If you replace that by round brackets for example, it should work (I think). 

### Summary

So action items for me to add to my long list:
- Check what Salt.Mine does on "click to view package details"
- Look at the angular bracket issue
- Implement some feature to install a package from an arbitrary URL

Best regards,

Matthias

klayoutmatthias, 2018-01-04:
Hi Lukas,

I can't confirm an issue with the "click here to view package details". 

It's important that you do not include the "https://github.com" prefix in the URL. So if you enter just the rear part: "lukasc-ubc/SiEPIC_EBeam_PDK.git/branches/v0.2.0/klayout_dot_config" into the entry box, I get these package details:

[image]

Best regards,

Matthias

klayoutmatthias, 2018-01-04:
I stand corrected: the angular bracket issue *is* an issue - the package is not installed. Please try to use round bracket to work around this issue. I'll look at this.

Thanks,

Matthias

klayoutmatthias, 2018-01-04:
One more update: 

the angular bracket issue seems to be an Qt4 issue. It's gone with Qt5. But I need to further investigate that. I can't say whether there is a workaround for this.

Plus when trying to install the package, I get some macro errors (SiEPIC library is missing). I assume that's intentional because it's nowhere to be found :-)

Best regards,

Matthias

lukasc-ubc, 2018-01-05:
Hi Matthias,

I removed the < > from the filename. Apparently Windows doesn't like it.  I committed it to SiEPIC-EBeam "master"

Missing files -- we split up SiEPIC-EBeam-PDK into two packages: Tools and EBeam-PDK.  We will also have several other PDKs.  Each will depend on Tools (and KLayout).

This works:
    lukasc-ubc/SiEPIC_EBeam_PDK.git/branches/v0.3.0/klayout_dot_config

    lukasc-ubc/SiEPIC_EBeam_PDK.git/trunk/klayout_dot_config

    lukasc-ubc/SiEPIC_EBeam_PDK.git/tags/v0.3.0/klayout_dot_config

    lukasc-ubc/SiEPIC-Tools.git/tags/v0.3.0/klayout_dot_config

(I don't know why it didn't work last night!)

thank you

lukasc-ubc, 2018-01-06:
Hi Matthias,

Things are working for me.  I have both EBeam and Tools registered, and I've gone through several updates and fixes...

1) I noticed one issue with Tools.  In going from v0.3.4 to v0.3.5, KLayout package manager came up with about 10 errors, where it couldn't delete/download the DLL files.  Hence the upgrade failed. I had to manually delete the entire salt folder locally, then install it fresh.

2) Is there a way to have multiple Salt mines?  (it is briefly mentioned in the documentation)

e.g., 

   export KLAYOUT_SALT_MINE=file:///home/yourself/index.xml; http://your_default_one

I am trying to figure out how to have both public Packages and Private packages.  

Documentation suggests that I would need my own single index.xml file hosted privately, and then I can point to multiple repos, including private and public ones.

You mentioned previously that it should be possible to do authentication.  But I don't think I can add a private GitHub repo to your salt.mine (your webdav would fail without the user/password).  I haven't tried creating my own index.xml yet, so perhaps my question is premature. 

3) The export KLAYOUT_SALT_MINE - instead of having the users modifying their system environment, could that be added to the GUI and saved in the .klayout_rc file instead?  This would also allow you to have multiple entries / a history (and perhaps the user chooses one of them only).

thank you
Lukas

klayoutmatthias, 2018-01-10:
Hi Lukas,

Maybe the DLL delete failed, because the DLL's are used by the KLayout  
process. Windows will then prevent those DLL's from being deleted.  
That's actually bad. I have not considered this case. I guess the only  
solution to this is to have the package manager as an external  
application.

Regarding the repository hierarchy:

Currently you can't have multiple repositories, but you can include  
other repositories. So you can basically build a hierarchy of them.

This is a setup I tested myself:

  * On a site that is owned by you, you host a private repository.  
This is basically just a XML file. This is where you point your  
KLAYOUT_SALT_MINE environment variable to
  * In this file you include the public repository:

    <salt-mine>
      .. your packages ..
      <include>http://sami.klayout.org/repository.xml</include>
    </salt-mine>

Best regards,

Matthias

lukasc-ubc, 2018-01-12:
Thanks Matthias.

It is working, fantastic!  it asks for a username/password for GitHub as expected, and installed the private repository.

DLL issue: my fix was to move those files to another repository. then quit KLayout and manually delete the salt folders. then install the two separate repositories.

Having the package manager as a separate application doesn't sound very appealing.  I really like the way you have it in the menu.  

Instead of deleting the DLLs, can you move them to the trash folder?  or some other temporary folder?  Then delete them next time when they are not in use.

closed 2018-02-25 by lukasc-ubc