KLayout/klayout #67: OSX feature request

lukasc-ubc, 2018-01-20:
Perhaps not a high priority but here are a few improvement suggestions for OSX:

1) MacBook touch bar support: https://developer.apple.com/macos/touch-bar/

2) Add support for Retina displays.  Retinizer app works well for fonts, but not for the icons and the layout itself.

3) when in Macro Development, the menus at the top stay the same and are attached to the layout window. Hotkeys such as Ctrl-W (close window) should act on the open file in the Macro Development. Instead, Ctrl-W closes the layout.  Command-F correctly goes to "Find text ...".  Some way of closing the Macro window would be nice

lukasc-ubc, 2018-01-21:
* From Finder.App, it should be possible to open a GDS in KLayout, e.g., right-click, Open With -> KLayout.  Or double-click on the GDS, KLayout should be able to open it.

thomaslima, 2018-03-07:
Last one is implemented in #87

lukasc-ubc, 2018-03-07:
WOW!  The lack of this functionality has been killing me for many many years!  I can’t believe you’ve done it!  Thank you soooo much.

I tested your build and it works fantastic!

I added “Py3” to the “SiEPIC 0.3.39” menu item to make it obvious to the user that we are running Python 3 in the Mac OS version….

thomaslima, 2018-03-07:
They have pretty sophisticated drag-and-drop functionality. I think you might be able to drag .lyp, .gds, and other files to klayout and it will do what it’s expected. But indeed open with was missing. Did the default file association work for you out of the box?

Thomas Ferreira de Lima
tlima@princeton.edu<mailto:tlima@princeton.edu>

lukasc-ubc, 2018-03-08:
@thomaslima 
Yes, default worked out of the box.  
* open DMG file
* copy to Applications
* double-click on a gds file in Finder.App
and that's it!

thomaslima, 2018-03-16:
I would like to reference #94 here (addressing number 2). I think we should have an issue per feature request. I think (3) is pretty easy to fix, (2) is in the horizon, and (1) needs more thought. I personally do not own a mac with touch bar.

klayoutmatthias, 2018-06-10:
Is this problem still active? I'd like to close this issue for cleaner bookkeeping.

closed 2018-06-10 by klayoutmatthias

thomaslima, 2018-06-11:
I think 1 is out of consideration, 2 was done and 3 was not done but should be a small fix.