KLayout/klayout #70: PYA:　*** ERROR: 'Polygon' object has no attribute 'shapes'

Kazzz-S, 2018-01-30:
I'm trying to filter/catch relatively large number of small polygons (meshes/fishes)
with relatively small number of large polygons (filters/cast nets).

The attached PYA script works fine except for the code block [7]-(E); Line 217 - 231.
The "Assistant" document shows Polygon::shapes() but I got a runtime error:
  *** ERROR: 'Polygon' object has no attribute 'shapes'

Is there any different way to do the same work?

Warm regards,
Kaｚｚｚ

[kmeshFilter-xxx.zip](https://github.com/klayoutmatthias/klayout/files/1676818/kmeshFilter-xxx.zip)

klayoutmatthias, 2018-01-31:
Hi kazzz,

I guess you mean "Shape#shapes", do you?

From your code I understand that you want to select shapes based on a region filter and later remove those shapes from the original.

The Region class is suited for extracting, but for manipulating the original shapes, you'll need a different approach. Please think of the Region as some copy of the original database shapes. You can manipulate the region, but this won't have an effect on the original shapes. For this you'll need to know the original "Shape" objects - and the Region does not provide this information.

There is a general issue here: the original shapes can be embedded in some hierarchy. If that is the case, selecting shapes by region may only select certain instances of the shape and if you remove the original one, you will also remove the ones in other instances of their cell. This may not have the desired effect. So manipulation of a hierarchical layout in general is a difficult task.

One solution to this issue is flattening. If you can afford this, a solution might be to use the set difference method of the region, i.e.
```
all = pya.Region(topCellMesh.begin_shapes_rec(layMesh))
remaining = all.not_members_of(regOverlapping)
topCellMesh.layout.clear(layMesh)
topCellMesh.shapes(layMesh).insert(remaining)
```
This approach has the side effect of flattening the "layMesh" layer and turning all shapes into polygons. Performance is probably poor for large layouts, so I'd use this on small layouts only.

If you don't care about the hierarchy effect and your filter region is of Manhattan kind, a different approach may work too:
```
sitMesh = pya.RecursiveShapeIterator( layoutMesh, topCellMesh, layMesh, regFilter, true )
# copy the selected shapes (uses the Region to quickly iterate)
topCellOutput.shapes(layOutput).insert(pya.Region(sitMesh))
# and remove the shapes (not "is_valid" which accounts for the same shape being selected multiple times in the hiearchy)
while not sitMesh.at_end():
    shape = sitMesh.shape()
    if shape.is_valid():
        shape.shapes().erase(shape)
    sitMesh.next
```

This approach exploits the ability of the RecursiveShapeIterator to confine iteration to region and will directly delete the shapes without flattening but inducing the mentioned hierarchical effect.
   
Best regards,

Matthias

Kazzz-S, 2018-02-01:
Hi Matthias,

> I guess you mean "Shape#shapes", do you? :-)

Yes, it's my silly mistake ;-(

Thank you very much for clarification on "Region."
Fortunately, the flattening approach is suitable for my task.
In fact, there is a single top cell in the mesh design.
The modified code works perfectly.
[kmeshFilter-yyy.zip](https://github.com/klayoutmatthias/klayout/files/1684541/kmeshFilter-yyy.zip)

Once again, thank you for your help.
Warm regards,
Kazzzz

This will close #70.

closed 2018-02-01 by Kazzz-S