KLayout/klayout #76: KL Layout while  importing Gerber crashes KL layout

steradianradar, 2018-02-11:
While importing Gerber to KL Layout I get the following error - crashes with the following error report. 

Apu Sivadas

Exception code: 0xc0000005
Program Version: KLayout 0.25 (2017-11-12 r20a6c5e) AMD64

Backtrace:
0x2ff25a0 - (klayout_laybasic.dll) +861600
0x6b21ced2 - (klayout_ext.dll) +1167058
0x303bc8b - (klayout_laybasic.dll) +1162379
0x668e82bb - (Qt5Core.dll) +1737403
0x5364296 - (Qt5Widgets.dll) +17046
0x54cc61d - (Qt5Widgets.dll) +1492509
0x54d2894 - (Qt5Widgets.dll) +1517716
0x54d685e - (Qt5Widgets.dll) +1534046
0x53ae0e8 - (Qt5Widgets.dll) +319720
0x54d721b - (Qt5Widgets.dll) +1536539
0x536a3bc - (Qt5Widgets.dll) +41916
0x53705c3 - (Qt5Widgets.dll) +67011
0x6a54704b - (klayout_lay.dll) +28747
0x668c7812 - (Qt5Core.dll) +1603602
0x536eca9 - (Qt5Widgets.dll) +60585
0x53c4cb0 - (Qt5Widgets.dll) +412848
0x53c6edb - (Qt5Widgets.dll) +421595
0x536a3bc - (Qt5Widgets.dll) +41916
0x536f746 - (Qt5Widgets.dll) +63302
0x6a54704b - (klayout_lay.dll) +28747
0x668c7812 - (Qt5Core.dll) +1603602
0x4f164fc - (Qt5Gui.dll) +156924
0x4f17c25 - (Qt5Gui.dll) +162853
0x4efcdde - (Qt5Gui.dll) +52702
0x6690effa - (Qt5Core.dll) +1896442
0x7ff9b159b85d - (USER32.dll) +47197
0x7ff9b159b1ef - (USER32.dll) +45551
0x6690e683 - (Qt5Core.dll) +1894019
0x6dd32cd5 - (qwindows.dll) +470229
0x668c6a5d - (Qt5Core.dll) +1600093
0x668cd827 - (Qt5Core.dll) +1628199
0x6a54ea2d - (klayout_lay.dll) +59949
0x6a54fa78 - (klayout_lay.dll) +64120
0x401f41 - (klayout_app.exe) +8001
0x1916f8 - (klayout_rba.dll) +5880
0x67ae6cc6 - (x64-msvcrt-ruby240.dll) +1469638
0x67af43e0 - (x64-msvcrt-ruby240.dll) +1524704
0x67aed16c - (x64-msvcrt-ruby240.dll) +1495404
0x67af23f4 - (x64-msvcrt-ruby240.dll) +1516532
0x679dff4e - (x64-msvcrt-ruby240.dll) +393038
0x679e2b2f - (x64-msvcrt-ruby240.dll) +404271
0x1982c5 - (klayout_rba.dll) +33477
0x401a7c - (klayout_app.exe) +6780
0x4013f8 - (klayout_app.exe) +5112
0x4014eb - (klayout_app.exe) +5355
0x7ff9b2321fe4 - (KERNEL32.DLL) +73700
0x7ff9b25cefb1 - (ntdll.dll) +454577

klayoutmatthias, 2018-02-11:
Hi,

thanks for the report, but unfortunately it's not telling much. Do you have a sample you can provide to reproduce the issue? If you don't want to paste it here, you can send test data to the mail address listed on the "Contacts" page on [http://www.klayout.org](http://www.klayout.org).

Kind regards,

Matthias

klayoutmatthias, 2018-06-10:
Any news on this?

steradianradar, 2018-06-11:
Matthias,

Sorry for my delay - I missed following up. Your dedication is moving. I am
attaching the .gbr file I have tested for the bug.

Plus all versions and error log.

Apu

-- 
*Steradian Semiconductors Private Limited*

Exception code: 0xc0000005
Program Version: KLayout 0.25 (2017-11-12 r20a6c5e) AMD64

Backtrace:
0xec25a0 - (klayout_laybasic.dll) +861600
0x6b21ced2 - (klayout_ext.dll) +1167058
0xf0bc8b - (klayout_laybasic.dll) +1162379
0x668e82bb - (Qt5Core.dll) +1737403
0x12f4296 - (Qt5Widgets.dll) +17046
0x145c61d - (Qt5Widgets.dll) +1492509
0x1462894 - (Qt5Widgets.dll) +1517716
0x146685e - (Qt5Widgets.dll) +1534046
0x133e0e8 - (Qt5Widgets.dll) +319720
0x146721b - (Qt5Widgets.dll) +1536539
0x12fa3bc - (Qt5Widgets.dll) +41916
0x13005c3 - (Qt5Widgets.dll) +67011
0x6a54704b - (klayout_lay.dll) +28747
0x668c7812 - (Qt5Core.dll) +1603602
0x12feca9 - (Qt5Widgets.dll) +60585
0x1354cb0 - (Qt5Widgets.dll) +412848
0x1356edb - (Qt5Widgets.dll) +421595
0x12fa3bc - (Qt5Widgets.dll) +41916
0x12ff746 - (Qt5Widgets.dll) +63302
0x6a54704b - (klayout_lay.dll) +28747
0x668c7812 - (Qt5Core.dll) +1603602
0x23c64fc - (Qt5Gui.dll) +156924
0x23c7c25 - (Qt5Gui.dll) +162853
0x23acdde - (Qt5Gui.dll) +52702
0x6690effa - (Qt5Core.dll) +1896442
0x7fff58276cc1 - (USER32.dll) +93377
0x7fff58276693 - (USER32.dll) +91795
0x6690e683 - (Qt5Core.dll) +1894019
0x6dd32cd5 - (qwindows.dll) +470229
0x668c6a5d - (Qt5Core.dll) +1600093
0x668cd827 - (Qt5Core.dll) +1628199
0x6a54ea2d - (klayout_lay.dll) +59949
0x6a54fa78 - (klayout_lay.dll) +64120
0x401f41 - (klayout_app.exe) +8001
0x1916f8 - (klayout_rba.dll) +5880
0x67ae6cc6 - (x64-msvcrt-ruby240.dll) +1469638
0x67af43e0 - (x64-msvcrt-ruby240.dll) +1524704
0x67aed16c - (x64-msvcrt-ruby240.dll) +1495404
0x67af23f4 - (x64-msvcrt-ruby240.dll) +1516532
0x679dff4e - (x64-msvcrt-ruby240.dll) +393038
0x679e2b2f - (x64-msvcrt-ruby240.dll) +404271
0x1982c5 - (klayout_rba.dll) +33477
0x401a7c - (klayout_app.exe) +6780
0x4013f8 - (klayout_app.exe) +5112
0x4014eb - (klayout_app.exe) +5355
0x7fff56b33034 - (KERNEL32.DLL) +77876
0x7fff58761551 - (ntdll.dll) +464209

klayoutmatthias, 2018-06-12:
Thanks a lot! 

But I don't see an attachment. Did I miss anything?

Thanks,

Matthias

steradianradar, 2018-06-13:
Matthias,

There was attachment - seems like this email address can not receive them
or kills it. Is there a proper email I can send to or a site I can upload
to.

I am not familiar with github.

Apu

thomaslima, 2018-06-15:
@steradianradar you can put the attachment via the GitHub website. Click this link: https://github.com/klayoutmatthias/klayout/issues/76

steradianradar, 2018-06-16:
I have uploaded it at

https://github.com/steradianradar/klayout

Apu

klayoutmatthias, 2018-06-17:
Please don't fork a project to upload files ... just drag and drop them to the window where you write your comments. 

For reference I have attached your files here:

[ststest1p2-F.Fab.gbr.zip](https://github.com/klayoutmatthias/klayout/files/2109367/ststest1p2-F.Fab.gbr.zip)

[error_import.txt](https://github.com/klayoutmatthias/klayout/files/2109365/error_import.txt)

[image]

[image]

[image]

klayoutmatthias, 2018-06-17:
Thanks for the test data.

I was able to read the files, but I cannot reproduce the issue. It loads well on my installation, but that's Linux. 

Maybe you can upgrade to 0.25.3 which is the latest version. I'll continue to try on Windows too.

Thanks,

Matthias

steradianradar, 2018-06-18:
Matthias ,

I tried after upgrading to 0.25.3. The problem still exist.

Apu

Kazzz-S, 2018-06-18:
### I've tested using 0.25.3 (the same as the binary package)
#### OS is still Yosemite (10.10.5)!
[image]
[image]
----
### Using the standard wizard ...
[image]
[image]
[image]
[image]
[image]
----
### Using "import.pcb" provided by Matthias...
[image]

Kazzz

Kazzz-S, 2018-06-18:
### 32-bit Windows  version (on Windows-7) works, too.
[image]

Kazzz

steradianradar, 2018-06-19:
Thank you for the update. I also found it to be working in my home computer
- seems related to Qt and my existing python installation.

Apu

klayoutmatthias, 2020-02-22:
I'll close this ticket as it seems not to be related to a code issue.

closed 2020-02-22 by klayoutmatthias