KLayout/klayout #81: Include numpy in Python version packaged with Windows

klayoutmatthias, 2018-02-24:


thomaslima, 2018-03-07:
Same for MacOS. I couldn't figure that out yet.

thomaslima, 2018-03-20:
Figured it out for macOS. my last .dmg was only 100mb. #100 

This is a big deal for salt packages.

klayoutmatthias, 2018-05-05:
numpy is included in 0.25.3

closed 2018-05-05 by klayoutmatthias

thomaslima, 2018-05-05:
Hi Matthias, so does that mean that klayout ships with its own python virtualenv in the windows build?

klayoutmatthias, 2018-05-06:
Hi Thomas,

That's true if "virtualenv" is just another word for "take what you get" :-)

I just included numpy in the installer. So it's available, although I  
don't know whether the version I picked isn't already outdated. Sadly,  
the MSYS2 platform I'm using is making rolling releases and being  
happy with the one I had, I did not follow their release cycles. Maybe  
I'll setup my CI to update the installation regularly. That should  
make KLayout follow MSYS2.

Frankly, I'd like to have KLayout available as MSYS2 package - then  
people could just install that themselves and use all those nice  
Python packages from MSYS2. But I have not made an attempt yet to  
initiate this.

Until then, building KLayout for MSYS2 isn't a big deal as MSYS2 is  
pretty much Linux-like. It just takes a while.

Best regards,

Matthias

thomaslima, 2018-05-07:
I am asking because I included python's framework in the mac build I've been providing. That means that people can install packages via pip. that is why I did not submit a pull request #100  until the other OS had the same functionality. But actually if you think it's okay I can do it.