KLayout/klayout #83: salt Package Manager - doesn't remove files

lukasc-ubc, 2018-02-26:
Hi @klayoutmatthias,

I notice that the Package Manager installer doesn't remove files that are no longer in the package, but rather. I am trying to solve the "Stream has unknown format" issue in my technologies, which I believe I have done on my repositories, but the user needs to manually delete all the files in their salt folder and re-install from scratch.

klayoutmatthias, 2018-02-26:
Hi Lukas,

I am wondering what comes after "but rather." :-)

Basically that's true - files just get added, not deleted. I can add  
this as a feature request, but I'm a bit afraid of users adding files  
dynamically.

I think that entirely removing the package in the package manager  
should help. Doesn't it?

BTW: the "stream has unknown format" isn't serious. It's just a bit  
annoying if you care for messages. I could basically supress it or  
just issue this warning on a higher verbosity level.

Best regards,

Matthias

lukasc-ubc, 2018-02-26:
> But rather add them only, as you gathered. 

As a feature request it would be good. Seems to me that an upgrade should have some built in diff functionality to remove deleted files. Also a diff would speed up the upgrade. 

I agree about the warning. It’s not an issue for me other than having 20 people asking over and over.