KLayout/klayout #84: Macro Development IDE in Windows, breakpoints not working

lukasc-ubc, 2018-02-27:
On MacOS, I can have breakpoints within modules and debug code directly (e.g., import SiEPIC).

On Windows it doesn't seem to work.

Is there a reason why the debugger in MacOS works better than in Windows?  Could this functionality be added to the Windows version?  There are some things that I'd like to debug in Windows (Windows-specific problems) and it doesn't seem very straightforward for code that we put in modules.

Thank you
Lukas

klayoutmatthias, 2018-02-28:
Hi Lukas,

I don't see a reason that shouldn't work on Windows except the Python version. I'll check this. Thanks for reporting this issue.

Best regards,

Matthias

closed 2018-03-02 by klayoutmatthias

klayoutmatthias, 2018-03-02:
The issue was caused by the "/" and "\" ambiguity of Windows paths. Module files get "\" path components while KLayout uses "/". The solution is path normalization.

commit 3d5fe871 (2018-03-02): Fixed #84 (Python modules: breakpoints not working on Windows)