KLayout/klayout #85: disable python "Caught the following exception:" window in Macro Development

lukasc-ubc, 2018-03-01:
Hi @klayoutmatthias,

This question is regarding handling try/except cases in python modules loaded in KLayout.

On MacOS (I haven't checked on Windows), when a module is within the KLayout python folder, e.g., urllib3 in ~/.klayout/python/urllib3, and when you run a macro that uses that module, with the Macro Development window open, each try/except creates exceptions which pop up a GUI:

> Caught the following exception:
> ....
> Press 'Ok' to continue and 'Cancel' to stop in the debugger

However, if that module is installed in the system, e.g., pip install urllib3, then the same exceptions don't trigger the GUI.  

And with the Macro Development window closed, you don't see these message and everything works fine.

What this means is that for modules that I develop and put in the KLayout path, I need to minimize the use of try/except, otherwise it makes debugging difficult (if you have to click "OK" 20 times before you get to the actual error you want to debug).

Is there any way to selectively disable the try/except dialogs for specific modules, or perhaps placing them in folders (salt/...) so they don't have these messages?

thomaslima, 2018-03-08:
Seems intentional...

Found the behavior in https://github.com/klayoutmatthias/klayout/blob/master/src/lay/lay/layMacroEditorDialog.cc#L2804-L2816

Note: it seems you can disable it in a macro debug setup dialog cf. https://github.com/klayoutmatthias/klayout/blob/master/src/lay/lay/layMacroEditorDialog.cc#L1975

lukasc-ubc, 2018-03-08:
Thanks Thomas.  

Interesting that it was designed this way, where other Python environments allow the code itself to deal with try/except cases, and don’t pop up GUIs.

It doesn’t look like the “config_set” method is exposed to Python, so perhaps we can't turn it off there?

> dir(pya.Application.instance().main_window().instance())

['AllowNestedDocks', 'AllowTabbedDocks', 'AnimatedDocks', 'DockOption', 'DrawChildren', 'DrawWindowBackground', 'ForceTabbedDocks', 'IgnoreMask', 'QFlags_DockOption', 'QFlags_RenderFlag', 'RenderFlag', 'VerticalTabs', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_class_connect', '_class_disconnect', '_create', '_destroy', '_destroyed', '_inst_connect', '_inst_disconnect', '_is_const_object', '_manage', '_unmanage', 'acceptDrops', 'accessibleDescription', 'accessibleName', 'actions', 'activateWindow', 'addAction', 'addActions', 'addDockWidget', 'addToolBar', 'addToolBarBreak', 'adjustSize', 'animated', 'asConstQObject', 'asConstQPaintDevice', 'asQObject', 'asQPaintDevice', 'autoFillBackground', 'backgroundRole', 'backingStore', 'baseSize', 'blockSignals', 'cancel', 'centralWidget', 'childAt', 'children', 'childrenRect', 'childrenRegion', 'clearFocus', 'clearMask', 'clone_current_view', 'close', 'close_all', 'close_current_view', 'cm_adjust_origin', 'cm_bookmark_view', 'cm_cancel', 'cm_cell_copy', 'cm_cell_cut', 'cm_cell_delete', 'cm_cell_flatten', 'cm_cell_hide', 'cm_cell_paste', 'cm_cell_rename', 'cm_cell_select', 'cm_cell_show', 'cm_cell_show_all', 'cm_clear_layer', 'cm_clone', 'cm_close', 'cm_close_all', 'cm_copy', 'cm_copy_layer', 'cm_cut', 'cm_dec_max_hier', 'cm_delete', 'cm_delete_layer', 'cm_edit_layer', 'cm_exit', 'cm_goto_position', 'cm_help_about', 'cm_inc_max_hier', 'cm_last_display_state', 'cm_layout_props', 'cm_load_bookmarks', 'cm_load_layer_props', 'cm_lv_add_missing', 'cm_lv_delete', 'cm_lv_expand_all', 'cm_lv_group', 'cm_lv_hide', 'cm_lv_hide_all', 'cm_lv_insert', 'cm_lv_new_tab', 'cm_lv_regroup_by_datatype', 'cm_lv_regroup_by_index', 'cm_lv_regroup_by_layer', 'cm_lv_regroup_flatten', 'cm_lv_remove_tab', 'cm_lv_remove_unused', 'cm_lv_rename', 'cm_lv_rename_tab', 'cm_lv_select_all', 'cm_lv_show', 'cm_lv_show_all', 'cm_lv_show_only', 'cm_lv_sort_by_dli', 'cm_lv_sort_by_idl', 'cm_lv_sort_by_ild', 'cm_lv_sort_by_ldi', 'cm_lv_sort_by_name', 'cm_lv_source', 'cm_lv_ungroup', 'cm_macro_editor', 'cm_manage_bookmarks', 'cm_max_hier', 'cm_max_hier_0', 'cm_max_hier_1', 'cm_navigator_close', 'cm_navigator_freeze', 'cm_new_cell', 'cm_new_layer', 'cm_new_layout', 'cm_new_panel', 'cm_next_display_state', 'cm_open', 'cm_open_current_cell', 'cm_open_new_view', 'cm_open_too', 'cm_packages', 'cm_pan_down', 'cm_pan_left', 'cm_pan_right', 'cm_pan_up', 'cm_paste', 'cm_prev_display_state', 'cm_print', 'cm_pull_in', 'cm_reader_options', 'cm_redo', 'cm_redraw', 'cm_reload', 'cm_reset_window_state', 'cm_restore_session', 'cm_save', 'cm_save_all', 'cm_save_as', 'cm_save_bookmarks', 'cm_save_current_cell_as', 'cm_save_layer_props', 'cm_save_session', 'cm_screenshot', 'cm_sel_flip_x', 'cm_sel_flip_y', 'cm_sel_free_rot', 'cm_sel_move', 'cm_sel_move_to', 'cm_sel_rot_ccw', 'cm_sel_rot_cw', 'cm_sel_scale', 'cm_select_all', 'cm_select_cell', 'cm_select_current_cell', 'cm_setup', 'cm_show_properties', 'cm_technologies', 'cm_undo', 'cm_unselect_all', 'cm_view_log', 'cm_zoom_fit', 'cm_zoom_fit_sel', 'cm_zoom_in', 'cm_zoom_out', 'connect', 'contentsMargins', 'contentsRect', 'contextMenuPolicy', 'corner', 'create', 'createPopupMenu', 'createWinId', 'createWindowContainer', 'create_layout', 'create_view', 'current_view', 'current_view_index', 'cursor', 'customContextMenuRequested', 'customContextMenuRequested_', 'deleteLater', 'destroy', 'destroyed', 'destroyed_', 'disconnect', 'dockNestingEnabled', 'dockOptions', 'dockWidgetArea', 'documentMode', 'dumpObjectInfo', 'dumpObjectTree', 'dynamicPropertyNames', 'effectiveWinId', 'enable_edits', 'enabled', 'ensurePolished', 'event', 'eventFilter', 'exit', 'find', 'findChild', 'focus', 'focusPolicy', 'focusProxy', 'focusWidget', 'font', 'fontInfo', 'fontMetrics', 'foregroundRole', 'frameGeometry', 'frameSize', 'fullScreen', 'geometry', 'getContentsMargins', 'grab', 'grabGesture', 'grabKeyboard', 'grabMouse', 'grabShortcut', 'graphicsEffect', 'graphicsProxyWidget', 'grid_micron', 'hasFocus', 'hasHeightForWidth', 'hasMouseTracking', 'height', 'heightForWidth', 'hidden', 'hide', 'iconSize', 'iconSizeChanged', 'iconSizeChanged_', 'index_of', 'inherits', 'initial_technology', 'inputMethodHints', 'inputMethodQuery', 'insertAction', 'insertActions', 'insertToolBar', 'insertToolBarBreak', 'installEventFilter', 'instance', 'internalWinId', 'isActiveWindow', 'isAncestorOf', 'isAnimated', 'isDockNestingEnabled', 'isEnabled', 'isEnabledTo', 'isEnabledToTLW', 'isFullScreen', 'isHidden', 'isLeftToRight', 'isMaximized', 'isMinimized', 'isModal', 'isRightToLeft', 'isSeparator', 'isTopLevel', 'isVisible', 'isVisibleTo', 'isWidgetType', 'isWindow', 'isWindowModified', 'isWindowType', 'is_const_object', 'keyboardGrabber', 'killTimer', 'layout', 'layoutDirection', 'load_layout', 'locale', 'lower', 'manager', 'mapFrom', 'mapFromGlobal', 'mapFromParent', 'mapTo [... cut ...] tDirection', 'unsetLocale', 'update', 'updateGeometry', 'updatesEnabled', 'view', 'views', 'visible', 'visibleRegion', 'whatsThis', 'width', 'winId', 'window', 'windowFilePath', 'windowFlags', 'windowHandle', 'windowIcon', 'windowIconChanged', 'windowIconChanged_', 'windowIconText', 'windowIconTextChanged', 'windowIconTextChanged_', 'windowModality', 'windowModified', 'windowOpacity', 'windowRole', 'windowState', 'windowTitle', 'windowTitleChanged', 'windowTitleChanged_', 'windowType', 'x', 'y']

klayoutmatthias, 2018-03-11:
Hi all,

you can disable the "stop on exception" feature with this code:

```
pya.Application.instance().set_config("macro-editor-stop-on-exception", "false")
```

But this flag is evaluated only when the macro IDE is started - hence it's not going to become efficient until you close and re-open the IDE.

I have tried to implement a feature that allows "teaching" which files are to be ignore: the "stop on exceptions" button now has a "ignore" button. If this is used, the corresponding file is not longer considered. The list of ignored files can be cleared in the setup dialog. In addition, the configuration mentioned above will be dynamic. The commit is https://github.com/klayoutmatthias/klayout/commit/5fbc20702e839c6721ca90bc3633d070371f28bd

Matthias

thomaslima, 2018-03-12:
Hi Mattias. I am getting this bug while trying to launch the application.
[image]

klayoutmatthias, 2018-03-12:
Sorry, my fault. Should be fixed with https://github.com/klayoutmatthias/klayout/commit/3ae71d7e72dd3929d07d6880e037445fa5275bba.

thomaslima, 2018-03-12:
Thanks. I thought I had caught that and it persisted. I will try again. Does it work fine for you with a fresh `~/.klayout`?

thomaslima, 2018-03-12:
Hi @klayoutmatthias Sorry, it is still not fixed here. I rebuilt the entire `lay` module, and it still persisted. Any way I can get the stack trace so I can show you?

klayoutmatthias, 2018-03-13:
Strange ... I tested with a new klayoutrc. Maybe your configuration file got corrupted. Could you try to delete ~/.klayout/klayoutrc?
Thanks,
Matthias

Von Samsung-Tablet gesendet

thomaslima, 2018-03-13:
Ok that fixed the initial error. But now when I open File->Properties, I get an application crash handled by klayout.

[image]

backtrace:
```
Signal number: 11
Address: 0x0
Program Version: KLayout 0.25.1 (2018-03-13 r6240f14)

Backtrace:
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x2132ab _ZN3lay19SaltDownloadManager7executeEPNS_17SaltManagerDialogERNS_4SaltE
/usr/lib/system/libsystem_platform.dylib +0x1f5a _sigtramp
/usr/lib/system/libsystem_c.dylib +0x53e50 notify_check_tz
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x7dc12 _ZN3lay12CrashMessage14cancel_pressedEv
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0xf4950 _ZN3lay14AlertLogButton17attention_changedEb
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0xf4502 _ZN3lay14AlertLogButton17attention_changedEb
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x21e5b8 _ZN3lay13salt_mine_urlEv
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x21d70d _ZN3lay13salt_mine_urlEv
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x1ada1e _ZN3lay12BookmarkList3addERKNS_19BookmarkListElementE
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x21f36b _ZN11QMetaObject8activateEP7QObjectiiPPv
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x67f5 _ZN7QAction8activateENS_11ActionEventE
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x217a31 _ZN7QObject5eventEP6QEvent
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0xfffd _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x1139a _ZN12QApplication6notifyEP7QObjectP6QEvent
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x5668b _ZN3lay14GuiApplication6notifyEP7QObjectP6QEvent
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1ee798 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1ef938 _ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x2a227 qt_plugin_instance
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x2aaf1 qt_plugin_instance
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0xa3721 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x15d0ac __CFRunLoopDoSource0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x86260 __CFRunLoopDoSources0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x856dd __CFRunLoopRun
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x84f43 CFRunLoopRunSpecific
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2fe26 RunCurrentEventLoopInMode
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2fb96 ReceiveNextEventCommon
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2f914 _BlockUntilNextEventMatchingListInModeWithFilter
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x41f5f _DPSNextEvent
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x7d7b4c -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x36d6d -[NSApplication run]
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x298dd qt_plugin_instance
[... cut ...]
/System/Library/Frameworks/Ruby.framework/Versions/2.3/Ruby +0x35061 ruby_run_node
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/qt5.pkg.macos-HighSierra-release/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x226e3 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout-qt10/./qt5.pkg.macos-HighSierra-release/klayout.app/Contents/MacOS/klayout +0xa94d
/usr/lib/system/libdyld.dylib +0x1115 start
/usr/lib/system/libdyld.dylib +0xffff80009eae1001 start
```

klayoutmatthias, 2018-03-14:
Hi Thomas,

that's strange .. the fix should be unrelated to anything else. 

What "Properties" do you mean? I tried various ways to reproduce the issue but failed. I removed ~/.klayout and loaded a file. File/Layout Properties works as well as Edit/Properties, both in viewer or editor mode. 

Any hints?

Thanks,

Matthias

thomaslima, 2018-03-14:
Sorry. I meant Preferences. klayout -> Preferences. Or same bug trying to open the Macro Editor settings button.

thomaslima, 2018-03-14:
To reproduce: open klayout without anything loaded. then open the macro development editor. then click the cogwheel button to open the setup page. crash.

klayoutmatthias, 2018-03-14:
Hmm ... thanks for the clarification. I reason that this issue is related to the change, but I tried hard and still could not reproduce it. I checked various potential missing initializers but without success. Even faithful valgrind did not report any particular issues here. If memory corruption causes non-reproducible behaviour, valgrind is usually able to find that.

Also the stack trace is weird. The symbols seem to be messed up. Their sequence simply does not make sense. Either the stack trace symbols are wrong on MacOS or the symbol information inside the dylibs is bad. 

Maybe it's worth trying a full rebuild.

Thanks and best regards,

Matthias

thomaslima, 2018-03-14:
I'm sorry this is hard to reproduce. I have done that already. Didn't seem to go away. All I have confirmed is that commit 5fbc207 is the offending commit. I have reverted that and the program runs fine. I've tried qt 5.9.4 and qt 5.10, same result.
I agree that the backtrace is weird. Also note that the addr2line command does not exist on Mac. I've tried installing it from binutils but that doesn't help at all. 

Seems like I have to debug this one my own. Do you have tips for improving that backtrace somehow? Printing things to terminal? That would help.
Feel free to close this issue and I will open another one. Or we can just keep this one open.

thomaslima, 2018-03-14:
warnings while building on the `layMacroEditorDialog.cc` file. I always ignore those... but maybe?
```
In file included from ../../../src/lay/lay/layMacroEditorDialog.cc:26:
In file included from ../../../src/lay/lay/layMacroController.h:28:
In file included from ../../../src/laybasic/laybasic/layPlugin.h:32:
../../../src/tl/tl/tlClassRegistry.h:223:12: warning: instantiation of variable 'tl::Registrar<lym::MacroInterpreter>::instance' required here, but no definition is available [-Wundefined-var-template]
    return instance;
           ^
../../../src/tl/tl/tlClassRegistry.h:206:9: note: in instantiation of member function 'tl::Registrar<lym::MacroInterpreter>::get_instance' requested here
    if (get_instance ()) {
        ^
../../../src/lay/lay/layMacroEditorDialog.cc:546:101: note: in instantiation of member function 'tl::Registrar<lym::MacroInterpreter>::begin' requested here
    for (tl::Registrar<lym::MacroInterpreter>::iterator cls = tl::Registrar<lym::MacroInterpreter>::begin (); cls != tl::Registrar<lym::MacroInterpreter>::end (); ++cls) {
                                                                                                    ^
../../../src/tl/tl/tlClassRegistry.h:230:24: note: forward declaration of template entity is here
  static Registrar<X> *instance;
                       ^
../../../src/tl/tl/tlClassRegistry.h:223:12: note: add an explicit instantiation declaration to suppress this warning if 'tl::Registrar<lym::MacroInterpreter>::instance' is explicitly instantiated in another translation unit
    return instance;
           ^
../../../src/tl/tl/tlClassRegistry.h:96:25: warning: instantiation of variable 'tl::Registrar<lay::PluginDeclaration>::instance' required here, but no definition is available [-Wundefined-var-template]
    if (! Registrar<X>::instance) {
                        ^
../../../src/lay/lay/layMacroEditorDialog.cc:3495:52: note: in instantiation of member function 'tl::RegisteredClass<lay::PluginDeclaration>::RegisteredClass' requested here
static tl::RegisteredClass<lay::PluginDeclaration> config_decl (new MacroEditorPluginDeclaration (), 1500, "MacroEditor");
                                                   ^
../../../src/tl/tl/tlClassRegistry.h:230:24: note: forward declaration of template entity is here
  static Registrar<X> *instance;
                       ^
../../../src/tl/tl/tlClassRegistry.h:96:25: note: add an explicit instantiation declaration to suppress this warning if 'tl::Registrar<lay::PluginDeclaration>::instance' is explicitly instantiated in another translation unit
    if (! Registrar<X>::instance) {
                        ^
2 warnings generated.
```

thomaslima, 2018-03-15:
Just tried a fresh build again. This time in your master branch, commit 02c9c64

Attaching the build log.

[image]

[qt5.build.macos-HighSierra-release.log](https://github.com/klayoutmatthias/klayout/files/1813572/qt5.build.macos-HighSierra-release.log)

thomaslima, 2018-03-15:
I figured it out by using old binary search debugging. Turns out you cannot give 0 as a pointer address.

klayoutmatthias, 2018-03-16:
Hi Thomas,

many thanks for this solution. I understand perfectly now what's the problem. I'm still using Qt4 and Qt4's QSyntaxHighlighter allows the parent to be 0. Qt5 tries to dereference this pointer and the segfault happens. Too bad they modified their interface assumptions.

Let me just modify your solution a little: actually the macro editor configuration page should be independent of the IDE itself, so the MacroEditorDialog may not be instantiated at the time the dialog page is used. 

The new patch is https://github.com/klayoutmatthias/klayout/commit/13062292feb0965632b1e3be7fd1aeea5c5cfccc

Thanks a lot,

Matthias

thomaslima, 2018-03-16:
Great! that fixed it. Does this solve Lukas's problem? Let me attach the build here (with qt5.10).

I've added the file `klayout-0.25.1-macOS-HighSierra-1-Qt5101-3295b63.dmg` to my [Dropbox public folder](https://www.dropbox.com/sh/034oi1afy682lsa/AABGTzS2_-H69ESatIOWY0m7a?dl=0).

lukasc-ubc, 2018-03-17:
Hi @thomaslima 

I'm not sure if you are referring to the original issue I posted above, or the Qt issues...

Old version (Feb, Python 2), showing the problems with Qt and keys:
https://www.youtube.com/watch?v=x4n3CQV0d9E

Here is what I see on SiEPIC-Tools using the version you uploaded (klayout-0.25.1-macOS-HighSierra-1-Qt5101-3295b63.dmg). Graphics are working great!

Here is what I tested that is now working, i.e., you fixed:

* screenshots appears correct
* menu hotkeys continue working after dialog messages come
* KLayout main screen rendering problems, after dialog boxes, working well.

http://www.youtube.com/watch?v=UbDsoy7h2es

Thank you very much!
Lukas

thomaslima, 2018-03-17:
Awesome! Good to know. Thanks for the videos. I am referring to the original issue about the exceptions being caught by the macro editor. I wonder if Matthias’s addition of an ignore button and an ignore list helps.

lukasc-ubc, 2018-03-17:
Ah, very nice.  The new “Ignore” button is nice, and I like the “Don’t stop inside these files”. I note that you can’t edit the list in the GUI, but I did find it in klayoutrc so that’s easy enough to edit if necessary.

klayoutmatthias, 2018-03-17:
Hi Lukas,

Right now, you can just clear it :-)

I felt that you would usually just fill the list by walking through your code and redoing this wasn't difficult.

Matthias

thomaslima, 2018-03-18:
Here’s what I guess it’s happening. He wants some python modules to be shipped with SiEPIC. But because he can’t control the user’s python site packages, he just included the source code of these modules in the path of siepic. So he wanted their exceptions to not be caught from day zero for the user. 

Right now I’m working on shipping python and a site-packages folder inside klayout.app, so that you can install new packages to klayout with pip. I considered putting it in the .klayout folder but that might be unstable for the user during klayout upgrade.

lukasc-ubc, 2018-03-18:
The exceptions actually are only a problem when developing.  For the end user, when the Debug window is closed, the exceptions do not generate windows.

Thomas: interesting idea to allow users to have access to pip in a KLayout-compatible way.

thomaslima, 2018-03-19:
I believe this issue can be closed. @lukasc-ubc check #100 !

closed 2020-02-22 by klayoutmatthias