KLayout/klayout #92: Build performance improvement: Precompiled headers

thomaslima, 2018-03-13:
Build time takes about an hour on my Mac. If we can bring it down to 30min at least, we can use Travis CI for free to test and deploy builds of klayout. I already have a formula working, but we need to work on the build time. I believe this can be done with precompiled headers. Instructions are in http://doc.qt.io/qt-5/qmake-precompiledheaders.html .

I haven't figured out a way to do this yet with Qt and qmake. Here's the `#include` usage throughout the project (excluding qt4)

```
 817 #include <memory>
 771 #include "gsiQt.h"
 768 #include "gsiDeclQtTypeTraits.h"
 767 #include "gsiQtCommon.h"
 715 #include <QObject>
 697 #include <QEvent>
 585 #include <QThread>
 585 #include <QMetaMethod>
 577 #include <QTimerEvent>
 577 #include <QChildEvent>
 329 #include <QWidget>
 321 #include <QSize>
 307 #include <QPainter>
 273 #include <QPoint>
 269 #include <QRect>
 223 #include <QFont>
 221 #include <QPixmap>
 219 #include <QRegion>
 215 #include <QKeyEvent>
 201 #include <QIcon>
 199 #include <QPaintDevice>
 199 #include <QCursor>
 195 #include <QInputMethodEvent>
 195 #include <QFocusEvent>
 193 #include <QGraphicsEffect>
 191 #include <QAction>
 189 #include <QLocale>
 189 #include <QBitmap>
 187 #include <QPalette>
 185 #include <QMargins>
 183 #include <QPaintEngine>
 183 #include <QLayout>
 181 #include <QWindow>
 179 #include <QSizePolicy>
 179 #include <QFontMetrics>
 177 #include <QStyle>
 177 #include <QMouseEvent>
 177 #include <QKeySequence>
 171 #include <QPaintEvent>
 169 #include <QShowEvent>
 169 #include <QResizeEvent>
 169 #include <QHideEvent>
 167 #include <QWheelEvent>
 167 #include <QFontInfo>
 165 #include <QTabletEvent>
 165 #include <QMoveEvent>
 165 #include <QCloseEvent>
 163 #include <QGraphicsProxyWidget>
 163 #include <QDropEvent>
 161 #include <QDragMoveEvent>
 161 #include <QDragLeaveEvent>
 161 #include <QDragEnterEvent>
 161 #include <QContextMenuEvent>
 161 #include <QBackingStore>
 161 #include <QActionEvent>
 151 #include <vector>
 149 #include <QPointF>
 141 #include <string>
 138 #include "tlString.h"
 117 #include <QRectF>
 115 #include "tlUnitTest.h"
 107 #include <QColor>
 106 #include "dbLayout.h"
  99 #include <QDir>
  99 #include "gsiDecl.h"
[... 1314 lines cut ...]
   1 #include "layDXFWriterPlugin.h"
   1 #include "layDXFReaderPlugin.h"
   1 #include "layCommonReaderPlugin.h"
   1 #include "layClipDialog.h"
   1 #include "layCIFWriterPlugin.h"
   1 #include "layCIFReaderPlugin.h"
   1 #include "layBrowseShapesForm.h"
   1 #include "layBrowseInstancesForm.h"
   1 #include "imgWidgets.h" // for interpolate_color()
   1 #include "imgWidgets.h"
   1 #include "imgPropertiesPage.h"
   1 #include "imgPlugin.h"
   1 #include "imgNavigator.h"
   1 #include "gsiQtHelper.h"
   1 #include "gsiClass.h"
   1 #include "fonts.cc"
   1 #include "fixedFont.h"
   1 #include "drcCommon.h"
   1 #include "dbWriterTools.h" 
   1 #include "dbTypes.h" 
   1 #include "dbLayout.h" 
   1 #include "contrib/dbGDS2TextReader.h"
   1 #include "contrib/dbGDS2Text.h"
   1 #include "algorithm"
```

klayoutmatthias, 2018-03-13:
Memory will be an issue too, not just build times. So I don't think TravisCI can be employed.

thomaslima, 2018-03-14:
It almost got it! https://travis-ci.org/lightwave-lab/klayout/builds/352482878

klayoutmatthias, 2018-03-14:
Hi Thomas,

I agree that TravisCI is a nice option as it's well integrated with GitHub and all the important targets are supported. I'm just sceptic about the resource requirements. Precompiled headers don't boost performance sufficiently in my experience and the resource hogs are the linker steps.

The big contribution is the Qt binding. From the log of your Travis Job I'd say it's halfway through. If you disable that (build -without-qtbinding), I'm sure you can handle that. I don't really consider Qt binding to be important for CI (in the sense of integration, not deployment) - what's more important is that the tests are run (run "ut_runner"). It can be configured to produce JUnit XML files and Travis should be able to parse them.

Over all, if one includes the whole Qt and language binding stuff, the project grew too big. I'd like to partition it, but frankly my priorities are different ones currently.

BTW: I don't see that Travis supports Windows. Neither does it seem to support CentOS. Currently, I'm running various Jenkins tasks on my own hardware using Docker containers for different Linuxes and native Jenkins for Windows. It's a private CI but sufficient for me so far. 

Matthias

thomaslima, 2018-03-14:
Thanks Matthias,

I will try the compilation without qt binding. You are right that travis is limited in this way, but it would be awesome for deployment especially for MacOS images. I was surprised they even had free VMs running MacOS available for opensource.

Precompiled headers did not speed up significantly on my Mac. But maybe I was not configuring it right.

But you might be right that private hardware would be the way to go. Maybe we can team up with some people to buy and install one of those on a university campus?

Alternatively, I think we are already in a position of easy scriptable compilation using dependencies from homebrew. (Thanks to @Kazzz-S ). I will work on making a homebrew package of klayout (#88), once we fix bug in #85.

So the conclusion for now is that we may have to move to private hardware. I am closing the issue now.

PS: As a curiosity, what would be your priorities for klayout? I think this software can be strategic for the Integrated photonics field, because we need precision layout at this stage. @lukasc-ubc is making significant progress in making a PDK based on klayout which frankly right now beats commercial ones, (to me especially because of the qt binding with python feature). I would like to have klayout+siepic pdk become some sort of alternative standard in this domain. Otherwise we'll all be hostage to these big companies charging 5K per year per user license fees for a mediocre software. (prob said too much already)

closed 2018-03-14 by thomaslima

klayoutmatthias, 2018-03-14:
No worries, I know this situation :-)  And 5K is little compared to what companies charge in the EDA business in general.

I'm trying to focus more on verification topics - DRC functionality and going beyond (towards connectivity extraction and maybe even LVS kind of functionality). And the DB API needs constant maintenance.

Best regards,

Matthias

lukasc-ubc, 2018-03-15:
Hi Matthias, Thomas

Regarding Matthias’ comment:
> I'm trying to focus more on verification topics - DRC functionality and going beyond (towards connectivity extraction and maybe even LVS kind of functionality). And the DB API needs constant maintenance.
> 

I spent some time on extraction.  We have a basic netlist extraction for the photonics, and we have some thoughts to add the metal extraction. We create a netlist in spice format and then send it to a circuit simulator.  During the extraction step, we perform a lot of important functionality checks, as well as Design For Test.  Some of these are quite interesting topics not addressed by commercial tools.  We also have an interest in the reverse: taking a netlist and generating a layout; my student has made some progress.  I started to write up a description of the current features here: https://github.com/lukasc-ubc/SiEPIC-Tools/wiki/SiEPIC-Tools-Menu-descriptions#verification
It has also been published in an industrial-focused conference: https://www.dropbox.com/s/wqgs0ftazc895a5/2016_SPIE_PW_schematic.pdf?dl=1

In terms of hardware for the project, please let me know what you guys need and I can contribute (hardware, hosting).

Regards,
Lukas

thomaslima, 2018-03-16:
Thanks for sharing your paper @lukasc-ubc . This seems like a really hard problem to crack. Right now I use klayout's python bindings to position pcells and connect them with waveguides. photonic compatible DRC+Netlists would be very very useful to us.

In terms of hardware contribution, I think the idea is to have a macOS computer server lying somewhere, with a couple of virtual machines and this Jenkins tool installed. This server would clone the repo at every commit to master, build klayout, test it, produce .dmgs and deploy them to some location. Matthias uses his personal hardware for windows and linux. 

Another possibility is to pay about $70 per month to travis-ci, which basically does the same thing in the cloud.. I just found out looking for this that they offer this plan for free for students. I am going to try it right now!

[image]

lukasc-ubc, 2018-03-16:
Hi Thomas,

Indeed, the travis-ci  company seems very keen to support university work and open source projects.

As for a mac, I have a macmini MacOS for university work onto which I can give you an account to implement the auto-compile.

paper, you mean this one?  http://onlinelibrary.wiley.com/doi/10.1002/lpor.201700237/full <http://onlinelibrary.wiley.com/doi/10.1002/lpor.201700237/full>

which problem are you referring to specifically?

regards,
Lukas

thomaslima, 2018-03-16:
I had meant the SPIE paper. I will read your new article with Bogaerts more carefully. Very fresh. Thanks!
I am referring to schematic driven layout in photonics, because it's very platform/foundry/tech specific, and it has many more constraints than metal, with many more fail conditions and smaller parameter space. So different people from different companies are going to come up with different standards etc. I am honestly not very familiar with the current state-of-the-art for SDL. Even if it existed, I wouldn't trust it, or I would like to tweak the layout somehow. And what is better than a python script for it? I have painful memories with Pyxis's Ample and Phoenix's.

I will read your paper to learn more.