KLayout/klayout #94: (MacOS / hiDPI) Retina display rendering of canvas

thomaslima, 2018-03-16:
Dear all, 

I've been working on this retina display rendering feature recently. I've scratched my head quite a bit and I think I'm starting to, dare I say, deal with some bugs in Qt. Here's the Qt doc about it: http://doc.qt.io/qt-5/qpainter.html#drawing-high-resolution-versions-of-pixmaps-and-images

When you have a `QImage`, it suffices to say `qimg.setDevicePixelRatio(2)` for retina Macs, then the `QPixmap` created after it respects the DPR and renders correctly. But a `QBitmap` (which inherits `QPixmap`) object, such as the ruler, fails to render correctly, even though [it should](http://doc.qt.io/qt-5/qpixmap.html#setDevicePixelRatio). I've tried many ways to make it render, and the only way i got it to work was to convert the bitmap to an image, and then back to pixmap. [This part of the code](https://github.com/lightwave-lab/klayout/blob/d97748df12f1841c33a327cf62ccebc1854118a3/src/laybasic/laybasic/layLayoutCanvas.cc#L717) would need some serious refactoring for that to work. I would need help for that.

I'm working on the `macos_retina` branch in my fork: https://github.com/lightwave-lab/klayout/tree/macos_retina , feel free to contribute! 

Build with `./build4mac.py -p B36 -q Qt5Custom && ./build4mac.py -p B36 -q Qt5Custom -Y && qt5.pkg.macos-HighSierra-release/klayout.app/Contents/MacOS/klayout`, with `qt` and `python` from brew.

I'm posting some preview pictures here, including failures in the case of 2x oversampling.

[image]
[image]
[image]
[image]

klayoutmatthias, 2018-03-18:
Hi Tomas,

I only have a VirtualBox, no Retina display :-(

I assume you have tried the refactoring already. QBitmap is supposed to be more efficient, but today that should not be a big issue. I understand that you're saying that using a QImage would solve the issue (if you use qimg.setDevicePixelRatio(2) ), right?

Currently the mentioned code works like this: 
 - the layout is painted to a QImage and transferred to the screen after subsampling
 - the foreground objects (such as rulers, selection markers, highlights and other markers) are painted to pre-allocated bitmaps ("planes") and transferred individually to the screen as overlays. Subsampling happens on bit-by-bit basis which also explains the uglyness of the rulers in subsampling mode.

So a quick solution would be to add the bitmaps to the QImage before subsampling and only subsample the image as a whole. Should be quick, cleaner and safe for Retina.

I'll propose a patch. Give me a some time. I'll also try to get rid of the #ifdef's as different buffer schemes are no longer required. That's a legacy concept from old X11 times.

Best regards,

Matthias

thomaslima, 2018-03-18:
hi Matthias, I spent some time dealing with this, and right now the code in my branch simply multiplies the viewport_l's dimensions by 2, which is the devicepixelratio for macs. But this variable can be captured by something like qApp.mainscreen.devicePixelRatio(), so that it would be more compatible to other platforms. I think i've put the *2 everywhere necessary for the canvas in my branch, if you can take a look. But everywhere else we use a QImage the same has to be done. We also need 2x versions of icons.

I would love to help you out but I feel like I need more insight of the code to tackle all the places you use QImage without messing up the build for other platforms.

Let me show you a few examples on how it renders on Macs (but I am sure same would happen in windows):

[image]
[image]
[image]

klayoutmatthias, 2018-03-19:
Hi Thomas,

I have just created a pull request with some refactoring attempt (https://github.com/lightwave-lab/klayout/pull/1). 

you did a remarkable job in identifying the root cause and actually your solution was pretty much the correct approach :-)  I just had to try to turn it into a generic solution.

I could only verify it's still working on a normal display - I hope high-DPI displays still work.

Here my reasoning:

As I understand the "devicePixelRatio" story, it goes like this: high-DPI displays have much denser pixels but there is still software around which uses pixel dimensions - for example 32x32 icons (me included). So the solution is to use a virtual pixel space which is 2 times coarser than the physical one, but as the rendering engines support floating-point coordinates, you can paint on half pixels. The effect will be the same than painting on integer pixels but at twice the size.

In order to support this, we need to establish images which have a bigger dimension (initially), then set the devicePixelRatio to 2. This will make the images shrink to a virtual size which is the half. The image data (such as scanline data etc.) will still be twice. Painting on images would happen with the virtual pixel size, but KLayout does not benefit from that as it has it's own rendering engine.

So we need to do this: 
 * For a high-DPI display rulers and other markers need to be drawn twice as thick normal so they don't appear too thin.
 * The width and height of the widget are virtual width/height. To obtain the true width and height we have to multiply their values with 2.
 * The QBitmap issue needs to be solved.

The third point was done by using an alternative implementation: instead of painting the subsampled bitmaps for the dynamic objects to the screen as bitmaps, I collect them (full resolution) in a QImage and subsample this. This path utilizes the same subsampling code than the normal (static) display. It just needs to be extended to support alpha channel subsampling, because it's drawn over the static content and needs to be transparent.

I turns out that this actually reduces the code and as a side benefit, the rulers look smoother in subsampling mode. 

Let me know if the Retina display has issues with this solution. 

Best regards,

Matthias

thomaslima, 2018-03-19:
Thank you so so much! I will try it out right now. The icons loaded from png files actually render in the say you describe already, as my screenshots suggest. The point is if we added a high resolution one, it would look prettier for those with high dpi displays.

thomaslima, 2018-03-19:
I believe your solution worked well for the rendering, but the mouse events are now trapped on the lower-left corner of my display. That is to say: trying to draw a ruler across the canvas actually draws it half-sized on the lower left side. But the rulers are drawn in high resolution right now, so it's great progress!

[image]

thomaslima, 2018-03-19:
The ruler labels also disappeared when I set the oversampling to 2x. I have no idea how to solve the mouse event issue. I have to wait for a hint from you.

thomaslima, 2018-03-19:
Another hint to you: the xy coordinates on the bottom-right, as I move the mouse, are also halved compare to true size.

thomaslima, 2018-03-19:
@klayoutmatthias I think it has to do with the `m_trans` variable in the `ViewObjectWidget::mouse_event_trans` method. Really excited to test this out.

klayoutmatthias, 2018-03-19:
Hi Thomas,

Oh yes ... the mouse coordinates are "virtual" too ... what a mess. 

I created a pull request, but maybe you're right and it's better (and easier) to change the transformation. Please ignore the pull request if you find the transformation approach is working better.

I have no idea why the ruler labels don't show. Maybe something with the stored fonts.

Matthias

klayoutmatthias, 2018-03-20:
Your approach is better :-) - I have created one more pull request modifying the transformation.

klayoutmatthias, 2018-03-20:
I got the reason for the missing ruler labels ... it's entirely stupid. There are fonts provided for resolutions 1, 1/2 and 1/3. No more was required so far. Now we also need 1/4 and 1/6.

I'll try to create the fixed font bitmaps. Currently they are compiled in.

Sorry for this confusion.

BR, Matthias

klayoutmatthias, 2018-03-20:
I have upgraded the pull request. There should be fonts for all resolutions to 1/6 now.

Matthias

thomaslima, 2018-03-20:
It works now after lightwave-lab#3. I had to fix the import headers for the new `fontgen.cc` but it just worked out of the box. Very nice job!

I shall change the title and keep this open for more hiDPI compatibility.

[image]

klayoutmatthias, 2018-03-20:
Ah yes, sorry for the fontgen.cc headers ... the fontgen.cc is just a helper tool. 

Instead of compiling in the font bitmaps I should generate them on the fly. The I'm going to remove fontgen.cc.

The other QImage objects should not require refactoring. The are used for generating printouts and screenshots. So I don't think they need the DPR scheme.

I have started to use SVG for the icons, but not all of them are translated yet. I can try to do so, but it will take a while.

I'm going to merge this into master (as you did). I would like to release 0.25.2 and we can include the Retina canvas support into this. 

Best regards,

Matthias

thomaslima, 2018-03-20:
Thank you for including that in .25.2. Let me know if we can help with the svg.

About the other QImage/QPixmaps objects, I meant to say the ones that impact the issues highlighted [in my comment above](https://github.com/klayoutmatthias/klayout/issues/94#issuecomment-374065226).

thomaslima, 2018-03-20:
Here's a windows discussion of display scaling: https://blogs.technet.microsoft.com/askcore/2015/12/08/display-scaling-in-windows-10/

klayoutmatthias, 2018-06-10:
I think we finally solved this, did we?

thomaslima, 2018-06-15:
@klayoutmatthias the layout canvas is solved, but there are other places which are not rendered properly still, outlined in my Mar 18 and Mar 20 posts. But these don't affect me much, personally.

thomaslima, 2018-08-03:
Made some progress on this. See PR #150

klayoutmatthias, 2022-12-06:
Try 0.28 :)

closed 2022-12-06 by klayoutmatthias