KLayout/klayout #95: salt package manager - disappeared?

lukasc-ubc, 2018-03-16:
I just noticed that several of my packages disappeared from salt manager, specifically SiEPIC-Tools, SiEPIC-EBeam-PDK...

Any idea what happened?

thank you
Lukas

lukasc-ubc, 2018-03-16:
I just added them in again.  I think this is the 2nd time this happened in the last month or so.

klayoutmatthias, 2018-03-16:
Hi Thomas,

I need to check that ... I noticed the same for my own packages. There is a background job running that disables packages when their URL is not accessible. If so, they are getting hidden because it is assumed they were removed. After a grace period of 3 days they are retired eventually.

Unless you really removed them, you don't need to re-add them. They are not really deleted. Just hidden. The background job runs once an hour so they should re-appear an hour later latest. In your case from the logs I see the scan failed (with some connection error) at 8.35 today (CET I think). I assume I had reappeared at 9.35 but you have been quicker and have re-added it ... :-)

Jokes aside - apologies for this. I'll try to figure out why the connection fails and I should modify the behaviour so packages are not getting hidden immediately. Or retry several times.

The log files tell me that has happened 25 times in the last 4 months, out of that 9 times in the last week. Looks like there is need for action. 

Best regards,

Matthias

klayoutmatthias, 2018-03-16:
Hi Lukas,

I could not really guess what's going on. The server ran out of sockets. My best guess is some kind of flooding by requests.

I have modified the behaviour so packages are not dropped from the list immediately. Only after the grace period expired, they are actually dropped. So a temporary glitch will not make the packages disappear.

Please let me know when this ever happens again.

Best regards,

Matthias

lukasc-ubc, 2018-03-16:
Thanks.  I’m glad it isn’t a human that was making the decision to remove them ;)

Feedback on the package manager: excellent for students and in general getting people up and running quickly.

klayoutmatthias, 2018-03-18:
Hi Lukas,

no human involved. Just algorithms :-)

And thanks for the feedback.

I'll close this issue and hope I don't need to reopen it.

Best regards,

Matthias

closed 2018-03-18 by klayoutmatthias