KLayout/klayout #98: Can not run python in klayout

Joyin-Hu, 2018-03-18:
Hi,

Incorporating python with klayout seems very cool. I am trying to query some info from layout. But I have hard time to get it work even after searching around for the solution and tried a few approaches:

1. Installed klayout 0.25.1 win64 from official website, installed the python packages under salt management as well, tried to run some python script under macro development, but console showed nothing.

2. So I thought maybe I had to build klayout to support python. But I got an error as shown below: 
$ ./scripts/deploy-win-mingw.sh -64
------------------------------------------------------------------
Running build for architecture win64 ..

  target = /home/Joyin/klayout/bin-release-win64
  build = /home/Joyin/klayout/build-release-win64
  version = 0.25.2

Scanning installation ..

Version Info:
    Version: 0.25.2
    Date: 2018-03-18
    Revision: e833edd

Using qmake: qmake
    Using Qt 5 API

Using Ruby interpreter: ruby
    Ruby library found: C:/msys64/mingw64/lib/libx64-msvcrt-ruby250.dll.a
    Ruby headers found: C:/msys64/mingw64/include/ruby-2.5.0 and C:/msys64/mingw      64/include/ruby-2.5.0/x64-mingw32
    Ruby installation is in:
    - C:/msys64/mingw64/lib/libx64-msvcrt-ruby250.dll.a (lib)
    - C:/msys64/mingw64/include/ruby-2.5.0 (headers)
    - C:/msys64/mingw64/include/ruby-2.5.0/x64-mingw32 (arch headers)
    Ruby version code is 20500

Using Python interpreter: python3.6m
    Python library found: C:/msys64/mingw64/lib/libpython3.6m.dll.a
    Python headers found: /include/python3.6m
    Python installation is in:
    - C:/msys64/mingw64/lib/libpython3.6m.dll.a (lib)
    - /include/python3.6m (includes)

Features:
    Qt bindings enabled
*** ERROR: wrong -pyinc path: /include/python3.6m/Python does not exist or is not a file
 ------------------------------------------------------------------

I also tried to specify pyinc="/c/msys64/mingw64/include/python3.6m/Python.h" , but the same error.

3. Next I tried to build under ubuntu subsystem 16.04 under win10, building seems fine. But trying to run "klayout" in directory of bin-release, terminal returned something like "no command found". There is a file name "klayout" under bin-release directory.

Could anyone help here?

Thanks!

lukasc-ubc, 2018-03-18:
You shouldn’t need to compile KLayout yourself.  I’ve been using Python within KLayout for over 2 years and never compiled it myself.

Start with the basic Macro Development system. Switch to “Python” by clicking on the bottom right.  Then type in things like:

     print(1+1)

And see if Python is working.

Then start writing your own macros.

The SiEPIC-Tools on the package manager is an example of what you can do.  Note that when installed with the package manager, it is read only in the “salt” folder.  You’ll need to move it if you want to edit.

Joyin-Hu, 2018-03-18:
I did try to print something from there. But nothing happened. Here is the simple script I tried.

print("Test")

import pya
layout = pya.Layout()
top = layout.create_cell("TOP")
l1 = layout.layer(1, 0)
top.shapes(l1).insert(pya.Box(0, 0, 1000, 2000))
layout.write("test.gds")

Any clue where I did wrong? I have anaconda3 with python 3.6.3 installed and included in my system Path. Will it affect?

lukasc-ubc, 2018-03-18:
Perhaps provide some screenshots.  Nothing obviously wrong here.

Doesn’t matter that you have installed Anaconda, since KLayout for Windows comes with it’s own Python install.

Did you try the code below line by line in the console?

Joyin-Hu, 2018-03-18:
I think the problem is that I tried to run the code under local folder previously. I just tried running from Local - python branch. It is now working. Thanks for the help!

[image]

klayoutmatthias, 2018-03-19:
Hi Joyin-Hu,

the "python branch" is intended for python modules - i.e. plain Python files you want to include in your macros. Normal macros ("wrapped" Python code that can be tied to a menu for example), simply go into "Local" where they are found by KLayout.

Can we close this issue?

Thanks,

Matthias

Joyin-Hu, 2018-03-19:
Thanks Matthias!

closed 2018-03-19 by Joyin-Hu