KLayout/klayout #100: (macOS Improvement) Packaging python3 with klayout.app

thomaslima, 2018-03-19:
Hi all,
I've made significant progress in embedding python3 from brew with klayout on a Mac.
It copies the Python.framework from `/usr/local/opt`, then replaces the links to all dynamic libraries inside, collects more dependent dynamic libraries (e.g. openssl), then patches the modules `site`, `distutils` and `pip` so that `pip install` would see the right path for site-packages (/Applications/klayout.app/Contents/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages`). I have included a script called `klayout_console` inside `/Applications/klayout.app/Contents/MacOS/`. What do you think?

Below is an example of when I installed numpy via pip directly to klayout.app.
[image]

I've included a .dmg in my [dropbox](https://www.dropbox.com/sh/034oi1afy682lsa/AABGTzS2_-H69ESatIOWY0m7a?dl=0) for testing. Commit hash 07dde1c .

PS: Here's the procedure for building:
`python3 ./build4mac.py -p b36 -q qt5brew`
`python3 ./build4mac.py -p b36 -q qt5brew -y`
Package deployed in `qt5.pkg.macos-HighSierra-release/klayout.app`

lukasc-ubc, 2018-03-20:
This looks very promising!  So our scripts in KLayout could check to see if numpy is installed, and if not, just do a pip.main(['install'...])?  That sounds very nice.

lukasc-ubc, 2018-03-20:
Well, within KLayout’s IDE, I was able to get numpy installed, using the commands you mentioned.  So no need for the klayout_console.  Can do in-app installation of missing packages.

Does import pip, and pip.main… work in Linux and Windows too?  If I recall, the Windows build doesn’t include pip (hence my Package which includes the missing libraries for Windows people…)

thomaslima, 2018-03-20:
I don't know how pip behaves in other OSs. I found [this article](https://stackoverflow.com/questions/4750806/how-do-i-install-pip-on-windows) for Windows. There is an issue #81 requesting embedding python in the window's klayout. As for linux, no idea what the situation is right now. Before we rely on using pip in Salt packages, we gotta check that pip will behave nicely without admin permissions in all three systems.

By the way the latest `.dmg` in my dropbox has both retina and this feature. I am going to use this version from now on.

PS: Can you show how you run one-time commands from the klayout IDE? Do you write a temp script for that? I'm asking because I also want to have my labmates become more comfortable scripting in klayout.

lukasc-ubc, 2018-03-21:
in klayout-0.25.2-macOS-HighSierra-1-Qt5101-07dde1c.dmg:

- File | Log Viewer: not resizable

You have two versions, one with brew and one without?  In both versions, in-KLayout pip installation of numpy and requests is working.  Seems like the Python3 installations are different for the two, since I need to install the missing modules separately.

One-time commands in KLayout IDE: 

- F5
- click on the bottom right "Python" in console
- then type away as usual in the Python console

lukasc-ubc, 2018-03-21:
https://youtu.be/oRt5bTERGOc

thomaslima, 2018-03-21:
@lukasc-ubc Both have brew's python. sorry about the confusing name. Both of them have embedded python. so packages you install are going inside their respective .app. They don't see each other. This was intended as a feature.

ps: I think you might have linked the wrong youtube video.

thomaslima, 2018-03-26:
@lukasc-ubc indeed nothing is resizable! Really annoying. I will try to fix that.

thomaslima, 2018-03-26:
This was fixed five days ago! https://code.qt.io/cgit/qt/qtbase.git/commit/?id=45c62aae4b068c3dc4549f841f6089ca826b37a5

thomaslima, 2018-06-26:
I have just uploaded a new version in my dropbox with this issue fixed. Fresh new Python 3.6.5 and Qt5.11.1.

klayout-0.25.3-macOS-HighSierra-1-Qt5111brew-Py365-896f734.dmg
klayout-0.25.3-macOS-HighSierra-1-Qt5111brew-Py365-896f734.dmg.md5

https://www.dropbox.com/sh/9t3f8yp5edztl6f/AADGWeZpjJgPh5HBwHXGmraYa?dl=0

PR #138 (2018-06-28): Macos embedding python and Qt5.11.1 bugfix
I tested this on Mac OS with qt11.1 installed from brew.
Closes #100 

feature:
- MacOS compilations can embed python's dynamic libraries with the app bundle

bugfix:
- Qt5.11 complains about using Qt::blue and similar as a valid QColor
closed 2018-06-26 by thomaslima