KLayout/klayout #102: Potential issue while upgrading from .25.1 to .25.2

thomaslima, 2018-03-20:
A labmate reported that when he installed the new klayout, he got that “Got an empty string where a real number was expected” [exception](https://github.com/klayoutmatthias/klayout/issues/85#issuecomment-372336582).

This could be solved by deleting the `~/.klayout/klayoutrc` file.

Can anyone help upgrade from a clean .25.1 to a .25.2 to check whether this will be a major issue for end users? Otherwise, more in the long term, we have to change klayout code so that it's failsafe with regards to "corrupted" klayoutrc.

klayoutmatthias, 2018-03-20:
Hi Thomas,

I will try to reproduce the issue.

Matthias

thomaslima, 2018-03-20:
Don't try too hard. Might be a false alarm, as he at some point installed a dev version.

klayoutmatthias, 2018-03-20:
No, not a false alarm. A real issue. Thanks a lot for testing!!! 

I did not see that myself (yet) because I did not try to upgrade (yet).

I have fixed that plus made the application safe against failure when reading the config. Config errors are reported but ignored otherwise.

Thanks again and best regards,

Matthias

closed 2018-03-20 by klayoutmatthias

thomaslima, 2018-03-20:
Thank you for figuring this out. We really appreciate it.

LG
Thomas

commit 21e2af2a (2018-03-20): Fixed #102 (Potential issue while upgrading from .25.1 to .25.2)

The reason was that for 0.25.1 "macro-editor-font-size" was allowed
to be an empty string (the default). Which meant: take default application
font size. In 0.25.2 this now was required to be a number and 0 was
the default for "auto" font size.

Two changes:
 - The default is back to empty string ("0" is still allowed as default)
 - The application was made safe against broken configuration files: an
   error is printed to the log, but apart from that the application
   will work (the configuration value is ignored however).