KLayout/klayout #103: 0.25.2 release available - MacOS images looked for

klayoutmatthias, 2018-03-21:
@kazzz / @thomaslima: if you drop me a link to a .dmg package I will include them on the server.

Kazzz-S, 2018-03-21:
Hi @klayoutmatthias,

Thank you for making the release!
I've started building the latest one: 0.25.2.
I'll let you know later.

Regards,
Kazzz

thomaslima, 2018-03-21:
Hi @klayoutmatthias.

I would trust @Kazzz-S's images better, because my builds are not done in a virtual machine. However, if you don't mind, I'd like you to include one of my versions built with python3.

Kazzz-S, 2018-03-21:
Hi @klayoutmatthias ,

Compressed DMGs are ready here:
https://www.dropbox.com/sh/x23vsv1374m3d5b/AABABTHU1N0jGNEqLqdkfv8Xa?dl=0 

DMG file|Target OS|Qt|Ruby|Python
----------|-----------|---|------|---------
klayout-0.25.2-MacOSX-Yosemite-1-Qt510Xmp.dmg|10.10 Yosemite,  10.11 El Capitan, 10.12 Sierra|5.10.0 MacPorts|System 2.0.x|System 2.7.x
klayout-0.25.2-macOS-HighSierra-1-Qt510Xmp.dmg|10.13 High Sierra|5.10.0 MacPorts|System 2.3.x|System 2.7.x

Regards,
Kazzz

thomaslima, 2018-03-21:
And here's my link. https://www.dropbox.com/sh/9t3f8yp5edztl6f/AADGWeZpjJgPh5HBwHXGmraYa?dl=0

| DMG file | Target OS | Qt | Ruby | Python |
| ------- | ---- | ---- | --- | --- |
|klayout-0.25.2-macOS-HighSierra-1-Qt510XbrewPy3.dmg | 10.13 High Sierra | 5.10.1 Brew | System 2.3.x | Brew 3.6.4 (Embedded) |

lukasc-ubc, 2018-03-21:
http://www.klayout.de/build.html has broken URLs and typos for the MacOS version.

klayoutmatthias, 2018-03-21:
@lukas - you too fast. I was still editing. It was out of sync for not more than few minutes ... :-)

klayoutmatthias, 2018-03-21:
@thomaslima @Kazzz-S 

I took kazzz's binaries, as you suggested. No offence meant :-)

Thanks both of you.

Matthias

Tho

lukasc-ubc, 2018-03-21:
@klayoutmatthias, 
Thanks! Will you also post Python 3 brew version?  I've been spending a lot of time with that one and really like it.

klayoutmatthias, 2018-03-21:
@lukasc-ubc Is that thomas' version? I lost track of the packages ...

lukasc-ubc, 2018-03-21:
yes.

klayoutmatthias, 2018-03-21:
@lukasc-ubc I see ... I'll put it on the server too. But I'll do this on my STAGING instance first. Just to avoid a race condition with you :-)

klayoutmatthias, 2018-03-21:
Done. And unless you picked the exact millisecond when I switched staging to production, the links should be functional :-)

@thomaslima: I think I can merge your development branch to master, right? That's the one the build was done with and should be integrated into master.

thomaslima, 2018-03-21:
I would rather not. It is not ready. We need to close #100 first. That is, there must be an equivalence between the mac version and windows, don't you think?

In my version, you can install packages with `pip` and they go directly into the `klayout.app`.

lukasc-ubc, 2018-03-21:
@klayoutmatthias , @thomaslima, @Kazzz-S 

Of the two versions that Matthias posted on klayout.de: 
- The kazzz version which uses Qt5.10 has the bug where the menu hotkeys don't work after a QMessageBox has been created.  There is also a new bug where KLayout freezes when you try to quit; it requires a force-quit.
- The thomas version which uses Qt5.10.1 fixes the two bugs. I believe there is a typo on http://www.klayout.de/build.html where the thomas version is using Qt5.10.1 rather than Qt5.10.

So I am using the thomas version.

thomaslima, 2018-03-21:
The typo is probably my bad. I wasn't paying attention when I named the .dmg file. @lukasc-ubc I am sorry about the issues with qt5.10. I haven't tested that version much, but it at least quits fine in here. How can I reproduce this bug?

lukasc-ubc, 2018-03-21:
@thomaslima 

You actually solved the Qt issues by using Qt5.10.1, so I'm happy.

the KLayout freezing bug is intermittent. I can't find a repeatable means to reproduce it. If I do, I will post it. 

Check out the following nice feature you added: auto-install numpy!  Namely, SiEPIC-Tools finds that numpy isn't installed, and does a pip install.  No sudo or anything required.  I just have a dialog that asks the user "do you want to install numpy", and Yes, triggers the install.  Then the simulations proceed as normal.

It is in SiEPIC-Tools and EBeam PDK.  Here is an example layout for which experimental data is available on GitHub: https://s3.amazonaws.com/edx-course-phot1x-chrostowski/Phot1x/EBeam_LukasChrostowski_E.gds

and a video showing what it does: https://youtu.be/oRt5bTERGOc

Note the circuit simulator is Lumerical INTERCONNECT, so if you don't have it you won't be able to do the whole thing.

Try SiEPIC | Measurement Data | Fetch.  if "requests" is missing in python, it will auto-install.

closed 2018-03-23 by klayoutmatthias