KLayout/klayout #104: Mac 0.25.1 Macro development Menu Action

OfficialVinsmokeSanji, 2018-03-21:
Hi 
First of all many thanks to the developers on the software, really appreciate it. 
I am trying to add a shortcut of "Move By" by using Macro Development after reading a post on the forum (https://www.klayout.de/forum/comments.php?DiscussionID=616&page=1#Item_0)

However, i cant find the key binded with Move By, since Mac version doesnt have this " File > Setup > Key bindings." 
Here is my script: 
module MyMacro
  include RBA
app = Application.instance
mw = app.main_window
menu = mw.menu
a = Action.new
a.title = "Move By"
#a.icon = "file path" # If picture icon is desired, else delete this line
a.on_triggered { menu.action("edit_menu.selection_menu.move-by").trigger }
menu.insert_item("@toolbar.end", "moveby", a)
end

Could you please kindly let me know what shoud i put for the move by function? and also where may i find all the key binding info?

Kazzz-S, 2018-03-21:
Hi,

Does this help?
[image]

By the way, version 0.25.2 will be available shortly.

Kazzz-S

OfficialVinsmokeSanji, 2018-03-21:
This is exactly what im looking for. 
Thank you so much!

klayoutmatthias, 2018-03-21:
It's available now :-)  ... and I will close this issue if you don't mind.

Thanks,

Matthias

closed 2018-03-21 by klayoutmatthias