KLayout/klayout #107: Undo action not working with shapes

thomaslima, 2018-03-31:
Hi,

I just laid out a box, or a path, then pressed Cmd+z (or manually via edit > undo create box/path), and the action was not undone. Moreover, the redo did not re-insert the object.

Note: Undo/redo action works fine with rulers. And also with Cell Instances. Can you reproduce?

thomaslima, 2018-04-02:
Hi Matthias,

I have tried with the -nc option. It must be only my compilation, or the Mac ones (qt > 5.10)

Another datapoint: The add layer event can be reversed.

Do you use qt5 as well? It would be helpful if we had the same qt in all platforms.

Best,

klayoutmatthias, 2018-04-07:
Hi Thomas,

sorry for the delayed response. I am using Qt4 still on Linux, but on Windows I'm using 5.6.2. On both platforms I don't see the issue.

But I *do* see the issue on my VirtualBox MacOS. I'll try to debug that.

Best regards,

Matthias

closed 2018-04-07 by klayoutmatthias

klayoutmatthias, 2018-04-07:
Found the issue. It should work now. It was a difference in the linking of RTTI information - clang requires classes to have uniform public visibility to allow dynamic_cast. But only on Mac and only in release build. That's why it was difficult to debug.

Thanks for reporting the problem.

Matthias

thomaslima, 2018-04-09:
Thank you so much! I will test it and let you know.

thomaslima, 2018-04-09:
Just tested without cleaning build directory. Problem persists.

thomaslima, 2018-04-10:
Rebuilt and tested afresh. problem still persists on my MacOS with qt 5.10.1.

klayoutmatthias, 2018-04-10:
(please imagine me cursing ...)

I was quite proud having found the issue ...

So what works for me (after the fix) with Qt 5.9.3 from MacPort:

  - create a new layout
  - create a new layer (1/0)
  - draw a box
  - use "Edit/Undo": the box vanishes

Are you testing the same? Or a different combination?

The root cause is not detected by the compiler nor does it give a
warning at runtime. So I cannot rule out other occurances of the
problem.

Any hints are welcome.

Best regards,

Matthias

thomaslima, 2018-04-10:
Unfortunately I have no hints so far, except that I am compiling with Qt 5.10.1 and doing the same steps that you are doing. I was even careful enough to start klayout with -nc -e -rx options. I saw you included a unit test on that but I still have to learn how to run it.

thomaslima, 2018-04-10:
Weird. The test passed!

thomaslima, 2018-04-10:
Ok I kind of have a false alarm. When I execute klayout from the built binary folder, your patch works. But when I execute Kaazz's deploy script to a klayout bundle, the undo button stops working. very weird. I have to investigate more.

thomaslima, 2018-04-10:
I'd like to share something very curious. I activated the flag DYLD_PRINT_LIBRARIES while launching klayout from both folders, and recorded which libraries were loaded while loading the app. Results are attached. 
[dyld_libs.txt (bin)](https://github.com/klayoutmatthias/klayout/files/1896644/dyld_libs.txt)
[dyld_libs.txt (pkg)](https://github.com/klayoutmatthias/klayout/files/1896647/dyld_libs.txt)

The packaged version loaded the same libraries, but also some extra ones:
```278a279,291
> dyld: loaded: /System/Library/CoreServices/RawCamera.bundle/Contents/MacOS/RawCamera
> dyld: loaded: /System/Library/PrivateFrameworks/MobileAsset.framework/Versions/A/MobileAsset
> dyld: loaded: /System/Library/PrivateFrameworks/StreamingZip.framework/Versions/A/StreamingZip
> dyld: loaded: /System/Library/Frameworks/MediaToolbox.framework/Versions/A/MediaToolbox
> dyld: loaded: /System/Library/PrivateFrameworks/CoreAVCHD.framework/Versions/A/CoreAVCHD
> dyld: loaded: /System/Library/Frameworks/MediaAccessibility.framework/Versions/A/MediaAccessibility
> dyld: loaded: /System/Library/PrivateFrameworks/Mangrove.framework/Versions/A/Mangrove
> dyld: loaded: /System/Library/Frameworks/CoreMedia.framework/Versions/A/CoreMedia
> dyld: loaded: /System/Library/Frameworks/VideoToolbox.framework/Versions/A/VideoToolbox
> dyld: loaded: /System/Library/PrivateFrameworks/CoreAUC.framework/Versions/A/CoreAUC
> dyld: loaded: /usr/lib/libspindump.dylib
> dyld: loaded: /usr/lib/libdscsym.dylib
> dyld: loaded: /System/Library/PrivateFrameworks/AppleVA.framework/Versions/A/AppleVA```

thomaslima, 2018-04-10:
Launched the klayout with debug level 100 and watched events play out. In the pkg folder's klayout, the menu item undo does not produce any output whatsoever. Just ignored.

In the bin folder's klayout, the debug shows that klayout captured the undo action and executed it.

The very same libraries are loaded for both events except the ones above. (In the bin's case, libspindump and libdscsym are loaded right when I actually draw a box.

thomaslima, 2018-04-10:
I just tested @Kazzz-S 's latest compilation and it works! It must have something to do with brew's qt 5.10.1, which is probably different than macport's one. :(

klayoutmatthias, 2018-04-11:
Hi Thomas,

when I debugged the issue it was pretty clear that it's a problem with  
the RTTI information the dynamic_cast operator did not found. So it's  
definitely not Qt related. It's either linker or compiler related.  
After linking with the right visibility of the class, the dynamic_cast  
implemented worked and the problem was gone for me.

I did not try HomeBrew and Qt5.10.1 myself yet. I'll try to reproduce  
the issue on that combination if you think that makes sense.

Matthias

thomaslima, 2018-04-11:
Thanks for the attention Matthias. Any idea why the built version actually works, but when I package it into a Mac bundle it stops working? I think I will use print statements to try to get a grasp of the issue.

thomaslima, 2018-04-11:
You are right. During compilation, I got a lot of warnings like these:
```ld: warning: direct access in function 'db::layer_class<db::box<int, int>, db::stable_layer_tag>::clear(db::Shapes*, db::Manager*)' from file 'dbShapes2.o' to global weak symbol 'vtable for db::layer_op<db::box<int, int>, db::stable_layer_tag>' from file 'dbShapes.o' means the weak symbol cannot be overridden at runtime. This was likely caused by different translation units being compiled with different visibility settings.```

Would that be a problem?

thomaslima, 2018-04-11:
Here's the full rebuild log after having changed the file dbshapes.cc.
[qt5.build.macos-HighSierra-release-bug107.log](https://github.com/klayoutmatthias/klayout/files/1900248/qt5.build.macos-HighSierra-release-bug107.log)

thomaslima, 2018-04-11:
I can also tell you, based on my latest debug, that the method `Shapes::undo (db::Op *op)` in `dbShapes.cc:926` is not called when I am in the bundled application. Incidentally, if I delete the shape(a box in my example) and click undo, it works.

thomaslima, 2018-04-11:
PROBLEM SOLVED
Okay now it is my time to curse myself. The "bug" was in the mac deploy tool by @Kazzz-S. It was copying over the old klayout 0.25.1 library files instead of the 0.25.2 ones because I had both in my bin folder. That is why klayout in bin ran but not the bundled one. This is going to go to my personal history of lame bugs.

Incidentally, my uploaded 0.25.2 build might have some issues. Would you consider re-uploading that?

thomaslima, 2018-04-11:
Here it is if anybody is interested: klayout-0.25.2-macOS-HighSierra-1-Qt5101brew-bb8ecb4.dmg in https://www.dropbox.com/sh/034oi1afy682lsa/AABGTzS2_-H69ESatIOWY0m7a?dl=0

klayoutmatthias, 2018-04-11:
Very good this problem is solved :-)

I have uploaded the new package to the server (bb8ecb4). Many thanks  
for providing it so quickly!

Best regards,

Matthias

commit da1daae1 (2018-04-07): Fixed #107 (Undo not working with shapes)