KLayout/klayout #109: Python 3 and GDS shape properties

klayoutmatthias, 2018-04-06:
This bug refers to [https://www.klayout.de/forum/comments.php?DiscussionID=1048](https://www.klayout.de/forum/comments.php?DiscussionID=1048).

The following code won't retrieve the shape property with key 2 on the attached layout with Python 3.

```
lv = pya.LayoutView.current()
for obj in lv.each_object_selected():
  print(repr(obj.shape.property(2)))
```

Reason: the strongly typed keys changed to 64 bit in Python 3.

This is a (slow) workaround:

```
# Don't do this too often:
key = pya.Expression.eval("2")

lv = pya.LayoutView.current()
for obj in lv.each_object_selected():
  print(repr(obj.shape.property(key)))
```

closed 2018-04-07 by klayoutmatthias

thomaslima, 2018-04-09:
This is a very obscure bug. Just reproduced on Mac. What is the main use case of shape.property, if you don't mind hinting?

Thanks,
Thomas

jurask, 2018-04-09:
Some GDS based CAM systems use these properties for storage of some additional information (for example target exposure dose in E beam lithography)

thomaslima, 2018-04-09:
Thank you @jurask !

klayoutmatthias, 2018-04-10:
Yes ... basically you can use the properties for any purpose you want. Some tools output net names for example as properties ("annotated GDS"). So you can basically select shapes by their circuit names. But AFAIK there is no common agreement what properties to use.

BTW: the bug should reproduce on Python 3/64 bit. The reason is long/long long differentiation in the compiler (actually there is no difference except on Windows, but the compiler still does not take these values as the same type). I could reproduce the issue on Linux myself.

jurask, 2018-04-13:
I have tried to build the last git version. It seems like that user properties are now correctly detected in the shape.property function. Unfortunatelly the shape.set_property(2, 'aaa') still creates #l2 instead of #2.

Thanks for any response
Juras

klayoutmatthias, 2018-04-14:
Hi Juras,

"#l2" should not be an issue. The lower case "l" indicates "long"  
(always 64bit) keys. That's required as OASIS supports long keys and  
Python 3 can potentially generate them.

When you save to GDS and load again, the "long" key will change to  
"#2" because it's normalized to "normal int" (32bit on 32bit Linux and  
on Windows, 64bit on 64bit Linux), but you can still retrieve it with  
property(2) - regardless whether the key is "#l2" or "#2". Both are  
interchangable except for different value range on Windows and 32bit  
Linux.

Thanks for testing,

Matthias