KLayout/klayout #120: unexpected result in subtraction of point and vector

thomaslima, 2018-05-11:
p1 - v1 should be a point, not a vector, right?

```
$ klayout_console -b
Python 3.6.4 (default, Mar 19 2018, 16:11:36)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
(KLayout Python Console)
>>> p1 = pya.DPoint(1,0)
>>> v1 = pya.DPoint(2,0)
>>> type(p1 - v1)
<class 'pya.DVector'>
>>> type(p1 + v1)
<class 'pya.DPoint'>
>>> type(v1 - p1)
<class 'pya.DVector'>
>>> type(p1 + -v1)
<class 'pya.DPoint'>
```

klayoutmatthias, 2018-05-13:
Hi Thomas,

but in your code v1 is a point :-)

The scheme is: 
 * point - point gives a vector  (a vector is the difference between two points)
 * point +/- vector gives a point (follows from first rule)
 * vector +/- vector gives a vector
 * vector + point gives a point
 * point + point is basically not defined, but will render a point
 * vector - point is also not defined, but will render a vector

Best regards,

Matthias

thomaslima, 2018-05-13:
Sorry. You are right. I use them interchangeably in my python code because it feels really hard to remember to place all points on the left.
Thank you!

Best,
Thomas

closed 2018-05-13 by thomaslima