KLayout/klayout #121: layout.read twice segfaults for a normal GDS

thomaslima, 2018-05-15:
When I do layout.read twice from python in a particular gds file, the second time it segfaults for some reason.
I am attaching the gds file here. It contains polygons and a Basic.TEXT pcell.

[inductor_1030pH.gds.zip](https://github.com/klayoutmatthias/klayout/files/2005571/inductor_1030pH.gds.zip)

thomaslima, 2018-05-15:
Luckily I realized my mistake:

https://www.klayout.de/forum/comments.php?DiscussionID=146

But it should still not segfault in my opinion.

klayoutmatthias, 2018-05-16:
You're right, segfaults shouldn't happen. I just wonder what's the mistake in particular ... the discussion mentioned is a bit lengthy :-)

thomaslima, 2018-05-16:
Hi Matthias. It was about not passing the layer map as a read option. That fixed this particular read.
But let me ask you this. Maybe it is minor, but when I read a second "almost identical (different names)" layout with a Basic.TEXT pcell, the first Basic.TEXT pcell gets replaced with the new one, instead of there being an extra one added.

thomaslima, 2018-05-19:
Sorry just to clarify, do you mean 'copy_tree'? That works, I believe. Except that for my application case I wanted to save individual cells with shared instances into individual .gds files and upon loading them together, the instances would point to the same cell.

klayoutmatthias, 2018-05-19:
Hi Thomas,

I found the root cause of the segfault and I think I can provide a  
solution for the "read atop" approach
also with PCells. At least for GDS2.

But if you want to merge the "same cell" into one, the "read atop"  
approach is questionable as it would
load the same cell twice (or many more times). So essentially the  
content would be duplicated. For a quick
solution this is feasible, but I think in general, a smarter approach  
is required: load all layouts individually,
identify the cells that are shared (i.e. by name or if a PCell by  
their parameters), prepare the shared
cells in a destination directory and then copy over the top cells  
while mapping the child cells to the
shared ones. Sounds a bit complicated, but I guess this is the safe  
way of doing it.

Best regards,

Matthias

closed 2018-06-10 by klayoutmatthias

thomaslima, 2018-06-15:
Thank you for looking into this. =)