KLayout/klayout #122: Feature request: Preference > autoreload

atait, 2018-05-15:
## Option for auto, manual, or prompted reload when file changes on disk
When a file changes on disk, the current behavior is to display a prompt asking whether to reload. Auto mode would be like automatically clicking "Yes". Manual mode would be like clicking "No" – user would then reload by going to the File menu. The mode could be selected by user in the preferences panel.

The hard part of detecting when a file changes is in place, defined here
https://github.com/klayoutmatthias/klayout/blob/887bcb3e044f89a8c168ad61586224b632072c4a/src/lay/lay/layMainWindow.cc#L1090

The corresponding prompt is defined on this line
https://github.com/klayoutmatthias/klayout/blob/887bcb3e044f89a8c168ad61586224b632072c4a/src/lay/lay/layMainWindow.cc#L1138

klayoutmatthias, 2018-05-16:
Hi atait,

thanks for the suggestion. This can be done of course, but I'm not sure about the consequences. Like if the file is reloaded, the selection is reset and pending edits will get lost and similar. So if someone is interacting with the view, autoload may not be a good idea.

Do you have a specific application in mind?

Matthias

lukasc-ubc, 2018-05-16:
I think the application is an external program creates a GDS file, and the user loads it in KLayout view-only.  Here are two packages that use this approach: 1) https://nazca-design.org/quick-tour/ <https://nazca-design.org/quick-tour/>, 2) https://www.mathworks.com/matlabcentral/fileexchange/46827-nicolasayotte-matlabgdsphotonicstoolbox <https://www.mathworks.com/matlabcentral/fileexchange/46827-nicolasayotte-matlabgdsphotonicstoolbox>

So these users would not be editing, hence the edits being lost don’t matter.  But you could have it pop up a window asking to save if edits have been made (dirty check before reload).

atait, 2018-05-16:
Hi Mattias,

Use case would be code-based and debugger-based interactive layout. Autoreload preference would speed this up a lot so you don’t have to switch window focus and press a button every line.

I’ve made some PDB++ (python debugger) shortcuts for layout.write. 
1. in big scripts, you can see which lines draw which polygons or place which cells, while also being able to inspect variables. 
2. in a python shell, you can type commands just to play around with the basic API elements – without having to reinitialize all the technology libraries.

Lukas is right: klayout would be used primarily as a viewer not an editor in this case. I like the idea of prompting if edits have been made in the window. The PDB scripts still need some polishing and then maybe SiEPIC-Tools?

Thanks,
Alex

amccaugh, 2018-05-18:
I'd like to chime in and support this suggestion.  I do a lot of GDS design work with python scripting, and it would be nice to immediately view the resulting .GDS from my scripts .GDS in a KLayout viewer window without clicking a prompt every time -- basically using it as an auto-refreshing viewing window which shows the results of my scripting as soon as I run it.

If we're concerned about overwriting changes accidentally, we have a few options:
1) Put the autoreload option in the Preferences, with default to OFF
2) On the "Changes have been detected" popup window, include a checkbox option that says "Automatically reload future changes to this file (tab) without prompting"
3) Or (I don't like this one very much as I think it makes a needless distinction between modes), make it auto-reload only when in Viewer mode, not in Editor mode

Personally, I'd like a combination of 1 and 2, where there's a global preference that lets me completley remove the prompting window if I want, but if I don't want to enable that, I still can do it on a per-tab basis by hiding the auto-reload prompt for just that tab

klayoutmatthias, 2018-05-26:
Hi all,

thanks for this input.

There is a detail about this proposal which makes implementation difficult: if you're writing a file, it will be growing and not be visibly fully until writing has finished. So there is a period, when the file is half-written - i.e. broken.

Unfortunately the time stamp will change at the beginning. KLayout goes by the time stamp. So if you write a file and KLayout checks the time stamp, it will see the file has changed and try to read it if auto-reload is enabled. So it will see a broken file and error out. 

That's a basic issue which happens always when writing a file takes considerable time (more than a second). I also see this issue on network share where there is a synchronization lag. So this approach bears a basic uncertainty. 

I proper solution is to establish a communication channel by which some controller can ask KLayout to reload and display. For example through a TCP/IP connection.

Matthias

klayoutmatthias, 2018-06-10:
Any comments on this?

atait, 2018-06-10:
Sorry for delay. If the timestamp changes before writing completes, then autoreload is probably not the way to go. Maybe a simpler approach would be
1. Preference for "manual-only reload" instead of auto query
2. API function to trigger a reload. Since write stalls the process, it would ensure the file is complete before reload is called
~~~
I have been focusing on [PHIDL](https://github.com/amccaugh/phidl) recently, which I think has a good solution to this feature. There is a function called `quickplot2` whose argument is a cell. Calling it will go through the polygons and references and draw them in a simple Qt viewer. This is great for debugging (e.g. in Spyder or PDB) for two main reasons:
1. No read, write, or hard drive footprint; everything stays in memory
2. Possible to debug cells within a large layout without having to draw the whole layout

atait, 2018-06-10:
Implementation might look like a modified Cell.write that creates a binary GDS. Instead of writing it to disk, it would be pipelined to a modified read somewhere on the viewer side. That read operation would be modified, of course, to get input from a stream instead of a file and also to refresh view when done. The approach should work over a network socket too since communicating processes don't have to be on the same computer.

amccaugh, 2018-06-12:
@klayoutmatthias I think your proposed solution is excellent.  Adding a network interface that allows a user to ask the application to reload the file would certainly fit the bill.  Along with @atait I would love to see something that allows me to stream polygon data directly into KLayout (perhaps also along a network pipe), but in the both near- and far-term, I think the network-socket-reload method will work nicely.

thomaslima, 2018-06-12:
Well the network solution is entirely feasible right now as a klayout salt package.

atait, 2018-06-28:
Ok following klayoutmattias' and tlima's suggestions, I've made an inter-process communication server/client. The server runs in a GUI instance of klayout. The client can be klayout or non-klayout. The client writes a gds file and requests that the klayout GUI load and display it. 

It does indeed fit within a salt package with one small exception to the reload prompt. I'll look into registering it soon. Meanwhile, I would appreciate if anyone has time to try it out and give feedback over on the repo https://github.com/atait/klayout-ipc

As far as this thread is concerned, it still would be nice to have a checkbox option to turn that "changed on disk" prompt on an off (i.e. manual reload only).

atait, 2018-07-18:
Hi all,

The IPC package I think is ready to go. It lets a lightweight layout creation script, such as standalone klayout.db (imminently) or phidl, communicate with a heavyweight klayout.app viewer. This is especially good for visual debugging, such as from Spyder. I've battle tested it on some other layout projects, and turned some of the useful patterns into examples.

It would be great if anyone, if interested, could test it out or propose some other applications of IPC. For example, a LAN socket probably wouldn't be too major of a change; does anybody have an application for that? The repo is [https://github.com/atait/klayout-ipc](https://github.com/atait/klayout-ipc)

_This issue thread can be marked closed_. The key preference for ignoring file changes on disk was already here – I just missed it before haha. Thank you all for your contributions in the thread! It ended up I think much better than the initial autoreload idea.

-Alex

closed 2020-02-22 by klayoutmatthias