KLayout/klayout #124: Feature Request - Reintroduce building with Visual Studio on Windows?

cowfarts, 2018-05-16:
As discussed in this [thread](https://www.klayout.de/forum/comments.php?DiscussionID=1004&page=1#Item_0) on the KLayout forum discussing the use of KLayout with Numpy on Windows, I would like to request the re-introducing the Visual Studio solution files that were present up to version 0.24 of KLayout if at all possible.

I would very much like to incorporate use of various packages from the numerical and scientific python ecosystem - Numpy, Scipy/SciKits, Pyomo (and its various optimization solvers) and other packages using C and Fortran code - into scripts that I could call from KLayout's Python, but doing so requires ABI compatibility between the Python installed with KLayout and the Python packages that are to be installed.  As far as I understand, use any of those packages with the MSYS2 version of Python included with KLayout means that those packages all need to be compiled using that same compiler.

This means that KLayout users cannot leverage the large body of work that is being produced in the conda and conda-forge community, nor in the standard PyPI repositories for those package where wheels do exist, for numerical computing in Python without recompiling all of those packages themselves.  In this way MSYS2 Python is not an equivalent to Anaconda, because it does not have the same ecosystem of pre-built binary packages available. 

If KLayout could be built with Visual Studio, then the included Python could be swapped out with Anaconda (or a Python downloaded from python.org as it also uses Visual Studio) by the end user after KLayout installation, and instantly the capabilities of the entire numerical Python community become accessible to users of KLayout on Windows...without the end user necessarily needing to use a C/Fortran compiler themselves.

Thank you for your consideration of this request.

klayoutmatthias, 2018-05-16:
I understand the intention and I'm aware of Anaconda. If anyone wants to support Anaconda via VisualStudio, please go ahead. But it's definitely not going to be me - I spent too many nights with this crabby compiler already and don't have the resources to go through this again. I'm very happy with the MSYS2 build system and this way I get a quality compiler with all the other prebuilt packages I need.

With VisualStudio I had countless issues which forced me to build Qt, Ruby and Python locally, switch to STLPort instead of VC's native STL and solve many weired compiler, linking and DLL issues. Not to speak of the build tool which not only takes ages, but rebuilds much too often (eventually taking a full night) ... really, I'm not missing this. Maybe VS2017 is better. Allow me to be sceptic.

Also, it wasn't my intention to pull in the whole ecosystem of Python. I'm using Python as an automation language and KLayout wasn't built as a Python driver. I finally want to release the KLayout core (non-GUI related functionality) as a standalone Python library, if that helps. But that's not going to be tomorrow.

If you're looking for a carefree system, use Linux. On Linux, KLayout uses the system's Python environment, so it's very easy to maintain.

Matthias

thomaslima, 2018-05-16:
@klayoutmatthias I would like to express my will to help you with the non-GUI functionality, since that is how I use it mostly. (I say express my will because I don't know much about interfacing c++ and python).  But having a builtin python interpreter helps a lot too for folks wanting to customize their editor. I assume both would share the same source code. That would be awesome! excited.

klayoutmatthias, 2018-05-17:
Hi Thomas,
thanks for offering your help - however there are many obstacles between a Python library and the current integrated solution. 
There are also legal questions regarding the license model: Qt requires me to adopt GPL, but that's not appropriate for a library. So the best solution would be to get rid of Qt in the non-GUI part.
I simply don't have a good plan yet.
Matthias

klayoutmatthias, 2018-06-10:
Hi all,

please see #130 for my latest efforts on this.

Matthias

ejprinz, 2018-06-19:
Hi All:

If KLayout with different Python modules works on Ubuntu, there is now a solution (not sure how enthusiastic I should be about that, but it does at least run). I've installed Ubuntu for Windows following the instructions in
https://seanthegeek.net/234/graphical-linux-applications-bash-ubuntu-windows/
and then installed the Ubuntu KLayout from klayout.de and (very much to my surprise) it is running and opening a large GDS file. The only minor issues are that the GUI comes from Ubuntu and so looks a little different from Windows, and I think Microsoft doesn't recommend that route for GUI programs anyway. I have built Python modules on Windows before and it has been a time sink. So being able to just stick to Ubuntu really helps. This is faster than Cygwin I think. And it runs Ubuntu binaries unchanged (which frankly beats me, technically, but I think this is due to Nadella the MSFT CEO. So as opposed to Cygwin one doesn't have to recompile the binaries.
Would it be possible to at least get the Python library approach to build (even with reduced features, on one platform - say Ubuntu) so we can start debugging and enhancing? I don't know where to start right now given that the build fails (of the pymod branch).

klayoutmatthias, 2018-09-15:
Just in case anyone is watching this issue ...

it was a long and painful journey, but within the pymod branch I have established compatibility with MSVC 2017. A specific issue is the availability of some third-party libraries (for the desktop app build Ruby/Python/zlib/pthreads, for the standalone Python module zlib/pthreads/expat/curl). I put together a tiny project to supply these: https://github.com/klayoutmatthias/klayout_bits4msvc2017

The current status is:
- builds and tests run cleanly (with a few understood exceptions) on Qt Creator 5.11 with MSVC 2017 64 bit using the 3rd party libs from klayout_bits4msvc2017.
- standalone Python module builds on MSVC 2017 and Anaconda with setuptools. Deployment is a bit rough since the 3rd party and runtime libs are not automatically included yet. I did not do a lot of testing yet.

Thanks to the many debug assertions inside MSVC's STL I was able to find some real issues, so there is some benefit for other platforms as well.

Matthias

closed 2019-03-21 by klayoutmatthias