KLayout/klayout #125: Package KLayout for flatpak/flathub

gyger, 2018-05-20:
Is there some interest in having KLayout packaged as a flatpak, which is a container format for applications in linux that emerged the recent years (https://flatpak.org/).
I prepared a first version of the file to generate the package, it is good enough for my computer, but could be polished if it is of interest to the project.

Flathub (the repository) likes to have the packaging done upstream, that is the reason for this request.
[de.klayout.KLayout.txt](https://github.com/klayoutmatthias/klayout/files/2020406/de.klayout.KLayout.txt)

Feel free to close the request, if you are not interested. I guess packaging is quite some annoyance in the life of a programmer.

klayoutmatthias, 2018-05-23:
Hi gyger,

thanks for spending your time on this and also thanks for sharing the results. You're right, packaging is a pain :-) 

I'm not familiar myself with flatpak. I personally do Linux deployment on a pool of Docker images and Jenkins. That's a feasible approach. But I'll take a look into flatpak.

Best regards,

Matthias

willsALMANJ, 2020-04-14:
@klayoutmatthias, if you are not interested in publishing KLayout on flathub, would you be willing to condone the community maintaining the package there? Flathub prefers upstream to publish but allows community versions if upstream is not interested. KLayout is not packaged for my distribution, and it would be nice to use it as a flatpak. I have made some small tweaks to @gyger's version here: https://github.com/willsALMANJ/flathub/tree/klayout (actually, I overlooked the fact that he had posted a config above until just and made mine from scratch but it came out almost the same).