KLayout/klayout #127: Build Error with KLayout 0.25.3 and GCC 8.1.0

t0nedef, 2018-05-28:
I run into a build error when building KLayout 0.25.3 against GCC libs 8.1.0:

> from ../../../src/db/db/dbCellMapping.cc:24:
> [...]
> /usr/include/c++/8.1.0/bits/stl_iterator_base_funcs.h:183:2: error: no match for 'operator--' (operand type is 'db::SortedCellIndexIterator')

I have a patch (see attached) that gets around the issue by defining the missing operator.

[missingOp.txt](https://github.com/klayoutmatthias/klayout/files/2045749/missingOp.txt)

plopresti, 2018-08-23:
I can confirm both the build failure and the fix.

Note that the error is that this class has "typedef std::random_access_iterator_tag iterator_category" but does not meet the requirements of a random-access iterator. The GCC 8.1.0 implementation of std::lower_bound uses the decrement operator.

klayoutmatthias, 2020-02-22:
Closing as this was fixed long ago. Cleaning up open tickets.

closed 2020-02-22 by klayoutmatthias