KLayout/klayout #128: Package Manager: Error99

tnoradam, 2018-05-29:
Hi,
every time I try to use the package manager I get:

Error 99: Network API error, fetching http://sami.klayout.org/repository.xml

Any ideas ? Mind you that I can reach this site just fine in iexplorer

Thank you !!
Thomas.

klayoutmatthias, 2018-06-02:
Hi Thomas,

Sorry for the late reply - I'm quite busy currently and I don't always monitor my mailbox. 

99 is an "Unknown network error" :-(

It's hard to tell what's the reason. Maybe some configuration prevents non-whitelisted programs to access the internet. I assume there are such settings in Windows. Proxy configuration may be another reason. I'm personally using KLayout in private and company environments and did not see these network errors so far.

Nevertheless there is a workaround: a package is nothing else than a set of files installed in a particular place.

In you home folder there should be a directory called "KLayout". There is a "salt" subfolder. If not, create one yourself. This is where the packages go.

Now you can manually download the sources from https://codeload.github.com/klayoutmatthias/xsection/zip/master and unpack them inside the "salt" folder. After restarting KLayout, you should see the XSection package and the menu entries in "Tools".

Regards,

Matthias

klayoutmatthias, 2018-06-10:
Closed as there is a workaround and the problem is not reproducible for me.

closed 2018-06-10 by klayoutmatthias

tnoradam, 2018-06-20:
Danke schoen ! Ich hab's geschafft ! Aber keine Ahnung warum Error 99 passierte.  

I think it may be because my employer doesn't give anyone admin rights. I run KLayout from MyDocuments but it may just be restricted from accessing the outside world because of that.