KLayout/klayout #131: Build Error

axhue, 2018-06-14:
Hey everyone,

I built klayout on ubuntu 18 and the build itself was successful (with a few minor warnings). However when I went to run the binary in bin-release/ I get the error `./klayout: error while loading shared libraries: libklayout_tl.so.0: cannot open shared object file: No such file or directory`. Anyone know what may be the issue?

Thanks!

klayoutmatthias, 2018-06-17:
Either set LD_LIBRARY_PATH to point to your bin directory or specify the install destination with

```
build.sh -prefix <your-install-destination> ...
```

this will set the RPATH inside the binaries and you won't need the LD_LIBRARY_PATH trick.

closed 2018-06-17 by klayoutmatthias

axhue, 2018-06-18:
Works like a charm! Thanks for the help,creation and maintenance of this repo!