KLayout/klayout #132: Build error on mac using clang++

thomaslima, 2018-06-15:
Hi, I have a build error that I don't quite now where to start looking. I still have to update Xcode to the latest version, but I was able to compile it before. I am in the `pymod` branch head (e5aa55a). Maybe you can offer an idea:

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -stdlib=libc++ -headerpad_max_install_names  -arch x86_64 -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.10 -Wl,-rpath,@executable_path/../Frameworks -Wl,-rpath,@executable_path/Frameworks -single_module -dynamiclib -compatibility_version	0.25 -current_version	0.25.3 -install_name	libklayout_QtWidgets.0.dylib -o libklayout_QtWidgets.0.25.3.dylib gsiDeclQtWidgetsAdd.o gsiQtWidgetsMain.o gsiDeclQAbstractButton.o gsiDeclQAbstractGraphicsShapeItem.o gsiDeclQAbstractItemDelegate.o gsiDeclQAbstractItemView.o gsiDeclQAbstractScrollArea.o gsiDeclQAbstractSlider.o gsiDeclQAbstractSpinBox.o gsiDeclQAccessibleWidget.o gsiDeclQAction.o gsiDeclQActionGroup.o gsiDeclQApplication.o gsiDeclQBoxLayout.o gsiDeclQButtonGroup.o gsiDeclQCalendarWidget.o gsiDeclQCheckBox.o gsiDeclQColorDialog.o gsiDeclQColormap.o gsiDeclQColumnView.o gsiDeclQComboBox.o gsiDeclQCommandLinkButton.o gsiDeclQCommonStyle.o gsiDeclQCompleter.o gsiDeclQDataWidgetMapper.o gsiDeclQDateEdit.o gsiDeclQDateTimeEdit.o gsiDeclQDesktopWidget.o gsiDeclQDial.o gsiDeclQDialog.o gsiDeclQDialogButtonBox.o gsiDeclQDirModel.o gsiDeclQDockWidget.o gsiDeclQDoubleSpinBox.o gsiDeclQErrorMessage.o gsiDeclQFileDialog.o gsiDeclQFileIconProvider.o gsiDeclQFileSystemModel.o gsiDeclQFocusFrame.o gsiDeclQFontComboBox.o gsiDeclQFontDialog.o gsiDeclQFormLayout.o gsiDeclQFrame.o gsiDeclQGesture.o gsiDeclQGestureEvent.o gsiDeclQGestureRecognizer.o gsiDeclQGraphicsAnchor.o gsiDeclQGraphicsAnchorLayout.o gsiDeclQGraphicsBlurEffect.o gsiDeclQGraphicsColorizeEffect.o gsiDeclQGraphicsDropShadowEffect.o gsiDeclQGraphicsEffect.o gsiDeclQGraphicsEllipseItem.o gsiDeclQGraphicsGridLayout.o gsiDeclQGraphicsItem.o gsiDeclQGraphicsItemAnimation.o gsiDeclQGraphicsItemGroup.o gsiDeclQGraphicsLayout.o gsiDeclQGraphicsLayoutItem.o gsiDeclQGraphicsLineItem.o gsiDeclQGraphicsLinearLayout.o gsiDeclQGraphicsObject.o gsiDeclQGraphicsOpacityEffect.o gsiDeclQGraphicsPathItem.o gsiDeclQGraphicsPixmapItem.o gsiDeclQGraphicsPolygonItem.o gsiDeclQGraphicsProxyWidget.o gsiDeclQGraphicsRectItem.o gsiDeclQGraphicsRotation.o gsiDeclQGraphicsScale.o gsiDeclQGraphicsScene.o gsiDeclQGraphicsSceneContextMenuEvent.o gsiDeclQGraphicsSceneDragDropEvent.o gsiDeclQGraphicsSceneEvent.o gsiDeclQGraphicsSceneHelpEvent.o gsiDeclQGraphicsSceneHoverEvent.o gsiDeclQGraphicsSceneMouseEvent.o gsiDeclQGraphicsSceneMoveEvent.o gsiDeclQGraphicsSceneResizeEvent.o gsiDeclQGraphicsSceneWheelEvent.o gsiDeclQGraphicsSimpleTextItem.o gsiDeclQGraphicsTextItem.o gsiDeclQGraphicsTransform.o gsiDeclQGraphicsView.o gsiDeclQGraphicsWidget.o gsiDeclQGridLayout.o gsiDeclQGroupBox.o gsiDeclQHBoxLayout.o gsiDeclQHeaderView.o gsiDeclQInputDialog.o gsiDeclQItemDelegate.o gsiDeclQItemEditorCreatorBase.o gsiDeclQItemEditorFactory.o gsiDeclQKeySequenceEdit.o gsiDeclQLCDNumber.o gsiDeclQLabel.o gsiDeclQLayout.o gsiDeclQLayoutItem.o gsiDeclQLineEdit.o gsiDeclQListView.o gsiDeclQListWidget.o gsiDeclQListWidgetItem.o gsiDeclQMainWindow.o gsiDeclQMdiArea.o gsiDeclQMdiSubWindow.o gsiDeclQMenu.o gsiDeclQMenuBar.o gsiDeclQMessageBox.o gsiDeclQPanGesture.o gsiDeclQPinchGesture.o gsiDeclQPlainTextDocumentLayout.o gsiDeclQPlainTextEdit.o gsiDeclQProgressBar.o gsiDeclQProgressDialog.o gsiDeclQPushButton.o gsiDeclQRadioButton.o gsiDeclQRubberBand.o gsiDeclQScrollArea.o gsiDeclQScrollBar.o gsiDeclQScroller.o gsiDeclQScrollerProperties.o gsiDeclQShortcut.o gsiDeclQSizeGrip.o gsiDeclQSizePolicy.o gsiDeclQSlider.o gsiDeclQSpacerItem.o gsiDeclQSpinBox.o gsiDeclQSplashScreen.o gsiDeclQSplitter.o gsiDeclQSplitterHandle.o gsiDeclQStackedLayout.o gsiDeclQStackedWidget.o gsiDeclQStatusBar.o gsiDeclQStyle.o gsiDeclQStyleFactory.o gsiDeclQStyleHintReturn.o gsiDeclQStyleHintReturnMask.o gsiDeclQStyleHintReturnVariant.o gsiDeclQStyleOption.o gsiDeclQStyleOptionButton.o gsiDeclQStyleOptionComboBox.o gsiDeclQStyleOptionComplex.o gsiDeclQStyleOptionDockWidget.o gsiDeclQStyleOptionFocusRect.o gsiDeclQStyleOptionFrame.o gsiDeclQStyleOptionGraphicsItem.o gsiDeclQStyleOptionGroupBox.o gsiDeclQStyleOptionHeader.o gsiDeclQStyleOptionMenuItem.o gsiDeclQStyleOptionProgressBar.o gsiDeclQStyleOptionRubberBand.o gsiDeclQStyleOptionSizeGrip.o gsiDeclQStyleOptionSlider.o gsiDeclQStyleOptionSpinBox.o gsiDeclQStyleOptionTab.o gsiDeclQStyleOptionTabBarBase.o gsiDeclQStyleOptionTabWidgetFrame.o gsiDeclQStyleOptionTitleBar.o gsiDeclQStyleOptionToolBar.o gsiDeclQStyleOptionToolBox.o gsiDeclQStyleOptionToolButton.o gsiDeclQStyleOptionViewItem.o gsiDeclQStylePainter.o gsiDeclQStylePlugin.o gsiDeclQStyledItemDelegate.o gsiDeclQSwipeGesture.o gsiDeclQSystemTrayIcon.o gsiDeclQTabBar.o gsiDeclQTabWidget.o gsiDeclQTableView.o gsiDeclQTableWidget.o gsiDeclQTableWidgetItem.o gsiDeclQTableWidgetSelectionRange.o gsiDeclQTapAndHoldGesture.o gsiDeclQTapGesture.o gsiDeclQTextBrowser.o gsiDeclQTextEdit.o gsiDeclQTextEdit_ExtraSelection.o gsiDeclQTimeEdit.o gsiDeclQToolBar.o gsiDeclQToolBox.o gsiDeclQToolButton.o gsiDeclQToolTip.o gsiDeclQTreeView.o gsiDeclQTreeWidget.o gsiDeclQTreeWidgetItem.o gsiDeclQTreeWidgetItemIterator.o gsiDeclQUndoCommand.o gsiDeclQUndoGroup.o gsiDeclQUndoStack.o gsiDeclQUndoView.o gsiDeclQVBoxLayout.o gsiDeclQWhatsThis.o gsiDeclQWidget.o gsiDeclQWidgetAction.o gsiDeclQWidgetItem.o gsiDeclQWizard.o gsiDeclQWizardPage.o  -F/usr/local/Cellar/qt/5.10.1/lib -L/Users/tlima/Documents/Princeton/Prucnal/GitProjects/klayout/qt5.build.macos-HighSierra-release/gsiqt/qt5/QtWidgets/../../.. -lklayout_tl -lklayout_gsi -lklayout_qtbasic -lklayout_QtGui -framework QtDesigner -framework QtWidgets -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework QtXml -framework QtPrintSupport -framework QtMultimediaWidgets -framework QtMultimedia -framework QtNetwork -framework QtSvg -framework QtXmlPatterns -framework QtSql -framework OpenGL -framework AGL
Undefined symbols for architecture x86_64:
  "gsi::qtdecl_QEvent()", referenced from:
      __GLOBAL__sub_I_gsiDeclQGestureEvent.cc in gsiDeclQGestureEvent.o
      __GLOBAL__sub_I_gsiDeclQGraphicsSceneEvent.cc in gsiDeclQGraphicsSceneEvent.o
  "gsi::qtdecl_QObject()", referenced from:
      __GLOBAL__sub_I_gsiDeclQAbstractItemDelegate.cc in gsiDeclQAbstractItemDelegate.o
      __GLOBAL__sub_I_gsiDeclQAction.cc in gsiDeclQAction.o
      __GLOBAL__sub_I_gsiDeclQActionGroup.cc in gsiDeclQActionGroup.o
      __GLOBAL__sub_I_gsiDeclQButtonGroup.cc in gsiDeclQButtonGroup.o
      ___cxx_global_var_init in gsiDeclQCompleter.o
      __GLOBAL__sub_I_gsiDeclQDataWidgetMapper.cc in gsiDeclQDataWidgetMapper.o
      __GLOBAL__sub_I_gsiDeclQGesture.cc in gsiDeclQGesture.o
      ...
  "gsi::qtdecl_QAbstractItemModel()", referenced from:
      __GLOBAL__sub_I_gsiDeclQDirModel.cc in gsiDeclQDirModel.o
      __GLOBAL__sub_I_gsiDeclQFileSystemModel.cc in gsiDeclQFileSystemModel.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [../../../../qt5.build.macos-HighSierra-release/libklayout_QtWidgets.0.25.3.dylib] Error 1
make[2]: *** [sub-QtWidgets-make_first] Error 2
make[1]: *** [sub-qt5-make_first] Error 2
make: *** [sub-gsiqt-make_first] Error 2```

klayoutmatthias, 2018-06-17:
Hi Thomas,

I have tested on Linux so far, but I see similar issues on Windows. I am about to update a few dependencies and I'm fairly sure these issues are gone then.

Please be a little patient :-)

Thanks, 

Matthias

ejprinz, 2018-06-19:
Hi Matthias, what's your primary development platform for this? It may be fastest to follow you on that and make one platform work first.

klayoutmatthias, 2018-06-20:
Hi Erwin,

I'm using Ubuntu 16 myself. But I do CI on CentOS 6+7 and Windows too. Right now the pymod branch is about to stabilize, so there is a chance these platforms will be available as trial downloads (.deb/.rpm/.exe) soon.

Ubuntu 18 is something I'll need to add anyway.

Best regards,

Matthias

thomaslima, 2018-06-22:
This error is gone as of 0d6da0e

closed 2018-06-22 by thomaslima