KLayout/klayout #134: Error in cell.fill_region

EmilKleijn, 2018-06-21:
In a Ruby script I am running, I use the fill_region function to add tiling. I have now encountered a problematic file that generates the following error in KLayout v0.25.3:
Internal error: ../../../src/db/db/dbPolygonGenerators.cc:303 m_open.emtpy () was not true in Cell:fill_region

The fill_region command runs sucessfully in v0.24.8

Required files for reproducing the issue:
[Files.zip](https://github.com/klayoutmatthias/klayout/files/2123269/Files.zip)

Includes the export script, the fill cell, and the problematic cell.

Regards

klayoutmatthias, 2018-06-23:
Thanks a lot for the test case. I will look into this.

Matthias

klayoutmatthias, 2018-06-24:
Hi,

I'm able to reproduce this issue and isolate the internal data which cause this error. Internal a big polygon with zero-width spikes is generated. I assume due to some implicit normalization these got deleted in 0.24.x.

I'll try to provide a fix this this. I'm afraid there is no simple workaround currently except using 0.24.

Best regards,

Matthias

EmilKleijn, 2018-06-25:
Hi Matthias,

Thanks for looking into this! I'm glad you can reproduce the issue.

Regards,
Emil

commit 94387529 (2018-06-25): Fixed #134

The issue was caused by an internal error in the edge processor.
Effectively the weak attractor scheme was causing this problem.
As the weak attractors are making things worse rather than
better I dropped them.

In theory, the weak attractors render an edge undisturbed by
neighboring intersection points, but in cases or parallel edges
this lead to problems: omitting cut points violates the output
edge configuration warranties the the polygon stitcher fails.

In addition, to maintain the solution for bug #74, the cut point
capture condition was relaxed, so that edge crossing the exa [... cut]
commit 23842be0 (2018-11-25): Fixed #134

The issue was caused by an internal error in the edge processor.
Effectively the weak attractor scheme was causing this problem.
As the weak attractors are making things worse rather than
better I dropped them.

In theory, the weak attractors render an edge undisturbed by
neighboring intersection points, but in cases or parallel edges
this lead to problems: omitting cut points violates the output
edge configuration warranties the the polygon stitcher fails.

In addition, to maintain the solution for bug #74, the cut point
capture condition was relaxed, so that edge crossing the exa [... cut]
closed 2018-06-25 by klayoutmatthias