KLayout/klayout #136: Feature Request: Read Only for Cells

raphaelvalentin, 2018-06-22:
Dear Developers, 
when Editing a Top, it is still possible to modify the sub cells (layout view, not box view).
I would be very interested to have a read only option on cell when I am editing top layout. 
What do you think?
Thank you very much,

onaspringday, 2018-06-22:
I had the same wish before but then I found out an alternative. I now have the 'selection mode' set to 'select top level objects only' and then do in-place editing by selecting an instance of a cell to edit and then access it with Display/Descend (Ctrl+D) . And Display/Ascend (Ctrl+A) to go back up in the hierarchy. 
Would that work for you?

raphaelvalentin, 2018-06-22:
Yes, absolutely better ... :) I guess my issue is solved.

klayoutmatthias, 2018-06-29:
Very good! I'll close it then.

closed 2018-06-29 by klayoutmatthias