KLayout/klayout #137: KLayout MacOS - The operation couldn’t be completed. (OSStatus error 99999.)

lukasc-ubc, 2018-06-25:
Can't start the application.

> The operation couldn’t be completed. (OSStatus error 99999.)

This is running MacOS High Sierra, 10.13.4, and klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.dmg

thomaslima, 2018-06-25:
I think this is the version I compiled. Did this just happen out of nowhere? Can you launch from terminal to try to check what is generating the error?

thomaslima, 2018-06-25:
@lukasc-ubc i need to compile a version with the python ssl module embedded, otherwise pip 10 is going to fail to install these modules automatically, but this is a discussion to be had in siepic-tools. i don't know why python errors are causing this OS error 99999. my klayout doesn't crash. wish i could reproduce this right now.

lukasc-ubc, 2018-06-26:
In the older versions, I get the same list of errors in the File | Log Viewer.

Can you create an error in one of the lyp libraries, e.g., replace import numpy with import numpyxxx?

thomaslima, 2018-06-26:
Errors are treated normally over here.
[image]
Maybe try reinstalling from the .dmg?

lukasc-ubc, 2018-06-26:
1) Log viewer: doesn’t let me resize.

2) I fixed all the SiEPIC errors so that from Terminal it loads error free.  Note that I added a MessageBox query for all the places you had auto-pip-install.

It still doesn't start for me.  Here is the output of the Console.App:

[image]

I will run an OS update and see if that fixes it.

lukasc-ubc, 2018-06-26:
hmm. didn't help.  Yes, I did right-click open...  it did a check for a while, then asks if I want to open. then waits for a while, then the error.

thomaslima, 2018-06-26:
Can you show me the result of `xattr -l /Applications/klayout.app/` from terminal?

lukasc-ubc, 2018-06-26:
On another MacOS computer, I was able to get it working.  So something specific with my desktop or user account.

iMac-Hickup:~ lukasc$ xattr -l /Applications/klayout.app/
com.apple.quarantine: 0181;5b311c9b;Chrome;B495252B-B98B-4C48-832C-4F9206493D0D

on the working computer, I downloaded it using Safari...

thomaslima, 2018-06-26:
There it says that this file was downloaded using Chrome. If you trust the source, maybe try removing these special quarantine attributes? `xattr -dr com.apple.quarantine /Applications/klayout.app`

lukasc-ubc, 2018-06-26:
**wow, that fixed it!**  

prior to that, I tried on a different user account, but that didn't help.

thomaslima, 2018-06-26:
Great! I have no idea how you got there though. I think the bug might be in the OS itself.

I am working on compiling a new version with Qt5.11, which fixes the resizing issue.

thomaslima, 2018-06-26:
Done in #100 .

thomaslima, 2018-09-21:
@klayoutmatthias I believe this issue can be closed.

stefanottili, 2024-01-31:
this issue should be closed, no ?