KLayout/klayout #144: Item description in Salt package manager missing on Linux

philippm1988, 2018-07-16:
In Salt package manager on Linux, the item description for all items is missing.
On Windows this problem does not exist.
KLayout was built with the sources from https://github.com/klayoutmatthias/klayout/tree/v0.25.3
with the following command: ./build.sh ( -qt5 ) 

Here is how it looks like on Linux ( openSUSE Leap 42.3 / openSuse Leap 15 ):

[image]

... and how it looks like on Windows ( Windows 10 ):

[image]

What is the reason for this strange behavior ?

klayoutmatthias, 2018-07-16:
Hi,

this is strange - I do see the descriptions. I use Ubuntu 16.04. 

Maybe it's specific to Qt5 and/or OpenSuSE. I'll take a look.

Thanks, 

Matthias

klayoutmatthias, 2018-07-27:
Hi Phillip,

I have just built KLayout on plain OpenSuSE 42.3 with Qt 4.8.6, Ruby 2.1.0 and Python 3.4. I don't see that issue myself. 

I have postponed OpenSuSE 42.3 support bacause I saw some issues in the test suite. They seem to be fixed now, so I may be able to provide an RPM for this.

But this means it's not OpenSuSE or the compiler, but my guess is Qt5. Which is the Qt5 version you used?

Thanks,

Matthias

klayoutmatthias, 2018-07-28:
I have established a OpenSuSE 42.3 build on the CI system. If will become active for 0.25.4. Maybe you want to try this "official" RPM: http://www.klayout.org/downloads/staging/opensuse42_3/klayout-0.25.4-0.x86_64.rpm

It's based on the standard packages of OpenSuSE 42.3. With theses packages I don't see the issue you show in the screenshots.

Matthias

philippm1988, 2018-07-30:
Hi Matthias,

I build it on openSuse Leap 15 with the standard packages and Qt 5.9 ( libqt5-qtbase, libqt5-qtmultimedia, ... ) with the option -qt5. 
I got the same result when using Qt4 ( libqt4, libqt4-devel )

Would be great if you can have a look and if possible provide an rpm for openSuse Leap too !

Thanks,

Philipp

closed 2018-07-30 by klayoutmatthias

klayoutmatthias, 2018-07-31:
For OpenSuSE Leap 15, please try http://www.klayout.org/downloads/staging/opensuse15/klayout-0.25.4-0.x86_64.rpm

Matthias

philippm1988, 2018-08-01:
Hi Matthias,

for OpenSuse 15, the item descriptions  in package manager are shown now, but after I installed the above rpm on the latest OpenSuse 15 version, i get the following error each time I close klayout (x):

ERROR: Signal number: 11
Address: 0x0
Program Version: KLayout 0.25.4 (2018-07-31 r5efe5d20)

Backtrace:
/usr/lib64/klayout/libklayout_lay.so.0 +0x2b58e0 lay::enable_signal_handler_gui(bool) [??:?]
/lib64/libpthread.so.0 +0x122e0 __restore_rt [??:?]
/usr/lib64/qt4/plugins/inputmethods/libqimsw-multi.so +0x2c93 ?? [??:0]
/usr/lib64/qt4/plugins/inputmethods/libqimsw-multi.so +0x2d3e ?? [??:0]
/usr/lib64/libQtGui.so.4 +0x24b2be QApplicationPrivate::_q_alertTimeOut() [??:?]
/usr/lib64/libQtGui.so.4 +0x1da12d QApplication::~QApplication() [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0xeb8e9 lay::GuiApplication::~GuiApplication() [??:?]
klayout +0x2d15
/usr/lib64/klayout/libklayout_rba.so.0 +0x1e970
/usr/lib64/libruby2.5.so.2.5 +0x1ab9d1
/usr/lib64/libruby2.5.so.2.5 +0x1bed83
/usr/lib64/libruby2.5.so.2.5 +0x1b5a40
/usr/lib64/libruby2.5.so.2.5 +0x1bafed
/usr/lib64/libruby2.5.so.2.5 +0x97454
/usr/lib64/libruby2.5.so.2.5 +0x9928d ruby_exec_node
/usr/lib64/libruby2.5.so.2.5 +0x9b65e ruby_run_node
/usr/lib64/klayout/libklayout_rba.so.0 +0x1eec4 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
klayout +0x2674
/lib64/libc.so.6 +0x20f4a __libc_start_main
klayout +0x278a

klayoutmatthias, 2018-08-12:
Hi Michael,

sorry for the late reply. I found time now to play with a blank  
OpenSuSE 15 installation, but I failed to reproduce the issue.

This is my configuration:

  * Plain Docker image: opensuse/leap:15  on  Ubuntu-16.04.4LTS host
  * X11 enabled (see  
http://fabiorehm.com/blog/2014/09/11/running-gui-apps-with-docker/)
  * Packages installed: openssh java-1_8_0-openjdk rpm-build xvfb-run  
sudo wget lato-fonts
  * Download image from:  
http://www.klayout.org/downloads/staging/opensuse15/klayout-0.25.4-0.x86_64.rpm
  * After install run with "Qt_X11_NO_MITSHM klayout"

I can install packages and quit KLayout without error messages.

So the problem seems to be a configuration issue. From the stack trace  
I'd say it's related to  
/usr/lib64/qt4/plugins/inputmethods/libqimsw-multi.so - but these  
traces can be misleading.

Kind regards,

Matthias

commit a272cd36 (2018-07-30): Fixed #144 (description of salt packages isn't shown)

Plus: the text color of the description now uses the
palette and the color contrast better when the item
is selected.