KLayout/klayout #146: Automatical installation of packages fails

philippm1988, 2018-07-25:
When trying to automatically install a package with the command "klayout -y <package_name>" over a repository URL ( https:// ) where authentication is needed, the desired package does not get installed even after authenticating via the authentication dialog in Klayout.

Whats the problem here ?
Any advice how to solve that ?

Thanks !

thomaslima, 2018-07-25:
Phillip, would you mind sharing the OS and version of klayout you installed? And, if possible, the package you want to install?

philippm1988, 2018-07-26:
The OS is Linux ( openSuse 42.3 ) and the installed version of KLayout is 0.25.3.
I want to install packages from a local svn repository with setting KLAYOUT_SALT_MINE=<my_local_url>.

klayoutmatthias, 2018-07-28:
Hi Phillip,

authentication is not supported currently for package installation with -y. As this mode is unattended there is not password dialog. I could add options for configuring username/password for -y, but this will mean you have to put the password as clear text into some script if you want automatic installation. This is usually not the preferred way of handling credentials. 

I need to say that working with authenticated sources is not a strength of KLayout's network access module. KLayout only caches passwords inside the process, not between processes. Neither does it try to access any passwort cache provided by your system or Subversion. Hence working with authenticated sources is always somewhat tedious. 

KLayout allows building against libcurl rather than QtNetword. libcurl offers more options for authentication than "basic". Maybe it's possible for example to utilize Kerberos. 

Kind regards,

Matthias

klayoutmatthias, 2018-07-28:
Hi Philipp, 

a quick fix is to supply username and password in the Salt Mine URL:

```
export KLAYOUT_SALT_MINE=https://user:password@www.domain.com/path/to/subversion/repository.xml"
```

Regards,

Matthias

philippm1988, 2018-07-30:
Hi Matthias,

but the problem with the clear text password persists with this solution.
The only feasible solution would be a separate svn repository for the vendor specific packages with the same credentials for all users with read only access.

Regards,
Philipp