KLayout/klayout #153: Rounded path - Crash on simple rectangular path when clicking Apply.

gyger, 2018-08-08:
I try to create a ROUND_PATH from a guide path and I get the following crash (see the attached screenshot.) when I press Apply plus OK. If I just press OK, the curve is accepted and everything works fine.

[image]

If I press Ok and Apply again, I get the following "Problem Details". Pressing OK closes the program.

```
Exception code: 0xc0000005
Program Version: KLayout 0.25.3 (2018-05-04 r887bcb3) x86

Backtrace:
0x1948032 - (klayout_laybasic.dll) +1081394
0x6c0fb0d2 - (klayout_edt.dll) +241874
0x66ac3517 - (Qt5Core.dll) +1586455
0x25ffe2f - (Qt5Widgets.dll) +1048111
0x2601076 - (Qt5Widgets.dll) +1052790
0x25ffb03 - (Qt5Widgets.dll) +1047299
0x2601203 - (Qt5Widgets.dll) +1053187
0x25667ea - (Qt5Widgets.dll) +419818
0x66aa32cf - (Qt5Core.dll) +1454799
0x66aa380c - (Qt5Core.dll) +1456140
0x12067f7 - (Qt5Gui.dll) +157687
0x1207f88 - (Qt5Gui.dll) +163720
0x7465be6b - (USER32.dll) +245355
0x7465833a - (USER32.dll) +230202
0x74657bee - (USER32.dll) +228334
0x746579d0 - (USER32.dll) +227792
0x66ae7705 - (Qt5Core.dll) +1734405
0xeab4f77 - (qwindows.dll) +479095
0x66aa2a8a - (Qt5Core.dll) +1452682
0x66aa97e2 - (Qt5Core.dll) +1480674
0x63ece058 - (klayout_lay.dll) +57432
0x63ed26bb - (klayout_lay.dll) +75451
0x401f5d - (klayout_app.exe) +8029
0x67d417be - (klayout_rba.dll) +6078
0x710ddf59 - (msvcrt-ruby240.dll) +1433433
0x710db74d - (msvcrt-ruby240.dll) +1423181
0x710db74d - (msvcrt-ruby240.dll) +1423181
0x710ee62e - (msvcrt-ruby240.dll) +1500718
0x710eeb0e - (msvcrt-ruby240.dll) +1501966
0x710deea4 - (msvcrt-ruby240.dll) +1437348
0x710def42 - (msvcrt-ruby240.dll) +1437506
0x710e3d8d - (msvcrt-ruby240.dll) +1457549
0x70f9cb2e - (msvcrt-ruby240.dll) +117550
0x710db74d - (msvcrt-ruby240.dll) +1423181
0x70f9cb2e - (msvcrt-ruby240.dll) +117550
0x710db74d - (msvcrt-ruby240.dll) +1423181
0x70f9cb2e - (msvcrt-ruby240.dll) +117550
0x710ace77 - (msvcrt-ruby240.dll) +1232503
0x710962bd - (msvcrt-ruby240.dll) +1139389
0x7105a52f - (msvcrt-ruby240.dll) +894255
0x71096b79 - (msvcrt-ruby240.dll) +1141625
0x70fccfc2 - (msvcrt-ruby240.dll) +315330
0x710e8212 - (msvcrt-ruby240.dll) +1475090
0x70fe06ad - (msvcrt-ruby240.dll) +394925
0x401a91 - (klayout_app.exe) +6801
0x40885d - (klayout_app.exe) +34909
0x4013e3 - (klayout_app.exe) +5091
0x747c8484 - (KERNEL32.DLL) +99460
0x77da2fea - (ntdll.dll) +405482
0x77da2fba - (ntdll.dll) +405434

```

klayoutmatthias, 2018-08-12:
Hi,

thanks for reporting this issue. I'm trying to reproduce the issue but I failed so far.

What is the radius you're using? And what's your OS?

Thanks,

Matthias

gyger, 2018-08-12:
I am on Windows 10, x64 and on the newest version. The radius I use is 30um. I try to create a small file where I can see the issue and will post it here.

klayoutmatthias, 2018-08-12:
Thanks. I'm on Linux, but I can try on W10. 

This is what works for me:
* run KLayout in editor mode
* create a new layout
* create a new layer 1/0
* draw some path on 1/0
* select this path - for example an L-shaped one
* use "Edit/Selection/Convert To PCell"
* select the PCell (not the path) and double click
* select radius 30, press "Ok"
* select the path as guiding shape, double click
* enter the values you show in your screenshot
* Press Apply, then Ok

I do not see the error you get, but maybe it's about the order how things are done.

Regards,

Matthias

lukasc-ubc, 2018-08-13:
I've seen this error many times before during the past year.

It doesn't happen all the time.  To get it to crash, I edited the vertices several times, each time pressing Apply.  Here is a crash screenshot. Note that I didn't get the mp_v messages, which I previously had as well.

[image]

Signal number: 11
Address: 0x58
Program Version: KLayout 0.25.3 (2018-05-05 r9708d35d)

Backtrace:
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/libklayout_lay.0.dylib +0x21934a _ZN3lay25enable_signal_handler_guiEb
/usr/lib/system/libsystem_platform.dylib +0x1f5a _sigtramp
/usr/lib/system/libsystem_platform.dylib +0xffffffe08fd8e570 _sigtramp
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/libklayout_edt.0.dylib +0x8d6b2 _ZN3edt17EditGridConverter17from_string_pickyERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEERN2db6vectorIdEE
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x2bf2c2 _ZThn16_N3lay14PropertiesPageD0Ev
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x21f36b _ZN11QMetaObject8activateEP7QObjectiiPPv
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x107d2f _ZNK15QAbstractButton11isCheckableEv
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x107bc7 _ZNK15QAbstractButton11isCheckableEv
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x108e2f _ZN15QAbstractButton17mouseReleaseEventEP11QMouseEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x4ac3f _ZN7QWidget5eventEP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0xfffd _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x12d8d _ZN12QApplication6notifyEP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/libklayout_lay.0.dylib +0x55b3b _ZN3lay14GuiApplication6notifyEP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1ee798 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x10960 _ZN19QApplicationPrivate14sendMouseEventEP7QWidgetP11QMouseEventS1_S1_PS1_R8QPointerIS0_Eb
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x6ab0a _ZN14QDesktopWidget11qt_metacallEN11QMetaObject4CallEiPPv
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x697f5 _ZN14QDesktopWidget11qt_metacallEN11QMetaObject4CallEiPPv
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0xfffd _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0x1139a _ZN12QApplication6notifyEP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/libklayout_lay.0.dylib +0x55b3b _ZN3lay14GuiApplication6notifyEP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1ee798 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtGui.framework/Versions/5/QtGui +0x2e66b _ZN22QGuiApplicationPrivate17processMouseEventEPN29QWindowSystemInterfacePrivate10MouseEventE
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/QtGui.framework/Versions/5/QtGui +0x15b8b _ZN22QWindowSystemInterface22sendWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x2a22e qt_plugin_instance
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x2aaf1 qt_plugin_instance
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0xa3a61 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x15d47c __CFRunLoopDoSource0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x864c0 __CFRunLoopDoSources0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x8593d __CFRunLoopRun
[... cut ...]
/System/Library/Frameworks/Ruby.framework/Versions/2.3/Ruby +0x3516e ruby_exec_node
/System/Library/Frameworks/Ruby.framework/Versions/2.3/Ruby +0x35061 ruby_run_node
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/Frameworks/libklayout_rba.0.dylib +0x22583 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Applications/klayout-0.25.3-macOS-HighSierra-1-Qt5101brew-Py364-9708d35.app/Contents/MacOS/klayout +0xaa2b
/usr/lib/system/libdyld.dylib +0x1015 start
/usr/lib/system/libdyld.dylib +0xffff80008d2e9001 start

klayoutmatthias, 2018-08-13:
Hi Lukas,

thanks for sharing the stack trace. Frankly I'm a bit puzzled as it indicates a crash in a function where a crash hardly is possible. But every stack trace is a piece of the solution. I'll try to track this issue down.

Best regards,

Matthias

klayoutmatthias, 2018-08-25:
Hi all,

thanks again for reporting this issue. I was able to locate the root cause thanks to valgrind. 
I'll provide a fix.

Matthias

commit 842f39da (2018-08-25): Fixed #153 (crash on Apply when changing guiding shape properties)
closed 2018-08-25 by klayoutmatthias