KLayout/klayout #156: Application freeze on closing KLayout, MacOS

lukasc-ubc, 2018-08-13:
Have you ever noticed that KLayout often freezes when you try to close it?  This is in the latest version on MacOS, 0.25.3 with Brew, py3, Qt5101.

If I just open KLayout and close it, it works fine.  But after doing some layout work, it freezes upon closing…. This didn’t happen a long time ago…. I think it started sometime in the past 6 months.  It also happens in KLayout 0.25.2, Qt510, an old version from early 2018.

I can reproduce this in SiEPIC-Tools, after running a simulation with INTERCONNECT.  Perhaps there is some garbage cleanup routing that gets stuck?

To debug, I started klayout from the terminal. Here is what I see in the terminal after trying to close:  an infinite loop.

QMutex::lock: mutex lock failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument
QMutex::lock(): cv wait failure: Invalid argument

Any idea how to prevent this?

thanks

klayoutmatthias, 2018-08-25:
Hi Lukas,

things like this used to happen in the cleanup phase of the Python interpreter. The sharing of objects between Python an Qt needs some bookkeeping about the ownership of an object. Effects like the ones you see happen if Qt and Python try to delete the same object. 

I tried to find a leak in the mentioned bookkeeping code but with my current configuration (Qt 4.8.7 and Python 3.5) is wasn't able to see one. Even valgrind - which usually is good in spotting such issues - is quiet. 

I'll try to watch this issue. Usually it's a matter of finding the right trigger.

Best regards,

Matthias

klayoutmatthias, 2018-09-19:
I have found a few memory corruption issues while doing the MSVC port. Those might explain the issue seen. I have ported the fixes to 0.25.x and will release them with 0.25.5. Maybe this cleaning will solve this issue.

Matthias

lukasc-ubc, 2018-09-19:
Thanks Matthias. I haven’t been able to come up with a small test case, and I see this in a pretty complicated set of scripts in SiEPIC-Tools.  I’ll let you know if the next version fixes this.

thomaslima, 2018-09-22:
@lukasc-ubc sorry to hear! If you can produce a minimal test case without connection to INTERCONNECT, I can give it a try.

klayoutmatthias, 2018-09-22:
> I tried version qt5.pkg.macos-HighSierra-B37-release-0.25.5-2236a76.dmg
>
> Same problem.
>
> Using SiEPIC-Tools and SiEPIC_EBeam_PDK, I create a layout and run a
> simulation using an external program (Interconnect).  Simulation
> runs. I quit Interconnect. Then I quit KLayout and it freezes.

Hi Lukas,

thanks for the testcase - I'll try to reproduce the issue with my  
VirtualBox instance.

Best regards,

Matthias

klayoutmatthias, 2018-11-29:
Hi Lukas,

while looking at similar issue on CentOS7, I was able to fix this issue there. It will be released with the next minor release. Maybe this patch will fix this problem as well.

Matthias

lukasc-ubc, 2019-10-29:
Hi Matthias,

I wanted to provide an update: this issue is still persisting in KLayout 0.26, using the build #381 by Alex Tait.

I added a new menu item "Force Quit" in SiEPIC-Tools which uses pya.Application.quit() -- not too pretty, but at least exists quickly...

stefanottili, 2024-01-31:
There is the issue that with klayout + qt6 the app doesn't quit, but I've not seen any issues with qt5 for years now ...