KLayout/klayout #160: MacOS build for 0.25.4

Kazzz-S, 2018-08-29:
@klayoutmatthias 

Thank you for making the maintenance release of 0.25.4!
I've built different binary packages for MacOS as usual and available below.
https://www.dropbox.com/sh/82bqswuidiuakks/AAAYUP6OuVWn5aNZrG36rth4a?dl=0 

Srl.No.|Target OS|Build OS(1)|Qt version(2)|Binary Package
--------|-----------|----------|------------|------------------
1|Yosemite(10.10)|Yosemite(10.10)|5.10.1(3)|klayout-0.25.4-MacOSX-Yosemite-1-Qt5101mp.dmg
2|El Capitan(10.11) Sierra(10.12)|El Capitan(10.11)|5.11.1|klayout-0.25.4-MacOSX-ElCapitan-1-Qt5111mp.dmg
3|HighSierra(10.13)|HighSierra(10.13)|5.11.1|klayout-0.25.4-macOS-HighSierra-1-Qt5111mp.dmg

Notes:
(1) using Ruby and Python bundled with OS
(2) using packages from MacPorts
(3) looks the last available version on Yosemite

Best regards,
Kazzz

klayoutmatthias, 2018-08-30:
Hi Kazunari,

many thanks for your efforts! I have uploaded the packages to the server.

Best regards,

Matthias

closed 2018-09-01 by klayoutmatthias

thomaslima, 2018-09-18:
Sorry for coming here late. I've pushed a pull request for the binaries. Please check https://github.com/klayoutmatthias/klayout/pull/168. I think the builds are in order (logs are attached), but I'd appreciate help testing klayout in respective virtual machines. Thank you @klayoutmatthias and @Kazzz-S .

Kazzz-S, 2018-09-30:
## Quick Test on macOS 10.14 Mojave ##

The binary package built for macOS10.13 (HighSierra) looks work on macOS 10.14 (Mojave), too,  without any change :-). 

I've tested on a virtual machine as usual. 
Thank you for your continuous contributions: @klayoutmatthias and @thomaslima.

Just for your information,
Kazzz

[image]
[image]
[image]

klayoutmatthias, 2018-09-30:
Hi Kazzz,

thanks a lot - that's good to know.

Best regards,

Matthias